squishykid/solax #2: Incorrect sensor name

matthewjporter, 2019-06-12:
sensor.battery_energy seems to be pulling the wrong data, it looks to be the total montly yeild/generation

squishykid, 2019-06-15:
This sensor is correct for my inverter, so it seems there's something different about how your model is encoding the data. Let's figure it out!

Are you able to access the inverter in question and post the results you get when you hit the real time data endpoint? This would be "http://{inverter_ip}/api/realTimeData.htm".

Something like:

`{"method":"uploadsn","version":"Solax_SI_CH_2nd_20160912_DE02","type":"AL_SE","SN":"XXXXXXX",
"Data":0.5,0.4,202.0,194.3,2.0,234.0,444,40,17.0,238.2,-15,101,77,56.01,-6.36,-357,27,92,0.0,126.0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,373.90,38.60,0,0,0,0,0,0,0,50.02,0,0,0.0,0.0,0,0.00,0,0,0,0.00,0,8,0,0,0.00,0,8],
"Status":"2"}`

Please also provide a screenshot of the relevant information from another dashboard. Then we can compare the sensor values.

I imagine that depending on the firmware/version etc, the index of the data being sent from the inverter might be changing. But with those two we should be able to start figuring it out.

There is another description of the endpoint here: [https://github.com/GitHobi/solax/wiki/direct-data-retrieval](https://github.com/GitHobi/solax/wiki/direct-data-retrieval).

The mapping that this library uses is found here, in a global called `INVERTER_SENSORS` : [https://github.com/squishykid/solax/blob/master/solax/solax.py](https://github.com/squishykid/solax/blob/master/solax/solax.py)

squishykid, 2019-06-15:
Also, I just noticed that you wrote some of the reference I'm using! Awesome! [https://community.home-assistant.io/t/solax-solar-inverter-setup-guide/48008](https://community.home-assistant.io/t/solax-solar-inverter-setup-guide/48008)

matthewjporter, 2019-06-18:
Thats whats the comminity is for, Glad it helped with the component

Will get the info required and paste basck, It might be a firmware update on mine but getting it is a challenge from Solax

squishykid, 2019-06-30:
@matthewjporter have you had a chance to take a look?

matthewjporter, 2019-06-30:
Sorry for the delay, some personal matters to attend to and away on business.

My inverter respond with the following
SK-SU5000e:
{"method":"uploadsn","version":"Solax_SI_CH_2nd_20160912_DE02","type":"AL_SE","SN":"xxxxxxxxxx","Data":[1.2,1.6,271.7,122.0,3.7,245.8,878,42,16.4,5576.8,-43,326,195,50.06,-7.33,-368,5,38,0.0,2190.9,,,,,,,,,,,,,,,,,,,,,,0.00,0.00,,,,,,,,50.04,,,0.0,0.0,0,0.00,0,0,0,0.00,0,9,0,0,0.00,0,9],"Status":"2"}

Looks like we are using the same "Version" on my SK-SU unit which is the one I am testing on

As for the Battery Energy (sensor.battery_energy), I have lined up as close I can the two displays I have, One using nodered (right) and the other using the component (left)
[image]

As you can see, the Battery Energy amount matches the Monthly Yield of my unit

Here is my nodered flow that matches the realTimeData.htm for the Monthly Yield:
solarTotal2 = { payload: realtimeArray[0][19], topic: "solax/energy/x1a-solarTotal2" };

You already have the sensor.battery_power which matches mine:
battPower = { payload: realtimeArray[0][15], topic: "solax/energy/x1a-battPower" };

What are you using sensor.battery_energy for as the term power/energy are the same - Is this not actually needed if you alreadt refer to it as power

This is my mappings
[image]

These are the screenshots from my Solax Portal
[image]

[image]

Note: Figuers wont match up, taken at different times

matthewjporter, 2019-06-30:
On the history data, I only really got the Yield for Today, Month, Year and Total and the only one I cared about was the solarTotal as thats the "All Time" solar generated

[image]

squishykid, 2019-07-01:
Just checking, since there appears to be other differences in the field names too. I think the names I have set for the other fields are correct though.

I'll change the 'Battery Energy' to 'Monthly Energy'

matthewjporter, 2019-07-01:
@squishykid 
Thanks, Yes, it looks like the other are correct, I just have them named in my NodeRed how i like them :)

squishykid, 2019-07-02:
Fixed by #6

closed 2019-07-02 by squishykid

matthewjporter, 2019-07-24:
Just wondering if this has been rolled out to HA yet, Got all the new sensor in 0.96.3 or 4 i think but now  the sensor with the invertserial but still as battery_energy for the monthly yield

Cheers