squishykid/solax #5: SolaX Power X3-8.0 Support

squishykid, 2019-07-01:
@kprzeb

We can discuss here

rest77, 2019-07-05:
Hi!

I can't manage to connect to my inverter X3 Hybrid G3. I believe the API is changed. I use the WiFi with firmware USB 2.33.7. Does anybody know something about that?

/Reik

kprzeb, 2019-07-07:
Hello 
squishykid thenks to start this topic.
I try to sniff inverter connection. Please see the atehment:
https://iv.pl/images/97211814924075239263.png

192.168.7.15 its Inverter IP
192.168.7.12 its Home assistant with runing solax sensor...

squishykid, 2019-07-20:
@kprzeb I can't see the image any more. Could you upload the image by dragging and dropping it onto github?

kprzeb, 2019-07-21:
Oh sorry this screen is:
[image]

squishykid, 2019-07-22:
That looks like a good start. Now I just need to know the name of the endpoint and get a sample of the response. A wireshark file would be perfect.

Can you also have a look at the data field and see what the meaning of the fields is?

kprzeb, 2019-07-22:
OK wireshark file is in attachment 

[solax.zip](https://github.com/squishykid/solax/files/3418915/solax.zip)

squishykid, 2019-07-26:
@kprzeb the wireshark does not seem to include a sample of calling an endpoint on the inverter. I can see home-assistant trying to connect to the inverter (and failing to get data back). I can also see the inverter sending data off to a public IP address (47.254.152.24 - probably solax monitoring).

Have you found an endpoint on the inverter which returns the inverter data? We basically need to do the same work as in this thread: https://community.home-assistant.io/t/solax-solar-inverter-setup-guide/48008

rest77, 2019-07-26:
Every 5 min the inverter sents the data to the solax cloud. 47.254.152.24 is the solax server for the data collection. The pocket wifi used to connect the inverter always keeps its own wifi on. You can connect to the inverters Wifi an use the local connection in the app. I tried to sniff this network but did noch get any usable data. The app in lokal should be polling the data from the inverter (I thought). I'll keep on trying an send updates as soon I find something.

kprzeb, 2019-08-01:
Hello 
I try to sniff local connection to inverter (App solax cloud local to inverter WIFI IP 5.8.8.8 ) 
This is the communication application with the inverter:
[image]

I can try to decode data on a sunny day, but I'm not happy that the data can not be accessed from the home WIFI network. It will be necessary to have another one WIFI card in client mode ?

rest77, 2019-08-02:
I run hass.io on the ethernet port of the Pi. Maybe it is possible to use the wifi of the Pi to connect to the inverters wifi?

squishykid, 2019-08-02:
@kprzeb this is exactly what we need. Thank you! This is enough to get started, but if you could send me the request shown in the image as a wireshark capture I will be able to build the feature without further issues.

If you don’t want to upload the file publicly, you can send it to my keybase too: robinwohlers

squishykid, 2019-08-02:
@rest77 this is probably possible, but I’m not sure how. Give it a go and let us know if you have success.

kprzeb, 2019-08-03:
That looks good. Login is not necessary. I just connect to inverter WiFi (SSID Wifi_SW8EUXXXXX) and send http post request "http://5.8.8.8/?optType=RealTimeData" and get data in response. Data format is the same in previous inverter version:
1: PV1 Current
2: PV2 Current
3: PV1 Voltage
4: PV2 Voltage
5: Grid Current
6: Grid Voltage
7: Grid Power
8: Inner Temp
9: Solar Today
10: Solar Total
11: Feed In Power
12: PV1 Power
13: PV2 Power
14: Battery Voltage
15: Battery Current
16: Battery Power
17: Battery Temp
18: ???
19: Battery Capacity
20: Solar Total 2
.
.
.
.

Example data for me:

rest77, 2019-08-03:
> @rest77 this is probably possible, but I’m not sure how. Give it a go and let us know if you have success.

I tried, but my Pi (3+) just sees 5GHz networks. I have some trouble with Bluetooth as well. Is my hardware broken or does anyone have a hint for me?

kprzeb, 2019-08-03:
** @rest77 ** Maybe the distance from rpi to inverter is to big ? Can you find network witch telephone?

rest77, 2019-08-05:
No, the signal is fine. There is something wrong with the Bluetooth as well. Both run on 2.4 GHZ. Thats why I suspect a problem on the hardware.

rest77, 2019-08-11:
I fixed some wifi settings on my 2.4 wifi in the router. Now my pi is connected to the inverters wifi. 
If I use: http://admin:admin@5.8.8.8/?optType=ReadRealTimeData or http://admin:admin@5.8.8.8/api/realTimeData.html I get page not found. Witch URL should I use to try?

squishykid, 2019-08-18:
@kprzeb I'm working on adding this to the library at the moment. There's a couple of other changes which need to be made too so this works nicely and they are causing delays.

Would it be possible to get the HTTP request you made in the screenshot as a Wireshark capture? This would help me immensely with testing the new inverter support.

kprzeb, 2019-08-22:
Hey
In attachment you have wireshark pac file
[solax.zip](https://github.com/squishykid/solax/files/3531731/solax.zip)

squishykid, 2019-08-24:
@kprzeb could you please test the code on the branch of #9 with your inverter.

I'm not sure how much help you will need, so I will just provide as much information as possible:
* Download a zip of the branch (not master) [ zip link here](https://github.com/squishykid/solax/archive/add-x3-and-multiple-inverter-support.zip)
* You might need to install the dependencies: `pip install aiohttp voluptuous`
* Create a file called `test.py` in the root and paste the contents from below
* Run `python test.py` and copy the output into a comment here.

It should print something like this, but the fields will be a bit different (this is for an XHybrid):
```
RealTimeResponse(data={'PV1 Current': 0.0, 'PV2 Current': 0.0, 'PV1 Voltage': 0.0, 'PV2 Voltage': 0.0, 'Output Current': 0.8, 'Network Voltage': 248.3, 'Power Now': 85.0, 'Inverter Temperature': 27.0, "Today's Energy": 17.8, 'Total Energy': 16256.5, 'Exported Power': -12.0, 'PV1 Power': 0.0, 'PV2 Power': 0.0, 'Battery Voltage': 55.74, 'Battery Current': -2.58, 'Battery Power': -145.0, 'Battery Temperature': 14.0, 'Battery Remaining Capacity': 91.0, "Month's Energy": 2058.3, 'Grid Frequency': 50.0, 'EPS Voltage': 0.0, 'EPS Current': 0.0, 'EPS Power': 0.0, 'EPS Frequency': 0.0}, serial_number='XXXXYYYY', version='Solax_SI_CH_2nd_20160912_DE02', type='AL_SE', status=2)
```

Code:
```
import solax
import asyncio

async def work():
    r = await solax.real_time_api('5.8.8.8')
    return await r.get_data()

loop = asyncio.new_event_loop()
asyncio.set_event_loop(loop)
data = loop.run_until_complete(work())
print(data)
```

kprzeb, 2019-08-25:
Hello
I have this error:
[code]
root@odroidc2:~/solax-add-x3-and-multiple-inverter-support# python3 test.py
Traceback (most recent call last):
  File "test.py", line 10, in <module>
    data = loop.run_until_complete(work())
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "test.py", line 6, in work
    return await r.get_data()
  File "/root/solax-add-x3-and-multiple-inverter-support/solax/__init__.py", line 107, in get_data
    return parse_solax_real_time_response(resp)
  File "/root/solax-add-x3-and-multiple-inverter-support/solax/__init__.py", line 85, in parse_solax_real_time_response
    version=response['version'],
KeyError: 'version'
[/code]

Am I doing something wrong?
curl works on this host:
[code]
root@odroidc2:~/solax-add-x3-and-multiple-inverter-support# curl -X POST http://5.8.8.8/?optType=ReadRealTimeData
{"type":"X3-MIC","SN":"XXXXXX","ver":"2.033.20","Data":[0.0,0.0,0.0,0.0,0.0,0.0,0,0,0.0,0.0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0.00,0.00,0,0,0,0.0,0.0,0.0,0.0,0.00,0.00,0.00,0,0,0,0,0,0,0,0.00,0,8,0,0,0.00,0,8,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[0.000,7,"X3-MIC","XXXXXXX",1,0.00,0.00,1.09,1.02]}root@odroidc2:~/solax-add-x3-and-multiple-inverter-support#
[/code]

squishykid, 2019-08-25:
No, this is an issue with my code. I’ll fix this now.

rest77, 2019-08-29:
Hi! I'm back after some trouble with my old pi. Now I'm running on a pi 4. If I try "curl -X POST http://5.8.8.8/?optType=ReadRealTimeData" i get an answer whitch looks right. How can I support you? Would be rally nice to have this integration working.

squishykid, 2019-08-29:
I need to get some time to fix the bug @kprzeb found above. If you like you can try and replicate the bug. Then you will be ready to test the fix when it becomes available.

In the mean time, could you provide the output from that curl command please? Just to double-check your inverter returns the same data.

rest77, 2019-08-30:
I get: {"type":"X3-Hybiyd-G3","SN":"SWR*******","ver":"2.33.7","Data":[0.4,0.6,641.6,600.0,-0.7,231.0,114,33,2.0,3021.3,-35,298,386,428.00,0.80,368,20.0,21.3,0.0,288.3,0.00,79.00,0,0,0,0.0,0.0,0.00,0.00,0,0,0,0.0,0.0,0.00,0.00,0,0,0,0.0,0.0,2236.48,345.85,2,28,84,0.7,0.8,228.3,228.9,50.06,50.06,50.06,0.0,0.0,0.000000,0.00,0,0,0,0.00,0,8,0,0,0.00,0,8,2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,1,0,0,0.00,0.00,1],"Information":[10.000,5,"X3-Hybiyd-G3","H3PE10E*******",1,3.31,0.00,3.27,1.01],"battery":{"brand":"83","masterVer":"9.001","slaveNum":"8","slaveVer":[1.10,1.10,1.10,1.10,1.10,1.10,1.10,1.10],"slaveType":[1,1,1,1,1,1,1,1]}}

Could you tell me where to put the files from your zip? I will try to help you.
/Reik

squishykid, 2019-09-07:
@kprzeb I think I have fixed the issue you found, could you try again please?

rest77, 2019-09-07:
@squishykid Could you tell me where to put the files from your zip? Then I will try to help you too.

/Reik

kprzeb, 2019-09-07:
Good for you! 
Its working I received a reply:
`RealTimeResponse(data={'PV1 Current': 0.0, 'PV2 Current': 0.0, 'PV1 Voltage': 0.0, 'PV2 Voltage': 0.0, 'Output Current': 0.0, 'Network Voltage': 0.0, 'Power Now': 0.0, 'Inverter Temperature': 0.0, "Today's Energy": 0.0, 'Total Energy': 0.0, 'Exported Power': 0.0, 'PV1 Power': 0.0, 'PV2 Power': 0.0, 'Battery Voltage': 0.0, 'Battery Current': 0.0, 'Battery Power': 0.0, 'Battery Temperature': 0.0, 'Battery Remaining Capacity': 0.0, "Month's Energy": 0.0, 'Grid Frequency': 0.0, 'EPS Voltage': 0.0, 'EPS Current': 0.0, 'EPS Power': 0.0, 'EPS Frequency': 0.0}, serial_number='SYYYYYYYY', version='2.033.20', type='X3-MIC')`

It is night at my place so data is zero. I will test again tomorrow morning and check the correctness of the fields. How to integrate it with HA ?
Thanks.

squishykid, 2019-09-08:
@kprzeb Awesome! Let me know how it goes with sunlight. If it checks out (can you have a look that all the sensor values are correct too please) From here I just need to upgrade the solax integration in the home-assistant project.

@rest77 & @kprzeb I'm thinking there should be a Discord server for talking about tech support and these kinds of things, would be a lot faster than just replying to github issues! I don't think we can use the existing home-assistant Discord server. Watch this space....

kprzeb, 2019-09-08:
I test it :)
RealTimeResponse(data={'PV1 Current': 6.5, 'PV2 Current': 6.4, 'PV1 Voltage': 438.2, 'PV2 Voltage': 474.6, 'Output Current': 8.2, 'Network Voltage': 227.2, 'Power Now': 5636.0, 'Inverter Temperature': 31.0, "Today's Energy": 0.5, 'Total Energy': 35.3, 'Exported Power': 0.0, 'PV1 Power': 2837.0, 'PV2 Power': 3013.0, 'Battery Voltage': 0.0, 'Battery Current': 0.0, 'Battery Power': 0.0, 'Battery Temperature': 0.0, 'Battery Remaining Capacity': 0.0, "Month's Energy": 0.0, 'Grid Frequency': 50.0, 'EPS Voltage': 0.0, 'EPS Current': 0.0, 'EPS Power': 0.0, 'EPS Frequency': 0.0}, serial_number='SWXXXX', version='2.033.20', type='X3-MIC')

It's almost good :)
What is 'Output Current' ? this is the output by AC side? This inverter have 3 phase output so the current should be 3x ~8A see picture
[image]
[image]

I don't know why I don't see these values from solax answer
curl -X POST http://5.8.8.8/?optType=ReadRealTimeData
{"type":"X3-MIC","SN":"SWXXXXX","ver":"2.033.20","Data":[6.4,6.4,443.1,474.3,8.2,227.0,5619,31,0.5,35.4,0,2834,3011,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0.00,0.00,1833,1867,1919,8.1,8.3,234.4,236.2,49.99,49.99,50.00,0,0,0,0,0,0,0,0.00,0,8,0,0,0.00,0,8,2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[8.000,7,"X3-MIC","MCXXXXXXXXX",1,1.10,1.02,1.09,1.02]}

Otherwise its looks very good :)

rest77, 2019-09-10:
The discord server is not working. I tried when the message came an now again but the invite has expired.

squishykid, 2019-09-10:
@rest77 try this one https://discord.gg/PtvdBth

squishykid, 2019-09-18:
So I have merged a bunch of other changes in #9, moved the X3 support into #11.

closed 2019-09-19 by squishykid

ricard0g0mes, 2019-11-13:
so, how can i use this in my home assistant (hassio) system?
is it possible?

PR #9 (2019-09-18): Add multiple inverter support
closes #5
PR #11 (2019-09-19): Add x3 support
This closes #5 much more cleanly