squishykid/solax #7: Inverter offline - no sensor

matthewjporter, 2019-07-24:
Hi, Not sure what to title as, but here is my issue

Now that I can add both my inverters in HA (Great Job on that) and they display as sensor.solax_inverterserial_sensor_name I am seing both my X1 hybrid and my X1 boost and all is great and wonderfull until the sun goes down and I reboot HA after sunset

If I reboot HA when my X1 boost is offline (it goes dead when not generating solar) I seem to be loosing all the sensors for it and never come back at all even when it starts generating again. The X1 hybrid is fine as its is always connected to AC/DC as per its design

I think the answer would be to make the sensor a permanent sensor so even without an update it will just display 0, nothing or the last reading (last reading is not a nice way) as the value and then when the inverter comes back online and is polled again it can update with the now new data being received

Cheers
Matthew

squishykid, 2019-07-24:
We will need to figure out which of the three (0, n/a or last value) to apply. On the other hand I’m not sure this is a decision I can make for everyone. I’m leaning towards another option- we make each inverter only one sensor and then combine all the other values as attributes. Many other components use this pattern. Then it will be up to the user to create template sensors for the values they are interested in, where they can also decide if they want 0, n/a, or last value.

matthewjporter, 2019-07-25:
Cheers, 

I know that the Solax Portal does last value as my inverter was saying it was producing 8w all night. Not sure you can get solar energy from the moon :)

Personally I think a zero value is better which is what I currently get with node-red sensors

If you have a development branch, let me know and will setup a custom_component folder to test for you

squishykid, 2019-07-26:
Yes, but on the other hand, network voltage shouldn't be 0 overnight. It makes more sense to leave that at the last measured value.

squishykid, 2020-06-10:
Re-visiting this now, I think that it should be possible to modify the code in home-assistant to mark the inverter as 'unavailable' in this scenario.

closed 2021-08-22 by squishykid