squishykid/solax #13: Tested with X1

spectroman, 2019-09-20:
Hi,

Just want to give you notice that I am testing this with the X1 and it works just fine, it is just a bit more freaky to reach him.

```
./solax-test.py 
InverterResponse(data={'Network Voltage Phase 1': 237.2, 'EPS Power': 0.0, "Today's Energy": 9.8, 'PV2 Current': 0.0, 'AC Power': 1183.0, 'Total Energy': 14.8, 'Total Consumption': 0.0, 'Grid Frequency Phase 1': 49.96, 'Output Current Phase 2': 0.0, 'Grid Frequency Phase 3': 0.0, 'Battery Voltage': 0.0, 'EPS Frequency': 0.0, 'Battery Current': 0.0, 'EPS Current': 0.0, 'Exported Power': 0.0, 'Battery Temperature': 0.0, 'PV1 Voltage': 274.3, 'PV1 Power': 1285.0, 'Output Current Phase 3': 0.0, 'PV2 Voltage': 0.0, 'Power Now Phase 3': 0.0, 'Network Voltage Phase 3': 0.0, 'Power Now Phase 2': 0.0, 'Network Voltage Phase 2': 0.0, 'Grid Frequency Phase 2': 0.0, 'PV2 Power': 0.0, 'Battery Power': 0.0, 'EPS Voltage': 0.0, 'Power Now Phase 1': 0.0, 'PV1 Current': 4.6, 'Battery Remaining Capacity': 0.0, 'Total Feed-in Energy': 0.0, 'Inverter Temperature': 40.0, 'Output Current Phase 1': 5.0}, serial_number='notagoodidea', version='2.033.20', type='X1-Boost-Air-Mini')
```

To talk to this one, using the "Pocket WIFI" version 2.0, the user need to connect into the Wifi generated by the dongle and query `http://5.8.8.8` -- the IP Address from the regular LAN won't accept requests, the address via LAN is used only to send information out to solarcloud.com, .. although the port 80 is open there, it seems to drop all the requests not coming from the network 5.8.8.x -- I also noticed that only one device can be connected to his local wifi network... as such... I connected my RPI via cable on my regular network and used the his embedded wifi to reach the Solax pocket wifi... 

It became even better, because then my RPI is locking access to the Solax WIFI and only me can exchange data with it...

The important part here, is that ur stuff also work on this other version. Thank you man! Now lets bind it with Openhab!

squishykid, 2019-09-21:
Woohoo! That's awesome, great to hear. Shame it's a bit more difficult to set up though. You might be able to expose the inverter on your local network with a HTTP proxy on the PI.

It seems like the library is detecting your X1 inverter as an X3 model - could you try making a POST request to `http://5.8.8.8/?optType=ReadRealTimeData` and posting the reply?

`curl -X POST http://5.8.8.8/?optType=ReadRealTimeData` should do it.

From what I can tell, the X1 is only a single phase inverter- having all the 'Grid Frequency Phase 3' etc values in the reply isn't correct. The sensor map is defined here: https://github.com/squishykid/solax/blob/0.2.2/solax/inverter.py#L176

We should add another inverter to the project to make it nice and clean.

spectroman, 2019-09-21:
The Curl command was my starting point here. As they seem to ask for a authentication, which is completely bogus in the end. When I got the Curl working, thereafter I tested this module... nonetheless, having a more tailored comm is best for sure.

Here's the JSON response:

```
{"type":"X1-Boost-Air-Mini","SN":"Serialfromwifidongle","ver":"2.033.20","Data":[5.1,0.0,259.6,0.0,5.3,235.0,1244,41,1.7,18.1,0,1348,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0.00,0.00,0,0,0,0,0,0,0,49.97,0,0,0,0,0,0,0,0,0,0.00,0,0,0,0,0.00,0,0,2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[2.500,4,"X1-Boost-Air-Mini","serialfrominverter",1,3.30,2.09,1.16,0.00]}
```

From what I could see, the data now matches with the information on the solaxcloud app, seems ok, but I haven't check each value in detail.

spectroman, 2019-09-21:
if u pvt me I have some other info u might be interested as well.

fverschure, 2019-09-27:
I have a solax X1 and a X3 both with pocket lan cards, so I can do some testing as I also want to send the data to my HA platform.

wills106, 2019-10-01:
@fverschure Does your Pocket Lan cards expose port 502?
Have a look at [Issue 14](https://github.com/squishykid/solax/issues/14) as I am connecting to my inverter through Modbus over TCP

fverschure, 2019-10-01:
@wills106 It doesnt look like 502 is exposed.  I tried running a portscan and both cards do not seem to have any ports exposed below 1024...

fverschure, 2019-10-06:
@wills106: did you do anything to make the solax expose port 502?

wills106, 2019-10-06:
No, but I am using the built in LAN and not a PocketLAN.

Have you tried copying my setup on my GitHub?
Just so you know when I did a normal scan it did not show port 502, only when I did an Intense scan did it show up.

Timgerr321, 2019-12-14:
@fverschure Did you manage to get it working? i have the same setup with two inverters but no ports are open..

fverschure, 2019-12-27:
Hi, no unfortunately not. I am about to give up on this and work out a RS485 setup for reading out both of my inverters. It it insane that Solax disables local control...

Timgerr321, 2019-12-27:
I managed to get around it by using an emulator on my PC and monitoring the network traffic. While monitoring I found the url's that the API uses for the mobile app. So in hassio I make a RESTapi request and get all the data back. Underneath is a photo while it's almost dark. 

[image]

fverschure, 2019-12-27:
Could you document this with some some detail? It may help some others as well.
Does the inverter expose a api after all but you need to know the URI to use is or do you pull the data back from the cloud api?

I will setup some monitoring ports on my switch to sniff devices.

Timgerr321, 2019-12-27:
If I have the time for it I will do some documentation. I tried a lot of things on my local network but nothing seemed to work, the inverters kept blocking me out. So I am indeed pulling the data from the solax servers just like the app does. 
A nice advantage over the local connection is that it doesn't shut down overnight so hassio doesn't spit errors all over the place.

squishykid, 2019-12-28:
@fverschure In my experience some inverters need to be connected via Ethernet to access modbus/port 502. This is the case with the X hybrid (SK-TL5000E) I have access to. If you could draw a diagram with any inverters, ethernet cables, modems, ha instance and wifi cards, it will be easier to assist you.

I am planning to add modbus support to the python library, as it seems that many inverters which rely on the pocket wifi card are very difficult to interface with over rest. See issue #5 and the references to '5.8.8.8' for example.

thomascys, 2019-12-29:
You can indeed obtain all the data via their api:
base url: http://47.254.152.24:6080
login endpoint: /proxy/login/login?userName=<your-username>&password=<your-password>
data endpoint: /proxy/inverter/getDailyInfo?inverterSn=<your-inverter-serial-number>&wifiSn=<your-wifi-serial-number>&tokenId=<token-from-the-login-endpoint>&today=<Y-m-d>

I created a custom component and will open source it soon after some code refactoring.

fverschure, 2019-12-30:
> @fverschure In my experience some inverters need to be connected via Ethernet to access modbus/port 502. This is the case with the X hybrid (SK-TL5000E) I have access to. If you could draw a diagram [...]
I have included a diagram of the relavant part of the network. 
I have two inverters:
- X1-Mini
- X3 Mic pro
both of them are connected with a pocketlan 2.0 lan adaptor and plugged into a switch going down to my main switch which is plugged into my ISP router. 
The lan (where Home assistant and the inverters are connected to) is a single (class C /24) segment

I think that the pocket-lan adapters work differently than the wifi adapters but they might be better to sniff the data flowing. 

> I am planning to add modbus support to the python library, as it seems that many inverters which rely on the pocket wifi card are very difficult to interface with over rest. See issue #5 and the references to '5.8.8.8' for example.

[image]

fverschure, 2019-12-30:
> You can indeed obtain all the data via their api:
> base url: http://47.254.152.24:6080
> login endpoint: /proxy/login/login?userName=&password=
> data endpoint: /proxy/inverter/getDailyInfo?inverterSn=&wifiSn=&tokenId=&today=

I saw someone creating a nodered flow based on this api pushing the data back to MQTT into home assistant. I could not get it to work either.

squishykid, 2020-01-01:
@thomascys @fverschure It would be cool to add support for the API at http://47.254.152.24:6080 to the home-assistant integration, or maybe as a new integration. I think this is out of scope for this library though- I would like to focus on local inverter data access. For example, the home-assistant solaredge integration has two versions, one for local access and another using the internet API.

thomascys, 2020-01-02:
@fverschure @squishykid I hacked custom component together: https://github.com/thomascys/solaxcloud
It still needs work on the part where the credentials should be configurable instead of hard-coded. Maybe anyone can point me to a good example?

BasTijs, 2020-02-03:
@thomascys  thanks for the scripts, it worked for a while but now my inverter seems to be offline (are they blocking me because of too many requests?)

@squishykid  I managed to get my X3 locally connected by having my rpi connected by cable to the network and by using the build-in wifi to connect to the pocket wifi 2.0 module. For those that also want to try this, take the following steps in HASSIO:
1) install the "SSH & Web Terminal" from Frenck, with protection mode disabled
2) you will now have a terminal with commands to enable wifi (normal SSH server addon does not allow the use of nmcli)
3) command "nmcli radio" (check if wifi is enabled)
4) command "nmcli device" (check if wifi is connected)
5) command "nmcli device wifi rescan" 
6) command "nmcli device wifi list" (look for your inverter Wifi_YOURSERIAL)
7) command "nmcli device wifi connect Wifi_YOURSERIAL password admin"
8) add a solax sensor with ip 5.8.8.8 in your hassio config file
sensor:
  - platform: solax
    ip_address: 5.8.8.8

thomascys, 2020-02-04:
@BasTijs Might be the case that you get blocked. They're API is pretty slow.

fverschure, 2020-02-09:
> @thomascys thanks for the scripts, it worked for a while but now my inverter seems to be offline (are they blocking me because of too many requests?)
>
> @squishykid I managed to get my X3 locally [...]

Thanks @BasTijs : As I have two inverters I dont know if this is an option.
I tried connecting one but it does not broadcast any ssid. Do I need to disconnect it from the cloud first?
I just ordered two rs485 to serial converters. If those arrive I will try this route as I do not suspect that they will close the modbus.

BasTijs, 2020-02-09:
@fverschure I dont think it will work with the pocketLAN adaptors, I also have one because I though the LAN version would give me direct access but I didnt see any open ports. If your rs485 doesnt work you could try 2x wifi and use the rpi build-in wifi and a dongle, or find some smart way to connect and read them one by one.

fverschure, 2020-02-10:
@BasTijs : You are right. No open ports. 
A trace hoever shows that it uses MQTT to send out packages to the cloud
Not sure yet what all the items are used for yet but the 11.1 is my daily yield for today in KwH.
Now if i only would be able to send this data to my own mqtt server... 

src: xxxxx
dst: 47.254.152.24
**topic:** loc/SPYQSBJYZQ
**message:** {"type":"X3-MIC","SN":"xxxxxxxxxx","ver":"2.007.20","Data":[0.0,0.0,0.0,0.0,0.0,0.0,0,0,11.1,3566.8,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0.00,0.00,0,0,0,0.0,0.0,0.0,0.0,0.00,0.00,0.00,0,0,0,0,0,0,0,0.00,0,8,0,0,0.00,0,8,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[6.000,7,"X3-MIC","xxxxxxxxxxxx",2,1.10,1.02,1.09,1.02]}

jk-5, 2020-03-05:
I have a solax X1 with ethernet dongle. The mqtt stuff looked very interesting so i tried hacking it. The inverter does a DNS lookup to mqtt002.solaxcloud.com when it starts up, then it connects to it with destination port 2901. I made a dns override in my network to redirect it to a local mosquitto instance. There is, however, a protocol violation in the solax mqtt implementation, and because of that mosquitto closes the connection. I could write a simple translation proxy to which the inverter connects that allows the protocol violation, which then connects to mosquitto itself. That will be a very ugly setup but might work.

I also tried setting up static routes for 5.8.8.8 or 11.11.11.0/24, but that gave no luck with the wired dongle

Kratoklastes, 2020-03-07:
I can confirm that the 5.8.8.8 method works. My preferred approach -

```
# [import block]
# [SNIP]

# config for requests
url = "http://5.8.8.8/"
myObj = {'optType': 'ReadRealTimeData'}
# [SNIP]

# Get list of all available SSIDs
r = subprocess.run(["netsh", "wlan", "show", "network"], capture_output=True, text=True).stdout
ls = r.split("\n")
ssids = [v.strip() for k,v in (p.split(':') for p in ls if 'SSID' in p)]
#[SNIP]

# if there are SSIDs named "Solax[something]", connect to each of them in turn, fetch data
tgtList = [s for s in ssids if 'Solax' in s]

for tt in tgtList:
    # connect to the SSID
    cmd = "netsh wlan connect name={0} ssid={0}".format(tt)
    k = subprocess.run(cmd, capture_output=True, text=True).stdout
    # Wait until it's actually connected, not when Windows says it's connected
    # [SNIP]
    # Get data
    x = requests.post(url, data=myObj)
    v = json.loads(x.text)
    # parse v and insert into a PostgreSQL data table
```
The primary annoyance with this method is the requirement to connect to the inverter SSIDs in turn, to retrieve each one's data in turn. (We have 2 5kW X1 inverters). 

Windows is well-known for reporting back connection a few seconds before it's actually connected, so there has to be wasted time checking if you're actually connected each time. 

The 'connect-check-fetch-connect-check-fetch' for the 2 inverters means that the script takes 11 seconds - which is woeful and a good reason to detest Solax for crippling Wifi so it can force people to their insecure garbage app/data-gatherer.

When the script is connecting to other SSIDs, the connection to the home wifi must be released. As a result I only run the script every 15 minutes intraday (from ~7:30a.m. to 7:30p.m.) - as a Scheduled Task.

My guess is that there are other `optType` values besides `ReadRealTimeData` (e.g., I bet there's something for history data, as there was for the API before it was deliberately rendered useless) but I have had no luck identifying them. (My main 'beat' is quant finance, not device sniffing!)

I am not a vindictive chap, but anyone who deliberately disables a data API deserves to contract coronavirus (don't worry, it's basically harmless if you're relatively healthy).

Selfie21, 2020-03-09:
Thanks for all the replies here. This whole Github Page helped me a lot sad what they did to the API seemed a lot easier with the old Inverters. I did something really similiar to @Kratoklastes now I'm gonna turn it into an Alexa Skill will see how that goes

Selfie21, 2020-03-11:
I added a way to get Information from the Solax Cloud Portal Remotely. It's pretty hacky and shouldn't be used too often, since it significantly slows down if you try to often, but if anyone needs it,it's the SolaxCrawler: https://github.com/Selfie21/SolaxCloudApi

marvandorth, 2020-03-27:
I'm desperate for some help.
I've got a Solax X3 inverter and I'm trying almost everything to get the data into home assistant or into grafana.
I've tried the Node Red approach here: https://community.home-assistant.io/t/solax-solar-inverter-setup-guide/48008/66
Thought it worked but gave me some strange Chinese error in the debug.
I've tried https://github.com/thomascys/solaxcloud

But nothing seems to work.

fverschure, 2020-03-27:
@marvandorth : I have done the same. 
With a pocketlan adapter you are basically screwed.  
nodered flows did not work for me either. Also tried RS485 route, but so far I did not get it to work yet. May be the rs485 port is not active when you have a active pocket-lan but I would expect some data to flow out of the modbus meter.  Documentation is not very detailed either.

Best option would be to hack to data upload and transform it in such a way that it is a valid mqtt package so you could send it to your local mqtt server. If you would do that it would also break the connection with solax cloud which is not ideal either.

I also contacted solax multiple times (go a contact from the head-office) but that contact is not responding to my email. 

I was exploring whether Solax could consider to allow to duplicate the package to a local mqtt server as well to leverage the data locally, but it seems that they are not very interested to talk to their customers...

wills106, 2020-03-28:
@marvandorth 
Have you tried my ModBus over TCP method?
It's [Issue 14](https://github.com/squishykid/solax/issues/14) on here, or have a look at my GitHub https://github.com/wills106/homeassistant-config/tree/master/packages you will want to use solax_hybrid_g3.yaml As that has the extra X3 (3 Phase) Values. I have not tested them, so please comment if the formatting needs correcting.

@fverschure 
Another user has mentioned that when they are using the built in LAN port their Pocket WiFi gets disabled.
Might be worth removing your PocketLAN while trying to use the RS485 port. It seems like the SolaX inverter might only allow communication from one method at a time.

@ All
Just so you all know I have working battery control now on my ModBus Method.
With my method you can control:
Battery Charge Rate
Battery Discharge Rate
Min % Battery Discharges to
Alter Between, SelfUse, BackUp & ForceTime

ForceTime allows you to charge your battery from the Grid.

marvandorth, 2020-03-28:
@wills106 
I assume the ModBus over TCP method needs a wired connection?
As of this moment my X3 inverter is connected in my Wireless Network via the Pocket Wifi 2.0 USB adapter.

@fverschure 

> Best option would be to hack to data upload and transform it in such a way that it is a valid mqtt package so you could send it to your local mqtt server. If you would do that it would also break the connection with solax cloud which is not ideal either.

I personally wouldn't mind breaking the connection with the Solax Cloud. I think I can make better graphs via Grafana. But I've no idea where to start!

@spectroman 
Did the RPi thing worked out for you?
There's a brief description at the bottom of the page on the Home Assistant Integrations page about this one.
https://www.home-assistant.io/integrations/solax/

wills106, 2020-03-28:
@marvandorth 
You will either need to connect to the Ethernet port / RS485 port depending on model.

This way it's totally local and not reliant on the cloud.

marvandorth, 2020-03-28:
@fverschure 
The strange thing is.
When I try to the the HTTP Post request to retrieve a tokenId:
47.254.152.24:6080/proxy/login/login?userName=*****&password=*****&userType=5
as described here:
https://community.home-assistant.io/t/solax-solar-inverter-setup-guide/48008/30?

I receive a chinese message saying my credentials are invalid but this isn't the case...

spectroman, 2020-03-28:
@spectroman
Did the RPi thing worked out for you?
There's a brief description at the bottom of the page on the Home Assistant Integrations page about this one.
https://www.home-assistant.io/integrations/solax/

For me the problem is solved and it is working flawlessly.
I have also integrated with Zabbix and OpenHab

Julio C. Hegedus v.45.0
---------------------------------------------------------------------
Novell Netware & Linux Specialist
https://juliochegedus.com
ICQ: 3614823
Skype: brclimber
CAP #.: 928

marvandorth, 2020-03-28:
@spectroman 
Can you perhaps explain to me how you pulled this off?

squishykid, 2020-03-28:
@marvandorth
The method used by @spectroman requires you to run the Wifi adapter on the inverter as the access point and then connect as device (maybe raspberry pi) to the inverter.

You will need to manually set the IP Address of the raspberry pi to 5.8.8.9 and then you can query the HTTP api on the inverter at 5.8.8.8

You read more about how another member of the community did this over on issue #5

squishykid, 2020-03-28:
I would like to spend some time documenting a step-by step debug, but have not had the time to do so yet.

marvandorth, 2020-03-28:
@spectroman 
I've readied up a Pi with WiFi and Ethernet connection. Installed Debian on it.
Connected the WiFi from the RPi to the WiFi of the Pocket WiFi 2.0 inserted in the inverter.
Connected the Ethernet to my home network.
Installed NGINX on the RPi and tried the example config as shown on:
https://www.home-assistant.io/integrations/solax/

Unfortunately that doesn't work.

Do you have a working Nginx Reverse Proxy and would you mind sharing it?

spectroman, 2020-03-28:
@marvandorth not sure why you want to use NGINX, as u can see, if you just have the IP on the same network 5.8.8.x, it is done, and u can use the script on this repo. Done. You don't need reverse proxy or anything like that, it is just an over complication.

Once you have the address it should work with curl like this:

` curl -X POST http://5.8.8.8/?optType=ReadRealTimeData
{"type":"X1-Boost-Air-Mini","SN":"SERIALNUMBER","ver":"2.033.20","Data":[6.7,0.0,284.0,0.0,7.4,238.9,1762,44,9.0,570.7,0,1908,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0.00,0.00,0,0,0,0,0,0,0,49.97,0,0,0,0,0,0,0,0,0,0.00,0,0,0,0,0.00,0,0,2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[2.500,4,"X1-Boost-Air-Mini","XSERIALNUMBER",1,3.30,2.09,1.16,0.00]}`

marvandorth, 2020-03-28:
> @marvandorth not sure why you want to use NGINX, as u can see, if you just have the IP on the same network 5.8.8.x, it is done, and u can use the script on this repo. Done. You don't need reverse [...]

That's the point, my Raspberry Pi has a wired connection with ip: 192.168.x.x
The Wireless connection received IP address: 5.8.8.11.

The HTTP Post request you mentioned only works on the Pi itself, but I'm not sure how to expose the data on my normal WiFi network.

wills106, 2020-03-28:
@marvandorth 
Which version X3 do you own?

marvandorth, 2020-03-28:
> @marvandorth
> Which version X3 do you own?

I have the X3-MIC / PRO.

wills106, 2020-03-28:
@marvandorth 
I take it your Inverter doesn't have an ethernet Port then? Unless I am looking at an older Manual.
Do you have anything plugged into your RS485 port?

If possible I would seriously consider running a cable to your Inverter if practical.

marvandorth, 2020-03-28:
> @marvandorth
> I take it your Inverter doesn't have an ethernet Port then? Unless I am looking at an older Manual.
> Do you have anything plugged into your RS485 port?
> 
> If possible I would seriously consider running a cable to your Inverter if practical.

Nope no Ethernet Port.
In order to connect it to a network I had to buy a USB adapter (the Pocket Wifi 2.0 in my case).
There's nothing plugged into my RS485 port.

I'd rather cable every device in my house but in this case that isn't a possibility....

I managed to do: 
`$ curl -X POST http://5.8.8.8/?optType=ReadRealTimeData
`
and received the following data:
`{"type":"X3-MIC","SN":"XXXXXX","ver":"2.033.20","Data":[7.9,7.9,308.7,296.0,6.3,239.3,4479,39,23.1,618.9,0,2415,2332,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0.00,0.00,1475,1480,1524,6.3,6.4,239.7,240.4,50.02,50.02,50.03,0,0,0,0,0,0,0,0.00,0,8,0,0,0.00,0,8,2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[5.000,7,"X3-MIC","XXXXXXXX",1,1.11,1.02,1.10,1.02]}`

I did this on the RPi that's connected wireless to the Inverter WiFi connection AND the Cabled home network. So I think I'm close....

Now I only need a way to access this data from my Home Assistant installation

marvandorth, 2020-03-28:
Ok, got a breakthrough!

I managed to create a HTTP Reverse Proxy with NGINX on the Pi.
The Inverter is now exposed on the 192.168.2.x range (where all other devices are).
Also doing a HTTP post works! 👍 

Now let's get this data into home assistant somehow

fverschure, 2020-03-28:
Any clue how to proceed with A X3 Mic / Pro with a pocketlan?

marvandorth, 2020-03-28:
> Any clue how to proceed with A X3 Mic / Pro with a pocketlan?

With Pocketlan you mean the cabled version of PocketWiFi?
Perhaps you can try to achieve the same as I have.

At the moment the data is being dragged into HomeAssistant so a success i could say!

[image]

fverschure, 2020-03-28:
@marvandorth : I tried to curl 5.8.8.8 with the ethernet adaptor on the same subnet, but it does not work.
I guess the pocketlan adapters are designed differently so I am stuck again.

marvandorth, 2020-03-28:
> @marvandorth : I tried to curl 5.8.8.8 with the ethernet adaptor on the same subnet, but it does not work.
> I guess the pocketlan adapters are designed differently so I am stuck again.

Can you describe your situation?
Can you do the Curl command when your computer is connected to the Pocketlan?

fverschure, 2020-03-28:
@marvandorth: my setup is in some posts above and see below:

[image]

I will directly connect one of my inverters to see if I can do a curl then. I’ll post the results later on here.

BasTijs, 2020-03-28:
> Ok, got a breakthrough!
> 
> I managed to create a HTTP Reverse Proxy with NGINX on the Pi.
> The Inverter is now exposed on the 192.168.2.x range (where all other devices are).
> Also doing a HTTP post works! 
> 
> Now let's get this data into home assistant somehow

@marvandorth cool you got the reverse proxy to work, maybe explain it in more detail for other users to also try this approach?

For me the pocketWifi approach with the wifi from the RPI as I explained above works great, but I only have one inverter. @fverschure You could consider getting 2x pocket wifis (they are cheap, ~30 euro) and maybe a wifi dongle for another wifi connection in the RPI? I dont know if it could handle 2 wifi networks while HA is on the RPI LAN?

sanmagal, 2020-05-01:
@marvandorth a detailed explanation of the reverse proxy to HA would be great.
Thanks

ricard0g0mes, 2020-06-16:
> > Any clue how to proceed with A X3 Mic / Pro with a pocketlan?
> 
> With Pocketlan you mean the cabled version of PocketWiFi?
> Perhaps you can try to achieve the same as I have.
> 
> At the moment the data is being dragged into HomeAssistant so a success i could say!
> 
> [image]

Can you pleeeeasssee tell us how did you do it?
I also have a X1 connected to my wifi and a rp4 with hassio.
How can i do it to read the data to my hassio?

Thank you

marvandorth, 2020-06-16:
Sorry, I accidentally unsubscribed to this thread.
I have Home Assistant running on another device in Docker but I think it doesn't really matter.
I've connected my RPi to my network via LAN and have another (Wireless) connection to my Solax via WiFi USB Dongle.

You should install Nginx on the Pi and make a .conf file in: /etc/nginx/sites-available/_solax.conf_.

[image]

Save it and give the .conf file a u can remember name. I've named mine solax.conf as you can see.
Now do a link with the following command:
`ln -s /etc/nginx/sites-available/solax.conf /etc/nginx/sites-enabled/solax.conf`

Then reload the nginx service with:
`service nginx restart`

If all is configured correctly you should be able to access the Solax PocketLan website via the IP-Address of your RPi.

[image]

If that's the case the only thing you need to accomplish is add the Solax integration to your configuration.yaml in Home Assistant:

[image]

b3luga, 2020-07-01:
> I have a solax X1 with ethernet dongle. The mqtt stuff looked very interesting so i tried hacking it. The inverter does a DNS lookup to mqtt002.solaxcloud.com when it starts up, then it connects to [...]

@jeffreykog Just to let you know that your approach is working for me. I have a Solax X1 with a LAN stick. I redirected all communication to mqtt001/mqtt002.solaxcloud.com to a local mosquitto instance via a local DNS entry. The only thing that I did on top of what you said is to include the line

listener 2901

in my mosquitto.conf. This is needed because the Solax is not using the standard MQTT port 1883.

I made my OpenHab instance an MQTT client of the mosquitto and now the data is feeding the smart home. Only need to make sense of the data now...

Thanks for showing the way!

jeffreykog, 2020-07-01:
@b3luga Interesting! I did add the `listener 2901` statement to let it listen on the custom port. Which mosquitto version are you using? It did raise a protocol violation and closed the connection. Did you do anything for authentication, or are you allowing every connection? If it works for you, i'm sure i am doing something wrong

b3luga, 2020-07-01:
I only installed it this morning on my Pi, so I guess it's the latest and greatest: v3.1. Nothing changed or added to the config except for that listener statement in /etc/mosquitto/conf.d/mosquitto.conf

jeffreykog, 2020-07-02:
@b3luga It still gives an protocol violation for me on a clean mosquitto install. v3.1 is the MQTT protocol version, not the mosquitto version. Can you check the version in the output of `mosquitto -h` for me? I have 1.6.9

b3luga, 2020-07-02:
@jeffreykog This is the first line of output:
`mosquitto version 1.3.4 (build date 2019-10-26 22:21:58+0000)
`
Interesting, that mine is so much older. 

BTW, what do you mean, when you say "protocol violation"? Are you sure you added the listener statement to the right config-file?

[36 more replies]

closed 2021-08-22 by squishykid