squishykid/solax #14: Add Support for Modbus

wills106, 2019-10-01:
I have a X1-Hybrid-5.0-D-E Inverter that has built in LAN, but the API is closed done for local connection. I can connect to the Cloud through this LAN port though.

I do not have a Pocket WiFi device connected to this Inverter.

I discovered port 502 is open on the LAN port. This port is Modbus over TCP.

Hopefully you can connect directly to the RS485 port on Inverters with no built in LAN port, using a USB - RS485 adaptor.

I found the Modbus registers as Hex on this GitHub page:
https://github.com/InfernoEmbedded/PowerScraper

I have converted it into decimal values and use the built in Modbus support in Home Assistant.
Some response values still don't make sense, but I have documented what I have found so far.
https://github.com/wills106/homeassistant-config/blob/master/packages/solax.yaml

Home Assistant thread:
https://community.home-assistant.io/t/solax-inverter-by-modbus-no-pocket-wifi/140143

squishykid, 2019-10-04:
For now I don’t have the resources to look into supporting modbus, either over TCP or Serial. I’m going to close the issue, but if there are others who would like to make a PR to support this then I’m happy to review it.

-

squishykid, 2020-01-01:
@wills106 are you still interested in testing modbus support?

wills106, 2020-01-05:
Yes sure. I have a few more registers for the 3 Phase / X3 models. But they are untested

https://github.com/wills106/homeassistant-config/blob/master/experimental/solax_test.yaml

I would love to be able to control the battery as well. But not sure if there is an issue with the built in Modbus over TCP implementation in HA as I can't seem to be able to write to the registers needed. Perhaps the ModBus library needs updating in HA?

matthewjporter, 2020-03-19:
I have just put my 5000E onto the local LAN port and can connect via modbus if you need any testing

squishykid, 2020-03-20:
@wills106 have you seen documentation to control the battery? Are talking about configuring when the battery supplies power to the house?

wills106, 2020-03-20:
The previous link is incomplete.

The new file https://github.com/wills106/homeassistant-config/blob/master/packages/solax_hybrid_g3.yaml

Allows you to alter the charge rate of the battery, alter the discharge rate. Change the min value the battery discharges to as a percentage.

You can also cycle from Self Use, Back Up and Force time.
With Force Time you can charge the Battery from the Grid.

I have not seen any official documentation. I have based the work from a known Modbus Register and noticing in one of the files on the SolaX Portal that a value matched up, bar with an offset. With knowing the offset I was able to work out some of the other Registers used to control the above mentioned.

The Web Portal server communicates via MQTT to the Inverter, but this is encrypted on port 8883.
The server I observed is based in Germany. Or at least the owner is Germany based.

There is more info: https://community.home-assistant.io/t/solax-inverter-by-modbus-no-pocket-wifi/140143/45

I don't know if this method works on the older models such as the 5000E. But I think I can setup a file for that inverter but I am unable to test it.

matthewjporter, 2020-03-21:
@wills106 
I have the 5000E if you need me to test

wills106, 2020-03-21:
@matthewjporter 
I will try and put together a complete package for you to try tomorrow. As you have found out my GitHub is a bit of a mess!

matthewjporter, 2020-03-21:
@wills106 
No Worries, I am happy to troubshoot, provide feedback and help in anyway I can if in the long run anyone else can benefit

In the UK and company has put a travel ban on me going to do onsite installation so have two weeks spare time at the moment :)

wills106, 2020-03-21:
@matthewjporter 

I guess I am one of the "lucky ones" still allowed to go to work hahaha
Is your system single or 3 Phase? Also what type and how many batteries do you have?

wills106, 2020-03-21:
@matthewjporter are you on the solax-dev chat on discord? https://discord.gg/V5VZKm
Might be easier to chat on there or my thread on Home Assistant https://community.home-assistant.io/t/solax-inverter-by-modbus-no-pocket-wifi/140143

matthewjporter, 2020-03-21:
I am on the discord chat now (Mr P) so we can take up there :)

I have a 2nd Gen (Single Phase) X1 Hybrid with 3 x Pylontech US3000 Batteries. As an added bonus I also have an X1-Air Boost inverter that dumps direct to the grid :(

jhorisberger, 2020-12-14:
Hi
I just got a X3-Hybrid 10.0T installed and i am currently in the process of integrating a custom batterypack (2nd life out of an Electric Vehicle). I had some contact with the Solax support concerning the CAN message protocols to connect the battery and they sent me the full specification for the Modbus over TCP instead, which i am happy to share 

Of course i can also offer to do the needed testing to get the X3-Hybrid on the list of supported Devices. I'm guessing it shoulden't be much of a step up from the X1 hybrid

wills106, 2020-12-15:
I have implemented the main X3 Specific registers in a package format if you want to drop something into Home Assistant straight away. https://github.com/wills106/homeassistant-config/blob/master/packages/solax_x3_hybrid_g3_triplepower.yaml

This is using the old single register reads, which puts loading on the ModBus bus.

In version 2020.12.0 of Home Assistant you can now read multiple ModBus registers in a single transaction. This reduces the loading on the bus, as you are reading in a more efficient manner.

I have started to move over to this method. I have most of the X1 Registers read in this method https://github.com/wills106/homeassistant-config/blob/master/packages/solax_x1_hybrid_g3_group_read.yaml this will work on the X3 but you will be missing the extra phases from the X3.

I have a little bit more work to do on the X1. Then it will be a simple process to add in the extra X3 registers.

dvisser, 2021-03-08:
> Hi
> I just got a X3-Hybrid 10.0T installed and i am currently in the process of integrating a custom batterypack (2nd life out of an Electric Vehicle). I had some contact with the Solax support [...]

can you share the "modbus over TCP" protocol document please?

jhorisberger, 2021-03-16:
[Hybrid X1&X3-G3 ModbusTCP&RTU V3.21- English.pdf](https://github.com/squishykid/solax/files/6147937/Hybrid.X1.X3-G3.ModbusTCP.RTU.V3.21-.English.pdf)

Hope this helps

sommo, 2021-05-07:
hi all,
is there a version of this project for modbus tcp connection?
thank you

deece, 2021-06-03:
@dvisser I'm keen to get the CAN docs, it would be nice to be able to pretend to be a supported battery (so one could hook up a different battery, say, a salvaged Tesla pack)

jhorisberger, 2021-06-03:
As of now i have not been able to get any CAN documentation from Solax themselves and the reply to my last mail makes clear that they don't intend to hand these out anytime soon. 
With the massive help from Araknid over at Second Life Storage we have been able to reverse egineer the Protocol to the point where i have my custom Battery running for the last month without any major issues. The whole process is documented over at: 
https://secondlifestorage.com/index.php?threads/three-phase-hv-hybrid-inverter-solax-x3-hybrid-8-0-solax-triple-power-t58-hv-battery.10747/#post-75108

Hope this is what you are looking for

deece, 2021-06-03:
Great work! Thankyou, I'll file this away in case I can find a big cheap aftermarket pack

deece, 2021-06-18:
FYI, I've added X-series support to PowerScraper now.

cngitty, 2021-06-28:
Great for your efforts getting X3-series (non hybrid) to work!
I wonder if with my plannend setup I can use:
1) RS485 with RS485->USB and your project connected to a Raspi (best case integrated OpenHAB)
2) AND in parallel the meter connection via RS485_METER to an SDM630 (Solax Version).
Anybody running this setup already? Is 1 & 2 working for you in parallel? Do I get via 1) also my self consumption (read via the SDM630)?
Thx for your feedback or other idea how I can get the data into OpenHAB as I would like with my go-e Charger a PV controlled car charging done.

ritschi1988, 2021-09-20:
Hello, my English is not the best, but I hope that can help...

[https://drive.google.com/folderview?id=187fKfQjEiIAEQ2Nbk2jyUBNkm8Gl286z]()

arwooldridge, 2022-08-09:
>     2. AND in parallel the meter connection via RS485_METER to an SDM630 (Solax Version).
>        Anybody running this setup already? Is 1 & 2 working for you in parallel? Do I get via 1) also my self consumption (read via the SDM630)?

I have tried two sdm630 meters in parrallel.  meter1( grid) on add, 1 meter2 (PV power) on add 2
both on 9600 baud no parity 1stop. but only get access to meter1 on the inverter and solax cloud. I have not set up home assistant yet.
with and without terminating 120 ohm resistors at the chain ends

johny-mnemonic, 2023-03-07:
@squishykid @wills106 I guess with the [modbus integration](https://github.com/wills106/homeassistant-solax-modbus) in full swing this issue could be finally closed and this project can focus solely on the local http API. What do you think?

wills106, 2023-03-07:
Yea might as well close this down?

wills106, 2024-06-23:
Closing down as I have a separate custom_component for Inverters over Modbus

closed 2024-06-23 by wills106