squishykid/solax #18: script reports different output

sommo, 2020-04-10:
Hi,
i'm running your solax script v.0.1.2.
when i run it, it reports different data every time:
pi@rpi_hue:~ $ python3 solax
RealTimeResponse(data={'Battery Remaining Capacity': 89.0, "Month's Energy": 4964.1, 'PV1 Current': 6.8, 'Exported Power': 995.0, 'Battery Voltage': 56.6, 'Inverter Temperature': 38.0, 'Battery Current': 19.95, 'PV2 Current': 6.9, 'Total Energy': 16955.4, 'PV2 Power': 1483.0, 'Grid Frequency': 50.0, 'EPS Current': 0.0, 'Network Voltage': 239.8, 'EPS Power': 0.0, 'Battery Temperature': 30.0, 'EPS Voltage': 0.0, 'Output Current': 11.3, 'EPS Frequency': 0.0, 'Battery Power': 1130.0, 'PV1 Power': 2451.0, 'PV1 Voltage': 360.5, "Today's Energy": 5.4, 'PV2 Voltage': 215.0, 'Power Now': 2663.0}, serial_number='230BCF70', version='Solax_SI_CH_2nd_20160912_DE02', type='AL_SE', status=2)

pi@rpi_hue:~ $ python3 solax
RealTimeResponse(data={'PV2 Voltage': 217.1, 'Total Energy': 16955.4, "Month's Energy": 4964.1, 'EPS Power': 0.0, 'Inverter Temperature': 38.0, 'EPS Current': 0.0, 'PV1 Current': 6.8, 'Battery Remaining Capacity': 89.0, 'Battery Temperature': 30.0, 'EPS Voltage': 0.0, 'Battery Voltage': 56.6, 'PV2 Power': 1497.0, 'PV1 Voltage': 363.8, 'PV2 Current': 6.9, 'Exported Power': 991.0, 'Grid Frequency': 50.0, "Today's Energy": 5.4, 'PV1 Power': 2473.0, 'Power Now': 2630.0, 'Battery Power': 1153.0, 'Output Current': 11.1, 'Network Voltage': 242.9, 'Battery Current': 20.35, 'EPS Frequency': 0.0}, serial_number='230BCF70', version='Solax_SI_CH_2nd_20160912_DE02', type='AL_SE', status=2)

pi@rpi_hue:~ $ python3 solax
RealTimeResponse(data={'PV2 Power': 1489.0, 'Exported Power': -924.0, 'Battery Temperature': 30.0, "Month's Energy": 4964.1, 'Network Voltage': 232.8, 'PV2 Voltage': 215.9, "Today's Energy": 5.5, 'Battery Remaining Capacity': 91.0, 'Battery Current': 43.74, 'PV1 Voltage': 366.5, 'EPS Current': 0.0, 'Power Now': 1249.0, 'Battery Power': 2497.0, 'Inverter Temperature': 38.0, 'EPS Power': 0.0, 'PV1 Power': 2455.0, 'Grid Frequency': 50.02, 'PV1 Current': 6.7, 'Total Energy': 16955.6, 'EPS Voltage': 0.0, 'EPS Frequency': 0.0, 'PV2 Current': 6.9, 'Output Current': 5.4, 'Battery Voltage': 57.02}, serial_number='230BCF70', version='Solax_SI_CH_2nd_20160912_DE02', type='AL_SE', status=2)

so if i want to grep some specific value (awk print$) that specific data is never in the same position.

also sometimes i recieve this error:

pi@rpi_hue:~ $ python3 solax
Received unexpected JSON from Solax API endpoint: required key not provided @ data['method']
{'successful': False, 'message': 'Read real time data timeout'}
Traceback (most recent call last):
  File "solax", line 8, in <module>
    data = loop.run_until_complete(r.get_data())
  File "/usr/lib/python3.5/asyncio/base_events.py", line 466, in run_until_complete
    return future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 293, in result
    raise self._exception
  File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
    result = coro.send(None)
  File "/usr/local/lib/python3.5/dist-packages/solax/__init__.py", line 140, in get_data
    3)
  File "/usr/local/lib/python3.5/dist-packages/solax/__init__.py", line 107, in async_solax_real_time_request
    raise SolaxRequestError
solax.SolaxRequestError

so i decide to update to the last version (sudo pip3 install solax --upgrade
) but i recieve this error:

pi@rpi_hue:~ $ python3 solax
Traceback (most recent call last):
  File "solax", line 8, in <module>
    data = loop.run_until_complete(r.get_data())
  File "/usr/lib/python3.5/asyncio/base_events.py", line 466, in run_until_complete
    return future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 293, in result
    raise self._exception
  File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
    result = coro.send(None)
  File "/usr/local/lib/python3.5/dist-packages/solax/__init__.py", line 52, in get_data
    3)
  File "/usr/local/lib/python3.5/dist-packages/solax/__init__.py", line 26, in rt_request
    return await inv.get_data()
AttributeError: 'str' object has no attribute 'get_data'

so i rollback to 0.1.2 and keep recieve different output everytime.
Someone could help me?

sommo, 2020-04-10:
this is my solax file:
import solax
import asyncio

r = solax.RealTimeAPI('192.168.0.106')

loop = asyncio.new_event_loop()
asyncio.set_event_loop(loop)
data = loop.run_until_complete(r.get_data())
print(data)

sommo, 2020-04-10:
i see that you update also that file, so i upgrade to 0.2.2, edited solax file, and now it works with last update, but the main issue (different output, values order) remains:

pi@rpi_hue:~ $ python3 solax
InverterResponse(data={'Battery Temperature': 30.0, "Month's Energy": 4964.1, 'Exported Power': 881.0, 'EPS Frequency': 0.0, 'PV2 Voltage': 209.0, 'PV1 Power': 2485.0, 'Battery Remaining Capacity': 95.0, 'Grid Frequency': 50.02, 'PV2 Current': 7.1, 'Total Energy': 16956.0, 'Output Current': 11.4, 'Battery Voltage': 57.26, 'Battery Current': 20.96, 'PV2 Power': 1483.0, 'EPS Current': 0.0, 'Inverter Temperature': 38.0, 'EPS Voltage': 0.0, 'PV1 Voltage': 350.0, 'Battery Power': 1201.0, 'EPS Power': 0.0, 'Network Voltage': 235.0, "Today's Energy": 5.9, 'PV1 Current': 7.1, 'Power Now': 2640.0}, serial_number='230BCF70', version='Solax_SI_CH_2nd_20160912_DE02', type='AL_SE')

pi@rpi_hue:~ $ python3 solax
InverterResponse(data={"Month's Energy": 4964.1, 'Total Energy': 16956.0, 'Battery Power': 1373.0, 'PV2 Current': 7.1, "Today's Energy": 5.9, 'Network Voltage': 231.7, 'PV2 Power': 1490.0, 'Battery Voltage': 57.33, 'Grid Frequency': 50.02, 'Output Current': 10.8, 'Battery Current': 23.93, 'Exported Power': 940.0, 'Power Now': 2459.0, 'Battery Temperature': 30.0, 'PV2 Voltage': 209.9, 'EPS Frequency': 0.0, 'Battery Remaining Capacity': 95.0, 'EPS Voltage': 0.0, 'EPS Power': 0.0, 'EPS Current': 0.0, 'Inverter Temperature': 38.0, 'PV1 Current': 7.0, 'PV1 Voltage': 354.2, 'PV1 Power': 2479.0}, serial_number='230BCF70', version='Solax_SI_CH_2nd_20160912_DE02', type='AL_SE')

how you could manage data to display it graphically? if the output values are everytime in different position, maybe i'm old style to use grep/awk but if someone could point me to the right direction to display data collected via your script could be very usefull, thanks!

sommo, 2020-04-13:
Anyway, if someone could point me, how do you visualize data? Home assistant?
Thank you

squishykid, 2020-06-10:
The output is a dictionary, so it will never be ordered. You could write some code to sort the output by key before printing it.

Sometimes the inverter provides an invalid response and this results in a SolaxRequestError. In this case your code can retry the request if appropriate.

closed 2020-06-10 by squishykid