squishykid/solax #19: solax inverter official page

sommo, 2020-04-10:
Sorry for using this github page, but official support/forum seems to not exist (or lack).
Have you some problem with your inverter data display via official app, seems having trouble since 2-3 weeks, the data is not display correctly like solax's server collect only a few moment of data.
Just for asking to know if its only my problem or not.
Cheers to all

sommo, 2020-04-13:
None of you that have Solax inverter use their app?
This is what I get:
[image]

squishykid, 2020-06-09:
Hey, apologies, but I don't know about the app. Maybe you could email them? This github project is only trying to make a python library so you can download data for yourself without the app.

closed 2020-06-09 by squishykid