squishykid/solax #20: Add support for X3-Hybrid (3 phase units)

bbnsn, 2020-05-21:


bbnsn, 2020-05-21:
Hi @squishykid I'm actually a Hubitat user (we share the same tech interests: solax, synology etc) and i'm keen to emulate your success with my solax inverter. Hubitat is a Groovy-based platform. do you know how feasible it is to convert your HA/python to something that will work with Hubitat?

squishykid, 2020-05-21:
I haven't had a chance to have a proper look at the hubitat docs, but if you can make a plugin that is able to make HTTP requests and parse the response, you could do the same as this library.

bbnsn, 2020-05-21:
thanks @squishykid i'm a beginner but i guess i'll try to figure it out, i do like my solax inverter and would love to use the data it provides more intelligently

closed 2020-06-09 by squishykid