squishykid/solax #25: Wifi Disable

xdubx, 2020-10-12:
Hello Guys,

i bought last days a solax inverter with one Wi-Fi dongle. After i connect the Wi-Fi with my router. i wanna run this script over the router connection. The Wi-Fi stick gets an IP address, but don't response on any requests. If i connect directly to the open Wi-Fi from itself the script works. Somebody have an idea to get this to work.

Besides i wanna disable the open Wi-Fi to get a better network security. An Apache password is really fast broken.
On the inverter runs a debian. Do you make more research on the system? Like port scanning or something in this way?
The next time i wanna maybe send them an email or start to reverse the firmware with IDA. Maybe i can grab some informations.

xdubx, 2020-10-18:
Get no answer form the company. I open the WIFI stick. It have some pins to flash. ARM Chip. But to reverse it is to expensive point timecosts. I found the https://github.com/arendst/Tasmota with a Solax X1 support. You can buy a esp8266 and plug it in the same usb slot. 

Port scan shows only port 80 on closed wifi und open wifi is open. 

_On the inverter runs a debian._  on the "wifi stick runs debian". The apache version is for debian.

wills106, 2020-10-18:
> I found the https://github.com/arendst/Tasmota with a Solax X1 support. You can buy a esp8266 and plug it in the same usb slot.

Are you sure you plug the ESP8266 into the USB?
There is no native USB on an ESP8266

Looking at the code it's using ModBus over RS485.

There is only a basic set of Registers in that implementation.

xdubx, 2020-10-18:
> Are you sure you plug the ESP8266 into the USB?
> There is no native USB on an ESP8266
>
> Looking at the code it's using ModBus over RS485.
>
> There is only a basic set of Registers in that implementation.

i will try to connect the gpio1 & 3 RX/TX Pins to the data pins like in the origial Pocket WIFI usb. In my optinion the wifi usb does the same. It connects to the RS485 intern and request the data over rs. 

[image]

wills106, 2020-10-18:
The USB stick won't be talking to the Inverter using ModBus.

If you want to communicate to your inverter over ModBus either use the proper RS485 terminals on your inverter. Or if your inverter has built in ethernet, use that via ModBus over TCP like I am using.

If you connect your ESP to the serial port best case nothing happens. Worse case you will blow the control chip on the Inverter. Rendering you with a rather large white box that doesn't do anything!

xdubx, 2020-10-18:
I opened the inverter to check the tracks from the different ports and yes it don't use the same input lanes as the other. 
Seems to be a little more work on reverse this WiFi usb stick.

closed 2020-10-18 by xdubx