squishykid/solax #26: Add Support for X1-Hybiyd-G3 (Solax X1 Hybrid 2xMPPT)

cheikemate, 2020-11-14:
I want to integrate my Solax X1 Hybrid 2xMPPT into Home Assistant.
I have connectivity with my inverter. Here are the curl responses taken from Home Assistant Linux command line.

**curl -X POST http://5.8.8.8/?optType=ReadRealTimeData**
{"type":"X1-Hybiyd-G3","SN":"SWKXFH86TL","ver":"2.033.20","Data":[2.8,0.0,130.8,0.0,1.5,239.8,380,30,1.6,1.9,-192,364,0,0.00,0.00,0,0,0.0,0,0.0,0,0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0,0,0,0,0,0,0,0,0,0,0,0.03,15.59,0,0,0,0,0,0,0,49.99,0,0,0.0,0.0,0,0.00,1,0,0,0.00,0,8,0,0,0.00,0,8,2,1,6,1.00,60000,100,10,20.00,20.00,0,0,0,0,0.0,0.0,1.2,0.0,0,0,0,0,0,0,0,0,0,0,0,1,1,0,0,0.00,0.00,1,0,0.0,0.0,0],"Information":[3.000,3,"X1-Hybiyd-G3","H1E302F8207046",1,3.11,0.00,3.13,1.05],"battery":{"brand":"0","masterVer":"0.00","slaveNum":"0","slaveVer":[0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00],"slaveType":[0,0,0,0,0,0,0,0]}}

**I have setup the integration yaml.**
sensor:
  - platform: solax
    ip_address: 5.8.8.8

**I have the following error.**
2020-11-13 23:39:33 ERROR (MainThread) [homeassistant.components.sensor] Setup of platform solax is taking longer than 60 seconds. Startup will proceed without waiting any longer.
  File "/usr/src/homeassistant/homeassistant/components/solax/sensor.py", line 30, in async_setup_platform
  File "/usr/local/lib/python3.8/site-packages/solax/__init__.py", line 37, in real_time_api
  File "/usr/local/lib/python3.8/site-packages/solax/inverter.py", line 89, in discover
solax.inverter.DiscoveryError: Unable to connect to the inverter at host=5.8.8.8 port=80, or your inverter is not supported yet.
Please see https://github.com/squishykid/solax/wiki/DiscoveryError

Maybe this is a typo error with the Mode-Type returned from API query ("Hybiyd" ??). But I am not sure because I don't know what kind of data is used to ID the inverter on the Home Assistant side.
I can provide more information if you need.

Regards.

squishykid, 2020-11-23:
Hi @cheikemate were you able to solve this issue?

cheikemate, 2020-11-23:
@squishykid 
Well... I did not resolved the issue but I found a way to make the inverter work.
I decided to read the REST API docs from Solax. 
I have created a sensor that pools the data from Solax Cloud directly into Home Assistant.
Nothing fancy.

```
sensor:        
  - platform: rest
    name: solaxcloud
    json_attributes_path: "$.result"
    json_attributes:
        - inverterSN
        - sn
        - acpower
        - yieldtoday
        - yieldtotal
        - feedinpower 
        - feedinenergy
        - consumeenergy
        - feedinpowerm2
        - soc
        - peps1
        - peps2
        - peps3
        - inverterType
        - inverterStatus
        - uploadTime
    resource: https://www.eu.solaxcloud.com:9443/proxy/api/getRealtimeInfo.do?tokenId=&sn=
    username: blablabla
    password: blablabla
    authentication: basic
    scan_interval: 360
    value_template: '{{ value_json.result.acpower }}'
  - platform: template
    sensors:
      solax_yieldtotal:
        friendly_name: 'inverter.AC.energy.out.total'
        value_template: '{{states.sensor.solaxcloud.attributes["yieldtotal"] }}'
        device_class: power
        unit_of_measurement: 'KWh'
      solax_feedinenergy:
        friendly_name: 'inverter.energy.toGrid.total'
        value_template: '{{states.sensor.solaxcloud.attributes["feedinenergy"] }}'
        device_class: power
        unit_of_measurement: 'KHh'
      solax_consumeenergy:
        friendly_name: 'inverter..energy.fromGrid.total'
        value_template: '{{states.sensor.solaxcloud.attributes["consumeenergy"] }}'
        device_class: power
        unit_of_measurement: 'KWh'
      solax_yieldtoday:
        friendly_name: 'inverter.AC.energy.out.daily'
        value_template: '{{states.sensor.solaxcloud.attributes["yieldtoday"] }}'
        device_class: power
        unit_of_measurement: 'KWh'
      solax_acpower:
        friendly_name: 'inverter.AC.power.total'
        value_template: '{{states.sensor.solaxcloud.attributes["acpower"] }}'
        device_class: power
        unit_of_measurement: 'W'
      solax_feedinpowerm2:
        friendly_name: 'inverter.address2meter.AC.power.total'
        value_template: '{{states.sensor.solaxcloud.attributes["feedinpowerm2"] }}'
        device_class: power
        unit_of_measurement: 'W'
```

cheikemate, 2020-11-23:
Preliminary results on my dashboard.

[image]

squishykid, 2020-11-23:
Nice! I did find why your inverter was not working though, and there is a patch coming in #27 that should fix it for you. If you know how to you can get the code from that branch and test it out for yourself.

It will probably take about 2 weeks to be in the next home assistant release.

closed 2020-11-23 by squishykid

squishykid, 2020-11-24:
https://github.com/home-assistant/core/pull/43564

Check the home-assistant releases, this should be available soon :)

cheikemate, 2020-11-24:
@squishykid 
THANK YOU!

PR #27 (2020-11-23): Add Support for X1-Hybiyd-G3 (Solax X1 Hybrid 2xMPPT)
closes #26
commit ce0c3604 (2020-11-23): Add Support for X1-Hybiyd-G3 (Solax X1 Hybrid 2xMPPT) (#27)

* Add Support for X1-Hybiyd-G3 (Solax X1 Hybrid 2xMPPT)

closes #26

* cleanup