squishykid/solax #32: Inverter X1 Unauthorized on WiFi

gungun974, 2021-04-09:
Hello.

I've been using this library for some time and I'm very happy with it.
(mostly for home assistant)

However, my Inverter X1 has stopped transmitting is information and now, i got this error message.

```html
<HTML>
<HEAD>
<TITLE>Error</TITLE>
<meta http-equiv="Content-Type" content="text/html; charset=UTF-8">
</HEAD>
<BODY><H1>401 Unauthorized</H1></BODY>
```
I never touch any parameter on my Inverter so i things this is maybe an automatic update that broke something.

Luckily I have already managed to find the source of this problem and a potential solution to it that I would therefore like to share today here for anyone who will have the same problem as me.

So when you connect to dongle, normally you can just directly receive Inverter Data from this URI
```/?optType=ReadRealTimeData```

So my first attempt for an 401 error was to add a header with a basic authorization but the inverter does not return any values if i do that.

In reality you have to call the inverter with the following URI with a password in the URL of this one
```/?optType=ReadRealTimeData&pwd=PASSWORD```

(Note: the default password is admin)

It would be nice if an optionally password field is add in this Library

squishykid, 2021-04-10:
This is interesting. Thanks for posting! I will make an update for this to the library. I can't promise changes to the home-assistant side of things though. Lets see how far we get.

gungun974, 2021-04-10:
Great. 
However i manage to find something really strange when i call this url with password.
If i forgot to add "&" in the url like in my previous message (I had already edited my first post), the inverter will completly crash until i disconnect WiFi card and reconnect it.

LuisPalacios, 2021-04-22:
@gungun974, thank you very much; I was getting crazy with this issue... works perfect.

LuisPalacios, 2021-04-22:
By the way, I'm using NGINX and I can't modify the source python file, so in case it's useful to someone else, here is my configuration on NGINX: 

```
server {
	listen 80;
	server_name inversor-proxy.parchis.org;
	access_log /var/log/inversor-proxy.access.log;
	location / {
		set $delimeter "";

		if ($is_args) {
			set $delimeter "&";
		}

		set $args "$args${delimeter}pwd=PASSWORD&";
		proxy_pass http://5.8.8.8;
	}
}

```

LuisPalacios, 2021-04-22:
> This is interesting. Thanks for posting! I will make an update for this to the library. I can't promise changes to the home-assistant side of things though. Lets see how far we get.

@squishykid, I'm using Home Assistant indeed and that's why I can't (don't know how) touch the integration. Meanwhile the "NGINX" workaround is working great :-)

Thanks

tomrys, 2021-05-18:
Hi,
I have Solax X3 Mic with Pocket WiFi and I was looking how to read data in real time, there are some suggestions here https://github.com/squishykid/solax/issues/5 that executing _curl -X POST http://5.8.8.8/?optType=ReadRealTimeData_ will give result, but I got the same result as it is at beginning of this issue.
After sniffing by wireshark the solax app and hanging WiFi module few times, reading about curl, I have command that works for me:
**curl -s -w '\n' -X POST http://5.8.8.8 --data 'optType=ReadRealTimeData&pwd=admin'**
"-s" to keep silent in scripts
"-w '\n\'" to add new line if saved in text file
the result is like this:
_{"sn":"Sxxxxxxxxx","ver":"2.034.06","type":7,"Data":[2427,2450,2437,13,13,13,272,277,278,2758,2686,18,16,475,424,5003,5003,5004,2,7456,0,14,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,8,0,0,0,0,0,8,5000,0,21,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,27,0,0,0,0,0,0,0,827,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],              "Information":[5.000,7,"MC502TG9083105",1,1.24,1.02,1.27,1.02,0.00,1]}_
if it is like this 'http://5.8.8.8/?optType=ReadRealTimeData&pwd=admin' then all is passed as URL and somehow that firmware does not handle that it might worked in past.

If i am not mistaken, there is no time value so to create graph there is need to add time to have nice timeline. 
I hope it will help :-)

commit 19ed75a4 (2021-08-03): X1 Mini with password (#33)

* X1 Mini with password

* fix linter

* fix Class method make_url should have 'cls' as first argument (bad-classmethod-argument)

* fix #32

support for pwd for InverterPost
fix #32

* pylint

* whitespace
closed 2021-08-03 by squishykid