squishykid/solax #36: X1Mini 'Received malformed JSON from inverter'

BoBiene, 2021-08-07:
so now i were able to test the pushed changes. Authentication is successful - but it seems like the json validation fails.

I'm getting this response from my X1Mini:

````json  
{"sn":"SWCPRQ3RVF","ver":"2.034.06","type":4,"Data":[2310,15,349,609,0,58,0,359,0,5000,2,3474,0,12,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,39,0,2518,0,0,0,0,0,0,0,0,0,0,0,0,0,41,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],               "Information":[0.700,4,"XMT072G3042340",1,1.19,0.00,1.32,0.00,0.00,1]}
````

I don't know how to change the schema to match both the old and the new respons for the X1Mini.
@squishykid can you assist with this?

ppetru, 2021-08-12:
Firmware v2.034.06 completely changed the exported data format. I just added support for it in PR #37, but I only have my X3 hybrid to test with.

BoBiene, 2021-08-12:
@ppetru thanks for your work on the X3. I startet to work on the new format last week for the X1 Mini and pushed my changes now to #38

commit 544a16bb (2021-08-16): X1Mini fix #36
commit 5b58e370 (2021-08-17): X1Mini fix #36 (#41)
closed 2021-08-17 by squishykid