squishykid/solax #45: Problem with New Energy monitors reporting

sm0keKings, 2021-09-03:
I setup the new energy monitor today.  It was working ok till the figures went off the chart with no explination.
[image]
The actual numbers are not wrong.
[image]

matthewjporter, 2021-09-03:
@sm0keKings 

Sorry I cant help with the energy monitors but I do Love your stats apge. Anychance you can share you yaml for that page and what cards you are using?

sm0keKings, 2021-09-03:
> @sm0keKings
> 
> Sorry I cant help with the energy monitors but I do Love your stats apge. Anychance you can share you yaml for that page and what cards you are using?

grid_power_entity: sensor.solax_swq5zx7rms_exported_power
production_is_positive: true
solar_power_entity: sensor.solax_swq5zx7rms_ac_power
title: Solar Power
type: custom:power-wheel-card
[image]

entities:
  - entity: sensor.energy_consumed_today_total
    name: Energy Used Today
  - entity: sensor.grid_sold_energy_day
    name: Energy Sold Today
  - entity: sensor.grid_consumed_energy_day
    name: Grid Energy Used Today
hours_to_show: 24
icon: mdi:graph
line_width: 3
name: Energy Used Purchased & Sold
points_per_hour: 4
type: custom:mini-graph-card

entities:
  - sensor.solax_swq5zx7rms_exported_power
  - sensor.solax_swq5zx7rms_ac_power
  - sensor.solax_swq5zx7rms_pv1_power
  - sensor.solax_swq5zx7rms_pv2_power
hours_to_show: 24
icon: mdi:graph
line_width: 3
name: Solar Power Graphs
points_per_hour: 8
type: custom:mini-graph-card
[image]

Any others you are interested in?

squishykid, 2021-09-03:
Howdy! This is probably something you want to raise against the home-assistant/core repo. This is just the adapter project which gets the data from the inverter for home assistant. You probably also want to include a bit more information for the developers over there which they can use to reproduce your problem. Screenshots aren't really enough. You could include your inverter model and maybe some sample responses from your inverter.
If you read the page here you can see how to get a response from your inverter to include with the issue https://github.com/squishykid/solax/wiki/DiscoveryError

squishykid, 2021-09-03:
Ok wait, my bad. let me double check this.

squishykid, 2021-09-03:
Are you referring to how the 'solax solar total yield` is in the megawatts? That is weird.

sm0keKings, 2021-09-03:
> Are you referring to how the 'solax solar total yield` is in the megawatts? That is weird.

All the figures were right for about 6 hours then it went south

sm0keKings, 2021-09-03:
> > Are you referring to how the 'solax solar total yield` is in the megawatts? That is weird.
> 
> All the figures were right for about 6 hours then it went south

Today it is fine at the moment.

[image]

squishykid, 2021-09-04:
Can you please supply some more information:
- inverter model
- Example response as described here: https://github.com/squishykid/solax/wiki/DiscoveryError

I can’t help you with just the information in the screenshots

Please do let me know though if it happens again. My theory right now is that the inverter returns a result with a really big or small number for kWh values, then goes back to the original count.

perhaps zoom in on the time where the issue occurred and show me what happened to the other sensors at the same time when the kWh sensors shot up like that

sm0keKings, 2021-09-04:
> Can you please supply some more information:
>
> * inverter model
> * Example response as described here: https://github.com/squishykid/solax/wiki/DiscoveryError
>
> I can’t help you with just the [...]

Thanks Squishykid!

Today it has been fine.  My inverter is a Hybrid X1.

squishykid, 2021-09-04:
I expect that you will experience this issue intermittently. Lets close this issue for now. Next time it happens it might be best to raise the issue on the home-assistant/core project.

closed 2021-09-04 by squishykid

sm0keKings, 2021-09-04:
> I expect that you will experience this issue intermittently. Lets close this issue for now. Next time it happens it might be best to raise the issue on the home-assistant/core project.

Thanks mate!