squishykid/solax #46: Potential "Work-Around" For Accessing API data via Pocket Wifi v2 over LAN

dvisser, 2021-09-22:
Hi squishykid, 

We have found a potential work-around for accessing the local api data via the pocket wifi v2 over a users LAN. 
First you need to ensure your pocket wifi v2 adapter is running firmware version v2.033.20 or less. 
(you can downgrade from newer firmware if you have the file, I have this file and can get it to you also if you wish). 
Then you need to add the "X-Forwarded-For" header to your http request. This we believe, is the magic that makes this work.
We have done this using the python requests library (code snippet below for you). 
It is worth mentioning that the user should then block all inbound and outbound traffic to their pocket wifi on their network to avoid the pocket wifi auto updating its firmware in the future also.

Cheers,

Dan

```
import requests
from requests.exceptions import Timeout

INV1URL = "http://192.168.0.203/?optType=ReadRealTimeData" # SolaX X1-HYBRID-5.0
TIMEOUT = 2
HEADERS={'X-Forwarded-For':'5.8.8.8'}

try:
    RESP1 = requests.post(INV1URL, headers = HEADERS, timeout = TIMEOUT)
    INV1PV1CURRENT = RESP1.json()['Data'][0]
    INV1PV2CURRENT = RESP1.json()['Data'][1]
    INV1PV1VOLTAGE = RESP1.json()['Data'][2]
    INV1PV2VOLTAGE = RESP1.json()['Data'][3]
    INV1ACCURRENT = RESP1.json()['Data'][4]
    INV1ACVOLTAGE = RESP1.json()['Data'][5]
    INV1ACPOWER = RESP1.json()['Data'][6]
    INV1TEMPERATURE = RESP1.json()['Data'][7]
    INV1ENERGYTODAY = RESP1.json()['Data'][8]
    INV1ENERGYTOTAL = RESP1.json()['Data'][9]
    INV1EXPORTPOWER = RESP1.json()['Data'][10]
    INV1PV1POWER = RESP1.json()['Data'][11]
    INV1PV2POWER = RESP1.json()['Data'][12]
    INV1BATTERYVOLTAGE = RESP1.json()['Data'][13]
    INV1BATTERYCURRENT = RESP1.json()['Data'][14]
    INV1BATTERYPOWER = RESP1.json()['Data'][15]
    INV1BATTERYTEMPERATURE = RESP1.json()['Data'][16]
    INV1BATTERYCAPACITYREMAINING = RESP1.json()['Data'][21]
    INV1TOTALFEEDINENERGY = RESP1.json()['Data'][41]
    INV1TOTALENERGYCONSUMPTION = RESP1.json()['Data'][42]
    INV1POWERNOW = RESP1.json()['Data'][43]
    INV1GRIDFREQUENCY = RESP1.json()['Data'][50]
    INV1EPSVOLTAGE = RESP1.json()['Data'][53]
    INV1EPSCURRENT = RESP1.json()['Data'][54]
    INV1EPSPOWER = RESP1.json()['Data'][55]
    INV1EPSFREQUENCY = RESP1.json()['Data'][56]
except Timeout:
    print('SolaX Hybrid 5.0kW Request Timed Out')
```

dvisser, 2021-09-22:
should also mention we have been using the above for over 12 months flawlessly too :-)

dvisser, 2021-09-22:
here is the version of pocket wifi firmware I mention als
[618.00122.00_Pocket_WIFI_V2.033.20_20190313.usb_.zip](https://github.com/squishykid/solax/files/7207711/618.00122.00_Pocket_WIFI_V2.033.20_20190313.usb_.zip)
o

dvisser, 2021-09-22:
@thezerobeast

dvisser, 2021-09-23:
pull request #47 submitted as I just tested this and it is working (although I had to fix some indents/tab spacing that home assistant didn't like).

[https://github.com/squishykid/solax/pull/47](url)

Johnreidsilver, 2021-10-02:
Works with "X1-Boost-Air-Mini" firmware 2.31.3 too, thanks for sharing!

dvisser, 2021-10-03:
Awesome. Yes pocket wifi firmware versions 2.33.20 or older should work with this method. Thanks for checking it out.

o0shojo0o, 2022-01-16:
Where can i view the wifi firmware versions?

o0shojo0o, 2022-01-17:
I have now found the wifi firmware version on me, I have the 2.034.06 and with that it does not seem to work.

simatec, 2022-01-25:
Is there also a similar possibility for the LAN stick to read out the data locally?

dvisser, 2022-01-26:
I didn't think the LAN dongle's were locked with a SolaX firmware update, but it is possible that they were. What output do you get when you try to retrieve data?

simatec, 2022-01-26:
How can I try to get Daren from the LAN stick?  I've tried through the browser and with curl

dvisser, 2022-02-05:
can you confirm the stick is appearing on your local network? you may need to ensure that the stick has an IP address is the correct range to suit your network. This is done via inverter programming.

simatec, 2022-02-05:
Yes … the LAN Stick have an local IP from my Local Network

squishykid, 2022-02-21:
I'm going to close this issue, as the root cause has been addressed, but feel free to continue the discussion. It should still be possible to reply to the issue.

closed 2022-02-21 by squishykid