squishykid/solax #48: X1-Boost-Air-Mini work around for aiohttp's bad header/(due to fake password prompt?)

Johnreidsilver, 2021-10-02:
aiohttp seems to be very peculiar regarding illegal headers, while curl and requests work fine it throws an error with X1-Boost-Air-Mini. Perhaps due to the fake password prompt that we can see in a browser at the local API URL:
http:/5.8.8.8/?optType=ReadRealTimeData

Running:
```
async def work():
    r = solax.inverter.X1Mini('5.8.8.8',80)
    return (await r.get_data())

loop = asyncio.new_event_loop()
#loop = asyncio.get_event_loop()
asyncio.set_event_loop(loop)
data = loop.run_until_complete(work())
```

returns:
`aiohttp.client_exceptions.ClientResponseError: 400, message='invalid character in header'`

but launching python3 with:
AIOHTTP_NO_EXTENSIONS=1 python3

returns fine
`InverterResponse(data={'PV1 Current': 0.4, 'PV2 Current': 0.0, 'PV1 Voltage': 140.5, 'PV2 Voltage': 0.0, 'Output Current': 0.5, 'Network Voltage': 245.4, 'AC Power': 77.0, 'Inverter Temperature': 31.0, "Today's Energy": 4.3, 'Total Energy': 7265.7, 'Exported Power': 0.0, 'PV1 Power': 87.0, 'PV2 Power': 0.0, 'Total Feed-in Energy': 0.0, 'Total Consumption': 0.0, 'Power Now': 0.0, 'Grid Frequency': 50.03}, serial_number='redacted', version='2.31.3', type='X1-Boost-Air-Mini')
`

Likewise launching HomeAssistent with this variable also makes the data fetch work
AIOHTTP_NO_EXTENSIONS=1 hass

squishykid, 2022-02-21:
Hi @Johnreidsilver, is there a chance you could please share a packet capture, one with `AIOHTTP_NO_EXTENSIONS` set, and another with `AIOHTTP_NO_EXTENSIONS` unset? The packet capture should show the TCP stream between your computer and the inverter. This will let us see what is going over the wire.

If you need some help with producing the packet capture, you could use tools such as Wireshark. Let me know if you need more help and I can get you sorted :)

Johnreidsilver, 2022-04-30:
Thanks for looking into this. Attached packets made with tcpdump 
sudo tcpdump -i w0 -w TCPDUMP.txt

pip3 freeze
solax==0.2.8

on another newer system I have solax==0.2.10 and it fails even after setting AIOHTTP_NO_EXTENSIONS=1 

looking at the packets the data is in both of them, but aiohttp seems to get halted with the error when extensions aren't disabled
[TCPDUMP.txt](https://github.com/squishykid/solax/files/8596742/TCPDUMP.txt)
[DUMPwithAIOHTTP_NO_EXTENSIONS.txt](https://github.com/squishykid/solax/files/8596743/DUMPwithAIOHTTP_NO_EXTENSIONS.txt)

VadimKraus, 2022-07-18:
Did you try using the discovery method,  cause it looks like you are using the wrong inverter to connect, in the 0.2.10 example

Johnreidsilver, 2022-08-01:
No, my commands were exactly as above.
Will test with solax.inverters.x1_boost when possible
https://github.com/squishykid/solax/blob/master/solax/inverters/x1_boost.py

squishykid, 2025-01-18:
Closing this for now, i think the library has changed a lot since this was last commented on. I think this may no longer be relevant.

closed 2025-01-18 by squishykid