squishykid/solax #49: Problems retrieving data from the Solax X1 Boost model using local IP

ignaciorecuerda, 2021-11-03:
Hello! 
first of all, congratulations for the work you have done. 
I am trying to use your application to see the real time data of my Solax X1 Boost. Before that I installed on my Solax a fireware that allows me to call it and receive the data (I tested it with the following curl call and it works: `curl -d "optType=ReadRealTimeData&pwd=admin" -X POST http://192.168.1.136) `
As you can see the IP of the "pocket wifi" is 192.168.1.136 in my local network, but when I execute the line `r = await solax.real_time_api('192.168.1.136')` of the function "work" I get the following message because it can't retrieve the data on that IP and on that port.

`solax.inverter.DiscoveryError: Unable to connect to the inverter at host=192.168.1.136 port=80, or your inverter is not supported yet.
Please see https://github.com/squishykid/solax/wiki/DiscoveryError
Failures=[InverterError('Could not connect to inverter endpoint', 'XHybrid'), InverterError('Could not connect to inverter endpoint', 'X3'), InverterError('Could not connect to inverter endpoint', 'X3V34'), InverterError('Could not connect to inverter endpoint', 'X1'), InverterError('Could not connect to inverter endpoint', 'X1Mini'), InverterError('Could not connect to inverter endpoint', 'X1MiniV34')]`

Do you know what I am doing wrong?
Thank you very much in advance

dvisser, 2021-11-03:
can you post the output of your curl command please (screenshot)? and were you accessing the pocket wifi AP direct or is this over your LAN?

ignaciorecuerda, 2021-11-04:
Hello @dvisser 
the command I am using is `curl -d "optType=ReadRealTimeData&pwd=admin" -X POST http://192.168.1.136` and the result is 
`{"sn":"XXXXXXXXXXXX","ver":"2.034.06","type":4,"Data":[2312,6,0,1175,0,0,0,0,0,4999,2,474,0,169,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,23,0,59,0,0,0,0,0,0,65271,65535,3027,0,3989,0,0,31,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],               "Information":[3.680,4,"XXXXXXXXXXXX",1,2.15,0.00,1.35,0.00,0.00,1]}`

Yes, it is  a call using my LAN

Thank you for your time

ignaciorecuerda, 2021-12-01:
Hello @dvisser @squishykid 
any news on this topic? is there compatibility with the Solax X1 Boost?

Thank you in advance

davoau, 2021-12-03:
Hi, I could not get it to work using the pocket wifi ip address on local lan with curl command. 
I had to connect to pocket wifi ip address 5.8.8.8 to get it to work.
What I have since done with home assistant on a raspberry pi is set up an access point using HOSTAPD on the pi, then connect solax wifi dongle to that, and use dnsmasq to redirect all requests to www.eu.solaxcloud.com, mqtt001.solaxcloud.com and mqtt002.solaxcloud.com to the Pi. All data from solar is sent directly to home assistant via mqtt. 
The benefit of this is I can turn on my hotwater when there is enough solar to heat it, plus do other home automation tasks.

Johnreidsilver, 2021-12-05:
The disadvantage of that solution is a longer interval between updates of the data. Your pocket wifi firmware might be too high to allow the local lan trick to work.

davoau, 2021-12-05:
Yes you are correct, you have no control over when they updates are received, however they are sent out every 5mins, which is good enough for me.
I did try some older firmware versions I could find, but nothing worked.

dvisser, 2021-12-19:
@davoau look in my PR for pocketwifi v2 firmware that allows LAN access with some minor changes.

Krettoss, 2021-12-30:
Hi, I had the same issue and I solved it with this:
` r = await solax.real_time_api('192.168.1.136', 80, 'admin')`

squishykid, 2021-12-31:
@ignaciorecuerda The answer above should resolve your issues.

closed 2021-12-31 by squishykid

arnemauer, 2022-03-25:
> above should resolve your issues.

No, it does not resolve the issues. Firmware 2.034.06 has a new data structure. 

**2.033.20**
{**"type":"X1-Boost-Air-Mini"**,"SN":"XXXXXXXXXX","ver":"2.033.20","Data":[8.9,0.0,240.1,0.0,8.1,244.6,1989,44,6.3,3033.4,0,2147,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0.00,0.00,0,0,0,0,0,0,0,49.99,0,0,0,0,0,0,0,0,0,0.00,0,0,0,0,0.00,0,0,2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[2.500,4,"X1-Boost-Air-Mini","XXXXXXXXXX",1,3.32,2.09,1.17,0.00]}

**2.034.06**
{"sn":"XXXXXXXXXX","ver":"2.034.06",**"type":4**,"Data":[2454,80,1989,2396,0,89,0,2148,0,4999,2,30333,0,62,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,51,0,6019,0,0,0,0,0,0,0,0,0,0,0,0,0,44,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],           "Information":[2.500,4,"XXXXXXXXXX",1,3.32,2.09,1.17,0.00,0.00,1]}

The attribute "type" is different in the newer firmware version.

CAP-Team, 2023-05-31:
On a X1-Boost/Air/Mini I get an error after:
curl -d "optType=ReadRealTimeData&pwd=admin" -X POST http://192.168.2.160

The error is:
curl: (7) Failed to connect to 192.168.2.160 port 80: Connection refused.

arnemauer, 2023-05-31:
Did you update the firmware of the wifi stick with the custom firmware?

CAP-Team, 2023-05-31:
No I did not. I'll look into that. Thanks