squishykid/solax #58: Success with X1-5.0-T-D(L)

Azelphur, 2022-03-18:
Readme asks us to report success with other inverters, I just had an X1-5.0-T-D(L) fitted and can confirm it works with this library and the home assistant integration (at least, apart from [this bug](https://github.com/home-assistant/core/issues/66912) but that's not the fault of this library I think)

Thanks for writing this! :+1:

straga, 2022-03-22:
> Readme asks us to report success with other inverters, I just had an X1-5.0-T-D(L) fitted and can confirm it works with this library and the home assistant integration (at least, apart from [this [...]

Hello. You user Solax Smart Meter for Export Control ?

Azelphur, 2022-03-22:
> Hello. You are using Solax Smart Meter for Export Control?

Believe not. I have a meter next to the inverter which monitors how much power is generated, which is an ECA2.z. I currently don't have anything monitoring my grid return, but in a few weeks I'll be getting a SMETS 2 meter fitted that will monitor my grid return, which is standard in the UK.

Azelphur, 2022-03-22:
Ah, I'm still new to this, only had my inverter for a week. I have never connected anything over RS485, so I'd imagine I'm not using export control. If you know a way to check, happy to do so :)

straga, 2022-03-22:
Ok. Thanks for the information.

squishykid, 2022-03-27:
Thanks for the nice message @Azelphur! As mentioned on the home assistant issue, I’ll have to have a look if we can figure out how to filter the output. All the best :)

closed 2022-03-27 by squishykid

straga, 2022-03-28:
If somebody needs more information: Now Emulator Solax Smart Meter works.
- https://github.com/syssi/esphome-modbus-solax-x1/issues/20
- https://github.com/straga/Smart-Meter-Gateway