squishykid/solax #66: Allow library users to construct specific inverter by name

squishykid, 2022-04-21:
Home assistant currently has a problem where by the integration fails to function if the home-assistant instance is rebooted when the inverter is offline.

The inverter might be offline at night if there is no battery attached, or if the battery runs out of power.

Solax should be adjusted so that integrations can use the discovery mechanism to find the inverter, then store the config in some external persistent store.

Later when integrations wish to reconnect to the same inverter, they should be able to use the stored config to reconnect, without going via the discovery mechanism

VadimKraus, 2022-04-22:
I guess more information is needed: Which home assistant version etc. 

I just tested it with a mock server on the current dev branch of home assistant and reconnect/discovery worked fine.

```python
import http.server
import socketserver
import json

PORT = 8000

response = json.dumps(
    {
        "method": "uploadsn",
        "version": "Solax_SI_CH_2nd_20160912_DE02",
        "type": "AL_SE",
        "SN": "XXXXXXX",
        "Data": [1] * 68,
        "Status": 2,
    }
).encode("utf-8")

class SolaxHandler(http.server.SimpleHTTPRequestHandler):
    def _set_response(self):
        self.send_response(200)
        self.send_header("Content-type", "application/json")
        self.end_headers()

    def do_GET(self):
        self._set_response()
        self.wfile.write(response)

    def do_POST(self):
        self.do_GET()

with socketserver.TCPServer(("", PORT), SolaxHandler) as httpd:
    print("serving at port", PORT)
    httpd.serve_forever()
```

Reproduce:
1. Start the mock server
2. Run home assistant (e.g. via devcontainer on the dev branch )
3. Add solax Integration
4. shutdown home assistant 
5. shutdown mock server
6. start home assistant -> integration will show up as failed
7. start mock server
8. wait (this might now depend on the retry back off )
9. integration will be valid again

However, 8. might be a cause for trouble if the back off is exponential, home assistant will probably be in a state that it tries after a very long time, so long that the inverter is already offline again? In the logs I saw a max value of 80 seconds, but this might change if it fails continuously for hours. There might also be a maximum retry count. Could not find anything about this in the documentation...

squishykid, 2025-01-18:
Since refactoring discovery, i think this is no longer necessary.

closed 2025-01-18 by squishykid