squishykid/solax #67: Solax integration occasionally returning all zeros

squishykid, 2022-04-21:
Cloned from https://github.com/home-assistant/core/issues/66912

VadimKraus, 2022-04-22:
See #68 for a possible solution. 
This solution relies a validation error and the home assistant retry mechanism(i.e. exception handling of user of this library).

joseal, 2023-10-29:
Any ETA, went his fix is going to be released? This is impacting a ton of users... thank you!

VadimKraus, 2023-10-29:
Updated #68.

VadimKraus, 2024-06-05:
@squishykid this is resolved I guess

jesserockz, 2025-09-16:
This is still an issue: https://github.com/home-assistant/core/issues/146493

jesserockz, 2025-09-16:
My inverter does report some of the values as 0 overnight, but the totals don't go away, except sometimes while it is starting up when the sun comes up in the morning.