squishykid/solax #73: Info for Solax X1AC Battery only inverter

rwg0, 2022-06-09:
Well, this seems to be the best place on the web to find info about how to decode the data from Solax inverters, and I got a lot of hints from looking through the code here, so I thought I would share what I discovered to allow you to add (what I think) is a currently unsupported model.

The X1-AC is a battery only inverter to add battery storage when you already have an existing solar panel array. There's also a separate 'RetroFit' version that has EPS (which we will be swapping to - the wrong one got ordered for us...)

Anyway, you can get the data by 

`curl -s -X POST http://espressif  --data 'optType=ReadRealTimeData&pwd=WIFISTICKCODE' `

This works on the main WIFI network as well as the 5.8.8.8 one, but you need the WIFI stick code, which I think is a thing from newer versions of the WIFI adapter. You may need to use the IP address instead of the espressif name. No traditional auth is required - the pwd parameter in the post body is fine.

Data that comes back is

`{"sn":"<WIFI STICK CODE>","ver":"3.003.02","type":9,"Data":[0,0,0,0,0,0,0,0,0,0,9,12,0,12,0,0,0,25,10,0,13,5,0,0,0,0,0,8,0,0,0,0,0,8,100,0,0,494,0,0,0,0,0,0,64863,65535,4307,0,8170,0,673,0,0,1314,900,318,200,226,206,33,0,0,0,0,0,0,0,0,0,0,0,0,33,1,1,1,3,0,2,0,0,0,0,0,20,0,221,0,0,0,0,0,0,0,0,0,3,0,68,0,0,0,0,0,13,2,0,0,0,0,0,0,0,0,0,0,1,256,2093,1811,5638,594,520,8995,0,0,0,0,0,17,1174,0,0,3264,3258,11647,0,21302,14389,18753,12593,16689,12355,13876,21302,14389,18753,12593,16689,12355,13876,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[3.680,9,"<INVERTER SERIAL NUMBER>",8,1.20,0.00,1.20,1.04,0.03,1]}`

Note that this is actually data with the battery asleep, so some things are zero that are valid when the battery is active.

Here's the data mapping I have worked out so far (zero based indices). Values in the data are all unsigned 16 bit, negative values are 65535=-1, etc.

0 : AC voltage at inverter * 10  (zero in sleep mode)
1 : AC current from battery * 10  (-ve = to battery)
2 : AC power from battery 
9 : AC frequency * 100
14 : Battery voltage * 100
15 : DC Battery current * 100 (positive = charging)
16 : DC Battery power (positive = charging)
17 : Battery temperature (C)
18 : Battery % remaining - note stops discharging at floor of 10%
44 : Grid power - negative = import
50 : Grid import + AC power draw from battery - sort of net site usage

46 (perhaps 46/47) and 48 (48/49) look like they are counters of some sort - low word / high word pairs? I haven't deciphered what though.

Anyway, hope this is useful to someone :)

squishykid, 2025-01-18:
Any chance that you have figured out what 46/47 and 48/49 represent?

rwg0, 2025-01-18:
Hi,

unfortunately not, and I don't have that model any more (it was a temporary one that we had for about 6 months until the one we had actually ordered came available - we now have the X1-Hybrid). I do have more fields worked out for the new hybrid one, but they are not always in the same order/position as for the X1-AC, so not much help.

In the end I found the best way to decode was to grab the data from the http endpoint and at about the same time grab the data via modbus over TCP. Since the modbus registers are documented, you can then try to match up values read from modbus with those read via http.

cheers,

Robin

```