squishykid/solax #76: Total yield in two registers?

rupertnash, 2022-07-17:
While looking at #75, I realised that if the data we're getting from the inverter is basically the same as the MODBUS data, then each datum is a 16 bit register value. Since the energy yield is in units of 0.1 kWh, that means the max yield it can respond with is 6,553.5 kWh - for my system that's probably about 1 year's generation

So - my question is: is the yield actually split over 2 data items? I see that the next item in the data array is currently zero for me. Similarly for the total feed-in energy and total consumption.

I just looked in the code of the Solax MODBUS thing (https://github.com/wills106/homeassistant-solax-modbus/blob/784153e6a0039402335026281ddca9aadc532a67/custom_components/solax_modbus/__init__.py#L802) and they clearly believe this to be the case.

If this is the case, then we will need to alter the response -> data code to cope with values that come from multiple registers.

VadimKraus, 2022-07-18:
I think it's already covered using this post processing method https://github.com/squishykid/solax/blob/ad975ae8a04299e1d4854c0d02ef65cd18937848/solax/utils.py#L21

rupertnash, 2022-07-18:
Ah - sorry I hadn't seen that - thanks for flagging for me. The choice of name ("reset") doesn't really reflect the way I was thinking about it (just encoding a 32 bit val over 2 16bit words). Those functions are only used in inverters I haven't just for my cribbing...

I implemented a solution already yesterday (https://github.com/rupertnash/solax/commit/21a2d19945e288123abf878aa6769af18b33c495)

If project leads want to keep using the existing code, that's cool otherwise I could also adapt `qvolt_hyb_g3_3p.py` and `x3_v34.py` to use my proposed code.

rupertnash, 2022-07-18:
I think the `resetting_counter` has an off-by-one error.

If there has been 1 reset/overflow of the least significant word, and the sensor LSW is zero, then the result should be 65,536. (Unless the inverter is doing something weird with its encoding of data, which is always possible...)

rupertnash, 2022-07-18:
Can anyone with an inverter that's been running long enough please confirm this by checking against the SolaX cloud view?

VadimKraus, 2022-07-18:
You are totally right. I have one with 8 "resets" there is an offset of 8 between cloud and this lib!

I vote in favor of your solution.

rupertnash, 2022-07-18:
Great- thanks for checking. I’ve made a PR (#77) which includes this feature. Feel free to separate that out and merge separately (if you have the permission lol)

VadimKraus, 2022-07-18:
btw. `to_signed` is probably also one off

```
> import struct
> print(struct.unpack("h",struct.pack("<H",40000)))
(-25536,)
> print(to_signed(40000))
-25535

VadimKraus, 2023-03-12:
This was fixed in #77,  right?

squishykid, 2025-01-18:
Looks like this was fixed in #77. Closing

closed 2025-01-18 by squishykid