squishykid/solax #80: Support for X3 Hbyrid G4

S1lent-Gh0st, 2022-07-24:
Curl command: `curl -d "optType=ReadRealTimeData&pwd=SVXXXXXXXX" -X POST http://10.0.1.6`
Output:             
 `{"sn":"SVXXXXXXXX","ver":"3.003.02","type":14,"Data":[2269,2213,2269,9,10,10,106,102,111,319,2932,2958,5,5,160,159,4996,4997,4996,2,0,0,0,0,0,0,0,0,0,0,0,0,0,1,54317,65535,0,0,0,30770,0,0,3063,0,0,1,48,11538,256,12308,5906,5639,100,0,37,0,0,0,0,0,0,0,0,0,0,0,0,0,1313,0,388,0,0,0,322,0,345,0,45,51,1342,0,406,0,0,0,26,0,4882,0,0,0,3339,0,0,0,0,0,0,0,0,0,1,9,1,33,8,256,3504,2400,264,162,266,225,32,32,4,1620,775,14135,14135,14135,0,0,0,3205,3191,18509,1,20564,12339,18497,12855,16695,12612,13616,20564,12339,18754,12595,16696,12612,12598,20564,12339,18754,12595,16691,13124,12854,20564,12339,18754,12595,16691,13124,13878,0,0,0,0,0,0,0,4098,259,1281,259,0,30770,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,3584,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[8.000,14,"H34A08I3438010",8,1.18,0.00,1.17,1.08,0.00,1]}`

FW: 3.001.02

Connection: `Proxy: 10.0.1.6:80`
		    `Solax: 5.8.8.8 (Wi-Fi dongle) via NAT:GW:10.0.1.2 -> 5.8.8.1 (Client mode: wifi_SVXXXXXXXX)`
Logs:
			
`2022-07-24 09:46:30 WARNING (Thread-3) [matplotlib.font_manager] Matplotlib is building the font cache; this may take a moment.`

Home Assistant version: core-2022.7.6 docker

I tried to write new inverter class with sensor map I found at 
https://github.com/home-assistant/core/issues/66617#issuecomment-1114612301

`"""
Made from x3_v34.py
Commented lines either don't have equivalent in X3_G4
_sensor_map or the equivalent isn't known yet.
Not sure if everything up to line 33 is how it 
supposed to be.

"""

import voluptuous as vol
from solax.inverter import InverterPost
from solax.utils import div10, div100, twoway_div10, to_signed, pv_energy, \
    twoway_div100, total_energy, discharge_energy, charge_energy, \
    feedin_energy, consumption, eps_total_energy

class X3HybridG4(InverterPost):
    """X3-Hybrid-G4 v2.035.01"""
    _schema = vol.Schema({
        vol.Required('type'): vol.All(int, 5),
        vol.Required('sn'): str,
        vol.Required('ver'): str,
        vol.Required('Data'): vol.Schema(
            vol.All(
                [vol.Coerce(float)],
                vol.Length(min=300, max=300), #0-299
                )
            ),
        vol.Required('Information'): vol.Schema(
            vol.All(
                vol.Length(min=10, max=10)
                )
            ),
    }, extra=vol.REMOVE_EXTRA)

    _sensor_map = {
        'Network Voltage Phase 1':               (0, 'V', div10),
        'Network Voltage Phase 2':               (1, 'V', div10),
        'Network Voltage Phase 3':               (2, 'V', div10),

        'Output Current Phase 1':                (3, 'A', twoway_div10),
        'Output Current Phase 2':                (4, 'A', twoway_div10),
        'Output Current Phase 3':                (5, 'A', twoway_div10),

        'Power Now Phase 1':                     (6, 'W', to_signed),
        'Power Now Phase 2':                     (7, 'W', to_signed),
        'Power Now Phase 3':                     (8, 'W', to_signed),

        'AC Output Power':                       (9, 'W', to_signed),

        'PV1 Voltage':                           (10, 'V', div10),
        'PV2 Voltage':                           (11, 'V', div10),
        'PV1 Current':                           (12, 'A', div10),
        'PV2 Current':                           (13, 'A', div10),
        'PV1 Power':                             (14, 'W'),
        'PV2 Power':                             (15, 'W'),

        'Grid Frequency Phase 1':                (16, 'Hz', div100),
        'Grid Frequency Phase 2':                (17, 'Hz', div100),
        'Grid Frequency Phase 3':                (18, 'Hz', div100),

        'Feed-in':                               (34, 'W', to_signed),

        'Total Energy':                          (68, 'kWh', total_energy),
        'Today\'s Energy':                       (70, 'kWh', div10),

        'Battery Voltage':                       (39, 'V', div100),
        'Battery Current':                       (40, 'A', twoway_div100),
        'Battery Power':                         (41, 'W', to_signed),
        'Battery Temperature':                   (105, 'C'),
        'Battery Remaining Capacity':            (103, '%'),
        'Battery Remaining Energy':              (106, 'kWh', div10),

        'Total Feed-in Energy':                  (86, 'kWh', feedin_energy),

        'Total Consumption':                     (88, 'kWh', consumption),
    }`
I didn't test the code since I'm newbie with HA and I don't know how to test it properly
I also found out that there's a pull request that was closed by author for the same model https://github.com/squishykid/solax/pull/74
I don't know what's the reason for closing it since it seems to work (based on the comments) so maybe it would be worth checking it out.

rupertnash, 2022-08-01:
See also my issue #75 and PR #77

It may be possible to address both of these in the same code since both are G4.

WikiHA, 2022-09-26:
Hi,
my Solax X3-Hybrid-G4 inverter has a wifi module with firmware version 3.001.02. The interface is also available on a local IP address (eg 192.168.*.*) in the format http://{IP}/ on port 80 with a POST request "optType=ReadRealTimeData&pwd={password}" where {password} is the SN of the module wifi. The data is in the format given in the first post and I think it exactly matches the QVOLTHYBG33P (and the inverter looks very similar too). 
However, I failed to install the SolaX Power integration, the installation ends with an Unexpected error.
I "clumsily" wrote a yaml module that reads the data like this and it works - I took the liberty of attaching it.
The word [54] in the Data section is probably the inverter temperature - measured on the heatsink.

[solaxg4.txt](https://github.com/squishykid/solax/files/9648324/solaxg4.txt)

ukshark, 2022-10-07:
Hi WikiHa, this is exactly what i need, does this script go into "automation" because when i add it, it errors with :
Message malformed: extra keys not allowed @ data['0']
Im guessing im putting it in the wrong place (Still pretty new at HA sorry)

adambogocz, 2022-12-20:
Hi,
you can watch out for my PR #100 😉  , I am using it already with HA at my home.

jansmets, 2022-12-21:
Also works for X3 MIC G2 
(output in night mode)

curl -d "optType=ReadRealTimeData&pwd=XYZ" -X POST http://192.168.0.191
{"sn":"XYZ","ver":"3.006.04","type":16,"Data":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,29,0,29,8000,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,65108,65535,166,0,6967,0,0,384,1,36352,3,3841,256,15360,15369,29707,39435,0,0,0,0,0,0,0,0,0,0,0],"Information":[8.000,16,"XYZ",8,1.00,0.00,1.13,1.01,0.00,1]}

adambogocz, 2022-12-21:
#100 integrated, so you should be able to try it on master now

S1lent-Gh0st, 2023-01-10:
Can confirm it works for my inverter.

98jan, 2024-01-04:
Hi,

stumbled over this post.
So I do have a X3-Hybrid G4, which uses a WiFi Dongle with the Firmware-Version: 2.034.06.

I tried with following command to get some data:
curl -d "optType=ReadRealTimeData&pwd=XYZ" -X POST http://5.8.8.8/

as password I tried:
-  admin (old password)
- my actual password which is used for basic auth
- SN Number of WiFi dongle

As response I get following json:
`0:code,message:"failed"}`

When I do not pass a pwd I get a 401 Unauthorized response.

Anyone who can help me here.

If you need additional informations, don't hestitate to ask me.

Best regards,

nomano, 2024-08-02:
> Hi,
>
> stumbled over this post. So I do have a X3-Hybrid G4, which uses a WiFi Dongle with the Firmware-Version: 2.034.06.
>
> I tried with following command to get some data: curl -d [...]

@98jan Hello, did you resolve mentioned issue?

darmach, 2024-11-10:
@98jan Is this the newest firmware for dongle?

I'm running:
```
$ curl -sd "?optType=ReadRealTimeData&pwd=SXXXXXX" -X POST http://192.168.1.100 | jq .Information
[
  10.000,
  14,
  "HXXXXXXXXXXXX",
  8,
  1.27,
  0.00,
  1.26,
  1.09,
  0.00,
  1
]
$ curl -sd "?optType=ReadRealTimeData&pwd=SXXXXXX" -X POST http://192.168.1.100 | jq .ver
"3.006.04"
```

What's interesting, in http://192.168.1.100 web interface the firmware is slightly different: 3.001.02

Is the latter the dongle firmware?

In my case api is accessible with curl above.

squishykid, 2025-01-18:
Just circling back, i think support for the X3 Hybrig G4 was added in #100. I will close this issue for now :)

closed 2025-01-18 by squishykid