squishykid/solax #84: Add Support for X3-MIC/PRO-G2

horamarques, 2022-08-01:
X3-MIC/PRO-G2 connected with Pocket wifi on local network
Data is acessible from local IP. 

curl -d "optType=ReadRealTimeData&pwd=XXXXXXXXXX" -X POST http://192.168.68.145
{"sn":"XXXXXXXXXX","ver":"3.003.02","type":16,"Data":[2394,2396,2384,20,20,21,478,479,500,2144,0,0,75,0,0,1617,0,0,4999,5000,5000,2,1437,0,21,4000,43,36,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,64551,65535,8264,0,15768,0,1562,4,1,11,1,0,27485,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[4.000,16,"MC204TIXXXXXXX",8,1.10,0.00,1.07,1.01,0.00,1]}%    

It seems to me that the output data should match the implementation already done for model **"qvolt_hyb_g3_3p"**

horamarques, 2022-08-05:
what I was able to validate so far:

{
   "sn":"XXXXXXXXXX",
   "ver":"3.003.02",
   "type":16,
   "Data":[
      2394, "Network Voltage Phase 1": (0, Units.V, div10),
      2396, "Network Voltage Phase 2": (1, Units.V, div10),
      2384, "Network Voltage Phase 3": (2, Units.V, div10),
      20,   "Output Current Phase 1": (3, Units.A, twoway_div10),
      20,   "Output Current Phase 2": (4, Units.A, twoway_div10),
      21,   "Output Current Phase 3": (5, Units.A, twoway_div10),
      478,  "Power Now Phase 1": (6, Units.W, to_signed),
      479,  "Power Now Phase 2": (7, Units.W, to_signed),
      500,  "Power Now Phase 3": (8, Units.W, to_signed),
      2144, "AC Power": (9, Units.W, to_signed),
      0,
      0,
      75,   "PV1 Voltage": (12, Units.V, div10),
      0,    "PV2 Voltage": (13, Units.V, div10),
      0,    "PV3 Voltage": (14, Units.V, div10),
      1617, "PV1 Current": (15, Units.A, div10),
      0,    "PV2 Current": (16, Units.A, div10),
      0,    "PV3 Current": (17, Units.A, div10),
      4999, "Grid Frequency Phase 1": (18, Units.HZ, div100),
      5000, "Grid Frequency Phase 2": (19, Units.HZ, div100),
      5000, "Grid Frequency Phase 3": (20, Units.HZ, div100),
      2,
      1437, Total Yield
      0,
      21,
      4000,
      43,
      36,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      1,
      0,
      0,
      0,
      64551,
      65535,
      8264,
      0,
      15768,
      0,
      1562,
      4,
      1,
      11,
      1,
      0,
      27485,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0
   ],
   "Information":[
      4.000,
      16,
      "MC204TIXXXXXXX",
      8,
      1.10,
      0.00,
      1.07,
      1.01,
      0.00,
      1
   ]
}

horamarques, 2022-11-18:
> do you have any update on data values?

Not yet, I will try to update it during the weekend.

Trilis29, 2022-11-22:
Have you tried?

horamarques, 2022-11-23:
I'm not able to identify all the values. Even so please find the ones I was able to identify

{
"sn":"XXXXXXXXXX",
"ver":"3.003.02",
"type":16,
"Data":[
2394, "Network Voltage Phase 1": (0, Units.V, div10),
2396, "Network Voltage Phase 2": (1, Units.V, div10),
2384, "Network Voltage Phase 3": (2, Units.V, div10),
20, "Output Current Phase 1": (3, Units.A, twoway_div10),
20, "Output Current Phase 2": (4, Units.A, twoway_div10),
21, "Output Current Phase 3": (5, Units.A, twoway_div10),
478, "Power Now Phase 1": (6, Units.W, to_signed),
479, "Power Now Phase 2": (7, Units.W, to_signed),
500, "Power Now Phase 3": (8, Units.W, to_signed),
2144, "PV1 Voltage": (9, Units.V, div10),				
0, "PV2 Voltage": (10, Units.V, div10),
0, "PV3 Voltage": (11, Units.V, div10),
75, "PV1 Current": (12, 'A', div10),
0, "PV2 Current": (13, 'A', div10),
0, "PV3 Current": (14, 'A', div10),
1617, "PV1 Power": (15, 'W',
0, "PV2 Power": (16, 'W'),
0, "PV3 Power": (17, 'W'),
4999, "Grid Frequency Phase 1": (18, Units.HZ, div100),
5000, "Grid Frequency Phase 2": (19, Units.HZ, div100),
5000, "Grid Frequency Phase 3": (20, Units.HZ, div100),
2,
1437, Total Yield
0,
21, Daily yield
4000,
43,
36,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
1,
0,
0,
0,
64551,
65535,
8264,
0,
15768,
0,
1562,
4,
1,
11,
1,
0,
27485,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0
],
"Information":[
4.000, "Rated Power" (KW)
16, "inverterType"
"MC204TIXXXXXXX", "inverterSN"
8,
1.10,
0.00,
1.07,
1.01,
0.00,
1
]
}

Trilis29, 2022-11-23:
I have smart meter connnected, later I will try to find out how to calculate current consumption in Wh. I think it should be something with 65535 - 64551.

Trilis29, 2022-11-26:
{
"sn":"XXXXXXXXXX",
"ver":"3.003.02",
"type":16,
"Data":[
2394, "Network Voltage Phase 1": (0, Units.V, div10),
2396, "Network Voltage Phase 2": (1, Units.V, div10),
2384, "Network Voltage Phase 3": (2, Units.V, div10),
20, "Output Current Phase 1": (3, Units.A, twoway_div10),
20, "Output Current Phase 2": (4, Units.A, twoway_div10),
21, "Output Current Phase 3": (5, Units.A, twoway_div10),
478, "Power Now Phase 1": (6, Units.W, to_signed),
479, "Power Now Phase 2": (7, Units.W, to_signed),
500, "Power Now Phase 3": (8, Units.W, to_signed),
2144, "AC Power": (9, Units.W, to_signed),
0,
0,
75, "PV1 Voltage": (12, Units.V, div10),
0, "PV2 Voltage": (13, Units.V, div10),
0, "PV3 Voltage": (14, Units.V, div10),
1617, "PV1 Current": (15, Units.A, div10),
0, "PV2 Current": (16, Units.A, div10),
0, "PV3 Current": (17, Units.A, div10),
4999, "Grid Frequency Phase 1": (18, Units.HZ, div100),
5000, "Grid Frequency Phase 2": (19, Units.HZ, div100),
5000, "Grid Frequency Phase 3": (20, Units.HZ, div100),
2,
1437, Total Yield div10
0,
21, Daily Yield div10
4000,
43,
36,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
1,
0,
0,
0,
64551,
65535,
8264,
0,
15768, Feed in total Consumed (same numbers in app) need to experiment more  div100
0,
1562, AC Power (W)
4,
1,
11,
1,
0,
27485,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0,
0
],
"Information":[
4.000,
16,
"MC204TIXXXXXXX",
8,
1.10,
0.00,
1.07,
1.01,
0.00,
1
]
}

adambogocz, 2022-12-20:
Hi @Trilis29,
try following decode table and tell me if it works for you and please copy to_signed32 function from my PR #100.

```
GridAVoltage: value: .1 * Data[0]
GridBVoltage: value: .1 * Data[1]
GridCVoltage: value: .1 * Data[2]
GridACurrent: value: .1 * to_signed(Data[3])
GridBCurrent: value: .1 * to_signed(Data[4])
GridCCurrent: value: .1 * to_signed(Data[5])
GridAPower: value: to_signed(Data[6])
GridBPower: value: to_signed(Data[7])
GridCPower: value: to_signed(Data[8])
Vdc1: value: .1 * Data[9]
Vdc2: value: .1 * Data[10]
Vdc3: value: .1 * Data[11]
Idc1: value: .1 * Data[12]
Idc2: value: .1 * Data[13]
Idc3: value: .1 * Data[14]
PowerDc1: value: Data[15]
PowerDc2: value: Data[16]
PowerDc3: value: Data[17]
FreqacA: value: .01 * Data[18]
FreqacB: value: .01 * Data[19]
FreqacC: value: .01 * Data[20]
RunMode: value: Data[21]
Yield_Total: value: .1 * pack(Data[22], Data[23])
Yield_Today: value: .1 * Data[24]
feedInPower: value: to_signed32(pack(Data[72], Data[73]))
FeedInEnergy: value: .01 * pack(Data[74], Data[75])
ConsumeEnergy: value: .01 * pack(Data[76], Data[77])
```

jansmets, 2022-12-21:
Works indeed!

adambogocz, 2022-12-21:
Glad to hear that :). I have give access to all mapping tables to @squishykid and I believe that all should get integrated at some point, see #101 for details.

Trilis29, 2022-12-23:
can you give me more sepcifics about this?
`feedInPower: value: to_signed32(pack(Data[72], Data[73]))`

Trilis29, 2022-12-25:
@adambogocz how can I convert this pack function and to_signed32 so simple yaml for home assistant?

adambogocz, 2022-12-27:
There is function in utility to convert 32 bit value to signed 32 bit value, called to_signed32 https://github.com/squishykid/solax/blob/5fdb6d18acd78350b33fd645f69b4eb301167a72/solax/utils.py#L77

It is used for example here https://github.com/squishykid/solax/blob/5fdb6d18acd78350b33fd645f69b4eb301167a72/solax/inverters/x3_hybrid_g4.py#L82

Trilis29, 2022-12-27:
I see where it used, how can I pack it in simple? I would like to use raw data with simple yaml template on home assistant

jansmets, 2023-01-06:
For reference, 

curl -d "optType=ReadRealTimeData&pwd=XXX" -X POST http://192.168.0.191
{"sn":"XXXX","ver":"3.006.04","type":16,"Data":[2271,2271,2276,4,4,8,33,21,0,4782,2629,0,1,1,0,75,30,0,5002,5002,5001,2,296,0,1,8000,24,25,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,65142,65535,1591,0,13532,0,56,0,1,7,6,0,12468,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[8.000,16,"MC20XXXXXX",8,1.18,0.00,1.15,1.01,0.00,1]}

curl -d "optType=ReadSetData&pwd=XXXXX" -X POST http://192.168.0.191
[26,0,0,25600,25700,0,30000,60000,0,0,0,0,0,0,0,0,0,2415,2300,118,0,0,115,101,11047,7946,5644,0,0,1,2645,2875,2875,1955,920,920,1,2530,5150,5200,4750,4700,0,0,2760,1150,60,1,5020,90,1,4980,100,1,900,0,10,10,0,100,2014,0,1,0,1,0,0,0,150,10,300,150,500,10,500,10,1,0,3000,5010,0,2530,2553,2576,2576,0,200,50,2139,2231,2369,2461,23134,200,500,800,1000,1,0,1,5010,0,50,4985,0,50,60,1000,500,0,0,435,0,0,65101,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]

eg, field 0 is the 'safety' setting for the inverter, field 7 is an export control setting, etc.
The "2014" is the standard inverter admin pin.

There are also the following commands - not sure how they're exactly used.
login_bd
SelfTestStart
SelfTestRequest
ReadRealTimeData
ReadSetData
setReg
setNet
setRTC
newParaSetting

jansmets, 2023-01-06:
Can we get support in for X3 G2 please?

Trilis29, 2023-01-16:
@adambogocz can we get help? 

I still feel feedinpower is wrong

horamarques, 2023-03-14:
any update n this issue? is there a branch to test this device?
Thanks

horamarques, 2023-06-29:
Hi all,

tested version 0.3.2 updating HA manually and its working fine.
Just need to wait for the HA update.

thank you all.

closed 2023-06-29 by horamarques