squishykid/solax #86: Suddenly not able to connect

Puntoboy, 2022-09-05:
I noticed this morning that all of a sudden HA can't communicate with my Solax Inverter (X1-Boost-3.6). It's been working find for the last week without a problem.

Today it suddenly stopped and all identities are unavailable. I tried rebooting everything, reloading the integration and deleting/reinstalling it. No change. I've not even upgraded HA to the latest version. 

This is the error in the logs.

`Logger: homeassistant.components.solax.config_flow
Source: components/solax/config_flow.py:36
Integration: SolaX Power ([documentation](https://www.home-assistant.io/integrations/solax), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+solax%22))
First occurred: 11:05:01 (1 occurrences)
Last logged: 11:05:01

Unexpected exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/solax/inverter.py", line 37, in get_data
    data = await self.make_request(
  File "/usr/local/lib/python3.10/site-packages/solax/inverter.py", line 183, in make_request
    response = cls.schema()(json_response)
  File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: required key not provided @ data['SN']

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/solax/config_flow.py", line 52, in async_step_user
    serial_number = await validate_api(user_input)
  File "/usr/src/homeassistant/homeassistant/components/solax/config_flow.py", line 36, in validate_api
    response = await api.get_data()
  File "/usr/local/lib/python3.10/site-packages/solax/__init__.py", line 51, in get_data
    return await rt_request(self.inverter,
  File "/usr/local/lib/python3.10/site-packages/solax/__init__.py", line 26, in rt_request
    return await inv.get_data()
  File "/usr/local/lib/python3.10/site-packages/solax/inverter.py", line 48, in get_data
    raise InverterError(msg, str(self.__class__.__name__)) from ex
solax.inverter.InverterError: ('Received malformed JSON from inverter', 'XHybrid')`

squishykid, 2022-09-10:
Can you please follow these instructions: https://github.com/squishykid/solax/wiki/DiscoveryError

If this still doesn't work, please follow these and share the response payload in this issue: https://github.com/squishykid/solax/wiki/Adding-Support-for-Your-Inverter

I wonder if your inverter is replying with a different payload, perhaps as a result of a firmware update?

Puntoboy, 2022-09-12:
Thanks. No firmware updates I'm aware of. 

I've tried the curl commands but not luck. 

[image]

I can get to the webpage for the inverter though, and login. 

[image]

squishykid, 2022-09-12:
https://github.com/squishykid/solax/issues/32 This might have some answers for you, but unfortunately I don’t know what password to try. But if it’s `admin`, the equivalent curl command would be:

`curl -X POST http://$hostname/?optType=ReadRealTimeData&pwd=admin&`

Puntoboy, 2022-09-12:
Unfortunately, that doesn't work. To login to the webpage, I have to use a username as well.

[image]

squishykid, 2022-09-12:
Could you please try it with quotes:
`curl -X POST 'http://$hostname/?optType=ReadRealTimeData&pwd=admin&'`

Puntoboy, 2022-09-12:
[image]

squishykid, 2022-09-12:
That’s annoying. I guess you might have to do some digging to find the right URL. When you find it I can make sure it’s programmed into the library.

Puntoboy, 2022-09-12:
I'm just not sure what's changed. It was working perfectly, then it stopped.

squishykid, 2022-09-12:
Perhaps the firmware has been updated remotely? But I think you have tried all the requests the library makes, via curl. You could try reading though some of the other issues?

Ill come back here if I can think of anything, please reply if you solve it too

Puntoboy, 2022-09-12:
It's the way it replies unauthorised, rather than an invalid URL or something similar. 

I'm not aware of Solax applying remote firmware updates, especially without an owner knowing.

Puntoboy, 2022-09-12:
The firmware version for the dongle is Solax_SI_CH_2nd_20160912_DE02

Puntoboy, 2022-09-15:
So not sure what's changed, but I managed to get a response using this command.

[image]

However, Home Assistant still doesn't connect.

Puntoboy, 2022-09-15:
If it makes any difference, I'm using the original Pocket Wifi dongle 1.0 which allowed you to access the API directly when it connects to your network, I don't need to use a reverse proxy to connect.

cazotlan, 2023-02-05:
Bonjour a tous, Nouveau sur GitHub, Je dois mettre en service mon installation solaire dotée d'un Solax-x1-boost 5000 qui est déjà raccordé en RS485 a son compteur "intelligent" DDSU666-CT pour la gestion du zéro injection. D'après Solax cette liaison ne permettrait pas l'ajout d'un 3em équipement (exemple narrow 644/1284 avec interface 485) pour venir lire la table Modbus de l'onduleur (paramètres d'exploitation Kwh, Etc). Je dispose donc par ailleur sur le solax d'un pocket Wifi2 dans l'hypothèse ou les nombreux échanges que j'ai lu sur le forum laisseraient à penser une possible extirpation des données en local (politique Solax du tout Cloud payant mauvaise). Pensez vous que celà soit possible en démarrant avec la fameuse commande (Curl https://5.8.8.8/?optType=ReadRealTimeData) depuis un pc-portable wifi et si celà s'avère probant d'enchainer avec une carte ESP8285 connectée au Narrow mega1284 (port rs232 et commandes at) ? Merci pour vos suggestions évetuelles. Cazotlan

andmagri, 2023-11-22:
Hi, if anyone needs this I managed to curl directly to the wifi dongle after a forced firmware upgrade using the following:

realtime=`curl -s -X POST --header 'X-Forwarded-For:5.8.8.8' 'http://192.168.0.57/?optType=ReadR
ealTimeData&pwd=admin&'` && echo $realtime

Adding the quotes after the pwd argument is what fixed it.

squishykid, 2025-01-18:
Closing for now as this is a very old issue.

closed 2025-01-18 by squishykid