squishykid/solax #88: Add global Support for Firmware v3.003.02

jaykijay, 2022-09-12:
With the newest commit we have the Class InverterPostData available.
This class supports Firmware v3.003.02 of the Pocket Wifi Sticks.

I already (in my fork) sucessfully tried this class with an X1 Mini as i have this available.
As i understand all Inverters with this Stick and Firmware Version will work fine if we change the connection Class to "InverterPostData".

I think the solution should be to globaly decide which way to connect is working (InverterPostData or the "old" InverterPost") and then try the different Inverter Models, as adding every Inverter again, just with the class changed is a bad idea and will result in asyncio Timeouts.

squishykid, 2022-10-09:
Sure! Could you please raise a PR and I'll take a look

kdehairy, 2022-11-15:
@squishykid can you have a look at this PR https://github.com/squishykid/solax/pull/96 ? I'm addressing this ticket here

closed 2022-12-15 by squishykid