squishykid/solax #90: New Wifi dongle support

rupertnash, 2022-09-12:
Various issues/comments suggest that new pocket wifi dongles have to be accessed via a POST with the request in the data rather than URL.
- #75 (now resolved by #77 - thanks Robin! @squishykid)
- #80
- maybe #84 
- https://github.com/squishykid/solax/pull/77#issuecomment-1243974594

This makes the detection harder and perhaps requires the cartesian product of inverters and dongles? 

Issue #80 prompted me to split out the method of access into a base class, but I would expect that the data received from different registers will need a different mapping onto sensor descriptions. One could hope that this will be the same mapping from indexes to meanings for the same model inverter meaning that part can live as a mixin and then the concrete types can just inherit from both `InverterPostData` and that?

I'm afraid I don't have the time right now to contribute but if tagged in a PR I am happy to run a test against my inverter (X1 Hybrid G4 with new wifi dongle - it returns gives version 3.003.02)

jaykijay, 2022-09-13:
See also #88.
The new Method via POST request works for my X1 Mini, too, i patched it myself in my own fork.

squishykid, 2024-03-03:
resolved in 96

closed 2024-03-03 by squishykid