squishykid/solax #93: PocketWifi v3.003.02 + X1 Boost support

lupine, 2022-10-06:
Hi there,

I've had a Solax X1 Boost G3 installed with the new v3 PocketWifi firmware. It's quite similar to the existing one, although there are some differences. I'm successfully scraping data from it with my own code now, but used a lot of hints from this repo, so thought I'd document my findings for when #88 is ready to go.

I've got it connected to a local network, but this works via the wifi network it exposes as well. I did observe it "locking down" to the first IP it saw, but I've not needed to try the `X-Forwarded-For` hack to see if that still works or not yet.

Default username is admin, default password is the serial number (which is also embedded in the wifi AP name :facepalm: ). I've got it listening on `10.0.2.100` below, and am accessing it from `10.0.2.1`. I've annotated the JSON with my best guess for each field:

```sh
$ curl 'http://10.0.2.100' -d 'optType=ReadRealTimeData&pwd=SXxxxxxxxx'
```
<details>

```json
{
  "sn": "SXxxxxxxxx",
  "ver": "3.003.02",
  "type": 4, // Inverter type per https://www.eu.solaxcloud.com/phoebus/resource/files/userGuide/Solax_API.pdf
  "Data": [
    2457, //  0 - AC Voltage
     120, //  1 - AC Output Current
    2942, //  2 - AC Output Power
    2926, //  3 - PV1 Voltage
       0, //  4 - PV2 Voltage
     103, //  5 - PV1 Current
       0, //  6 - PV2 Current
    3020, //  7 - PV1 Power
       0, //  8 - PV2 Power
    4944, //  9 - AC Frequency
       2, // 10 - status per https://www.eu.solaxcloud.com/phoebus/resource/files/userGuide/Solax_API.pdf
     413, // 11 - Total Generated Energy, low 16 bits
       0, // 12 - Total Generated Energy, high 16 bits
      66, // 13 - Today Generated Energy, low 16 bits
       0, // 14 - Today Generated Energy, high 16 bits
       0,
       0,
       0,
       0,
       0,
       0, // 20
       0,
       0,
       0,
       0,
       0,
       0,
       0,
       0,
       0,
       0, // 30
       0,
       0,
       0,
       0,
       0,
       0,
       0,
       0,
      35, // 39 - Inverter Temperature (External)
       0, // 40
      77, // 41 - Inverter runtime?
       0,
       0,
       0,
       0,
       0,
       0,
       0,
       0,
       0, // 50
       0,
       0,
       0,
       0,
      38, // 55 - Inverter Temperature (Internal)
       0,
       0,
       0,
       0,
       0, // 60
       0,
       0,
       0,
       0,
       0,
       0,
       0,
       0,
       0,
       0, // 70
       0,
       0,
       0,
       0,
       0,
       0,
       0,
       0,
       0,
       0, // 80
       0,
       0,
       0,
       0,
       0,
       0,
       0,
       0,
       0,
       0, // 90
       0,
       0,
       0,
       0,
       0,
       0,
       0,
       0,
       0
  ],
  "Information": [
    3,                // 0 - Nominal kW
    4,                // 1 - Status
    "XB32xxxxxxxxxx", // 2 - Inverter serial
    8,
    0,
    0,
    1.4,
    0,
    0,
    1
  ]
}
```
</details>

The inverter has two strings but I only have panels attached to the first, but I inferred the PV2s (and a bunch of others) from the existing `x1_boost.py`.

The "status" field at 10 is rock solid. While the status is 2 ("OK"), the inverter is running and lots of other metrics, including temperature, are reporting. When it's not, it goes to 0 ("Waiting"), 

I have field 41 down as "runtime". I see it increasing by one per hour as long as field 10 (status) is 2. When it's 0, field 41 stops increasing. Its value has no correspondence with generation, import, or export. Here's a graph of it:

<details>

[image]

</details>

The internal temperature value at 55 is a guess, but a convincing one. Here are the two graphed together (internal temp is higher):

<details>

[image]

</details>

Export stats are notable by their absence. They are included in the `x1_mini.py` for v2 firmware, but I see nothing like them :shrug:. I don't have FiT either, so can't check for those values.

I'd love to know what the remaining values in the `Information` field are. Anyone have any idea?

wranglatang, 2022-10-10:
Hey @lupine - I just installed a X1-Hybrid Gen4 which comes with a Pocket WiFi V3 (3.001.02) and I've been scanning the internet to find a way to near realtime values to feed into my homeassistant! It sounds like you have had some great success reading from your inverter/Pocket Wifi. Do you then feed this data into Home Assistant, or other DB. If so, How!?

lupine, 2022-10-10:
@wranglatang I put together a little program that scrapes the solax endpoint and publishes the data to MQTT. From there, I'm getting it into victoriametrics and graphing it in grafana; but home-assistant can read MQTT as well.

You'd probably be better off using https://www.home-assistant.io/integrations/solax/ or so. There's about a million different projects for the different access methods and inverters ^^

wranglatang, 2022-10-10:
Thanks for the info. Would be good to know how you are scraping the data. i might try using Node red - once I get the data in i will put it into HA Sensors no problem! Do you know if its possible to request specific sensors? getting an update for all the sensors takes time - im look to be able to refresh the data every 5 - 10 seconds.

wills106, 2022-10-10:
Be very careful using the Pocket WiFi in AP if you do decide to do so!

https://community.home-assistant.io/t/solax-inverter-by-modbus-no-pocket-wifi-now-a-custom-component/140143/715?u=wills106

lupine, 2022-10-10:
Yeah, it's pretty awful; the installers put it in despite me telling them I wanted luddite mode on everything. I'll be replacing it with a PocketLAN when I can. Took me a few days to realise they'd installed it in the first place, then a few more to realise just how bad it was.

wranglatang, 2022-10-10:
Ive connected my pocket wifi to my IoT Network so it doesn't have access to anything else. My assumption is that the Pocket Wifi stops being in AP mode once its connected to a wifi network? is this not correct?

wills106, 2022-10-10:
I don't have a PocketWiFi / PocketLan so can't comment 100% on the security.
But they are pretty limited devices if you want any control locally.

lupine, 2022-10-10:
Not correct, sadly. It continues to advertise and be accessible as an AP. If one can get the modbus port working on it, that's obviously a worst-case scenario, but the admin interface isn't great either. It's an unsecured AP so polling the interface with the password also means the password is regularly broadcast in cleartext.

PocketLAN is better just because people need to get access to the cabling to do bad things to your inverter.

wills106, 2022-10-10:
Or just buy a Waveshare and have full local control over your Inverter. Probably of limited use for @lupine with your Inverter but @wranglatang would get a lot of benefit from having a Gen4 Hybrid.

wranglatang, 2022-10-10:
Thanks for the input guys! My Inverter is in my garage which is about 50 meters from my house, getting ethernet in there is a Work in Progress - But i definitely want my energy secure (nobody changing to have the batteries charge from grid etc.)

Ill take a look at the Waveshare! I noticed that the X1 Gen 4 has a "LCD/COM" port... is this what i would need to use?

lupine, 2022-10-10:
Yeah, Modbus-RTU is best if you can get it. My inverter does have that too, but it's not well exposed.

Wonder how well powerline would work down the AC cable. I suspect not great :sweat_smile: 

Anyway, I opened this issue so the project owners could confidently add support for this specific inverter+firmware+access method combo. I don't want to derail it beyond that, so best of luck with your respective projects :thumbsup:

wills106, 2022-10-10:
@wranglatang  The Com port yes. Pin 4&5. You don't need Ethernet at your Inverter as such. You can bring the RS485 connection upto 1200m to where a convenient point is.

You will need power for the Waveshare.

@lupine If you are able to connect via RS485 let me know.
It's possible XB3 & XM3 Boost / MIni will work, but not had a massive amount of feedback so far.

wranglatang, 2022-10-10:
@wills106 Thanks. Ive ordered one of the waveshares & will set that up when it arrives. I will report back on your [homeassistant-solax-modbus](https://github.com/wills106/homeassistant-solax-modbus) git.

Side question - are you aware of what all the items are in the returned Data [] json?

lupine, 2022-10-11:
@wills106 out of curiosity I gave your solax-modbus home-assistant client a try against the port 502 that's open on the pocketwifi interface.

```
2022-10-11 14:01:57.980 WARNING (MainThread) [custom_components.solax_modbus] reading serial number from address 0 failed; other address may succeed
2022-10-11 14:02:02.987 WARNING (MainThread) [custom_components.solax_modbus] reading serial number from address 768 failed; other address may succeed
2022-10-11 14:02:02.988 ERROR (MainThread) [custom_components.solax_modbus] cannot find serial number, even not for MIC
2022-10-11 14:02:02.990 ERROR (MainThread) [custom_components.solax_modbus] unrecognized inverter type - serial number : unknown
```

I got this for modbus addresses 0 and 1.

I was also unable to get anything out of the port via `mbpoll`.It's definitely *listening* on the port, but without a response of some kind, I can't rule out that maybe it's just not connected to anything.

If that is the case, then changing the admin password and never sending it over the unsecured AP might be sufficient security, although it's still unpleasant to have the AP present.

I'm in correspondence with Solax at the moment about this port 502; perhaps they'll have something useful to say.

wills106, 2022-10-11:
If you connect to the PocketWiFi with it in AP Mode, they you can connect to it via Modbus.
But when it's connected to a Network you can't.

I would still be reluctant to use a PocketWiFi when the password is built into the SSID. I assume you can change the password, but it makes you wonder what other security risks are built into it.

theater, 2023-04-10:
> @wills106 out of curiosity I gave your solax-modbus home-assistant client a try against the port 502 that's open on the pocketwifi interface.
> 
> ```
> 2022-10-11 14:01:57.980 WARNING (MainThread) [custom_components.solax_modbus] reading serial number from address 0 failed; other address may succeed
> 2022-10-11 14:02:02.987 WARNING (MainThread) [custom_components.solax_modbus] reading serial number from address 768 failed; other address may succeed
> 2022-10-11 14:02:02.988 ERROR (MainThread) [custom_components.solax_modbus] cannot find serial number, even not for MIC
> 2022-10-11 14:02:02.990 ERROR (MainThread) [custom_components.solax_modbus] unrecognized inverter type - serial number : unknown
> ```
> 
> I got this for modbus addresses 0 and 1.
> 
> I was also unable to get anything out of the port via `mbpoll`.It's definitely _listening_ on the port, and I can establish a TCP connection, but without a response of some kind, I can't rule out that maybe it's just not connected to anything.
> 
> If that is the case, then changing the admin password and never sending it over the unsecured AP might be sufficient security, although it's still unpleasant to have the AP present.
> 
> I'm in correspondence with Solax at the moment about this port 502; perhaps they'll have something useful to say.

@lupine Have you got any meaningful response from Solax about this port 502? Have you managed to change the password? (I'm still reluctant to do so due to the unforseen consequences) :) Any more findings on the array of data? Has anyone reverse engineered the values or we have to do a guess work? :)

Cheers,
Konstantin

wills106, 2023-04-10:
@theater Do you want to raise an issue on my GitHub, as this is a different integration you are posting on compared to mine.

Also you need your PocketWiFi updating to V3.004.03 assuming you have a PocketWiFi v3

lupine, 2023-04-10:
@theater no, they were unhelpful so I replaced the inverter. It's now sat in a box in my garage :sweat_smile:

squishykid, 2025-01-18:
Closing this for now as it seems to me like this was addressed by #135 . Can you please comment and re-open if this was not the case? :)

closed 2025-01-18 by squishykid