squishykid/solax #101: Lookup tables for every(most) Solax Inverter

adambogocz, 2022-12-20:
Hi @squishykid,
I have lookup tables to decode data for following models:

- X1-Hybiyd/Fit
- X1-Boost/Air/Mini
- X3-Hybiyd/Fit
- X3-20K/30K
- X3-MIC/PRO
- X1-Smart
- X1-AC
- A1-Hybrid
- A1-Fit
- A1-Grid
- J1-ESS
- X3-Hybrid-G4
- X1-Hybrid-G4
- X3-MIC/PRO-G2
- X1-SPT
- X1-Boost/Mini-G4
- A1-HYB-G2
- A1-AC-G2
- A1-SMT-G2
- X3-FTH
- X3-MGA-G2
- "Inverter with type 24"
- "Inverter with type 100"
- "Inverter with type 101"
- "Inverter with type 102"

Let me know if you are interested in that, I didn't wanted to just dump it all here if you didn't wanted it.

squishykid, 2022-12-20:
Yes, I am currently looking at how to refactor the library to make it much easier to add and manage inverters. If you could share your findings here I will be able to add them in. Thank you for reaching out!

adambogocz, 2022-12-20:
You are welcome :)

```
3 = {
    Yield_Today: { value: .1 * bit16(1, t[13])},
    Yield_Total: { value: .1 * bit32(1, t[11], t[12])},
    PowerDc1: { value: bit16(1, t[7])},
    PowerDc2: { value: bit16(1, t[8])},
    BAT_Power: { value: bit16(0, t[18])},
    feedInPower: { value: bit32(0, t[48], t[49])},
    GridAPower: { value: bit16(0, t[2])},
    FeedInEnergy: { value: .01 * bit32(1, t[50], t[51])},
    ConsumeEnergy: { value: .01 * bit32(1, t[52], t[53])},
    RunMode: { value: bit16(1, t[10])},
    EPSAPower: { value: bit16(0, t[43])},
    Vdc1: { value: .1 * bit16(1, t[3])},
    Vdc2: { value: .1 * bit16(1, t[4])},
    Idc1: { value: .1 * bit16(1, t[5])},
    Idc2: { value: .1 * bit16(1, t[6])},
    EPSAVoltage: { value: .1 * bit16(1, t[44])},
    EPSACurrent: { value: .1 * bit16(0, t[45])},
    BatteryCapacity: { value: bit16(1, t[20])},
    BatteryVoltage: { value: .01 * bit16(1, t[16])},
    BatteryTemperature: { value: bit16(0, t[19])},
    GridAVoltage: { value: .1 * bit16(1, t[0])},
    GridACurrent: { value: .1 * bit16(0, t[1])},
    FreqacA: { value: .01 * bit16(1, t[9])}
},
4 = {
    Yield_Today: { value: .1 * bit16(1, t[13])},
    Yield_Total: { value: .1 * bit32(1, t[11], t[12])},
    PowerDc1: { value: bit16(1, t[7])},
    PowerDc2: { value: bit16(1, t[8])},
    feedInPower: { value: bit32(0, t[48], t[49])},
    GridAPower: { value: bit16(0, t[2])},
    FeedInEnergy: { value: .01 * bit32(1, t[50], t[51])},
    ConsumeEnergy: { value: .01 * bit32(1, t[52], t[53])},
    RunMode: { value: bit16(1, t[10])},
    Vdc1: { value: .1 * bit16(1, t[3])},
    Vdc2: { value: .1 * bit16(1, t[4])},
    Idc1: { value: .1 * bit16(1, t[5])},
    Idc2: { value: .1 * bit16(1, t[6])},
    GridAVoltage: { value: .1 * bit16(1, t[0])},
    GridACurrent: { value: .1 * bit16(0, t[1])},
    FreqacA: { value: .01 * bit16(1, t[9])}
},
5 = {
    Yield_Today: { value: .1 * bit16(1, t[21])},
    Yield_Total: { value: .1 * bit32(1, t[19], t[20])},
    PowerDc1: { value: bit16(1, t[13])},
    PowerDc2: { value: bit16(1, t[14])},
    BAT_Power: { value: bit16(0, t[26])},
    feedInPower: { value: bit32(0, t[65], t[66])},
    GridAPower: { value: bit16(0, t[6])},
    GridBPower: { value: bit16(0, t[7])},
    GridCPower: { value: bit16(0, t[8])},
    FeedInEnergy: { value: .01 * bit32(1, t[67], t[68])},
    ConsumeEnergy: { value: .01 * bit32(1, t[69], t[70])},
    RunMode: { value: bit16(1, t[18])},
    EPSAPower: { value: bit16(0, t[60])},
    EPSBPower: { value: bit16(0, t[61])},
    EPSCPower: { value: bit16(0, t[62])},
    Vdc1: { value: .1 * bit16(1, t[9])},
    Vdc2: { value: .1 * bit16(1, t[10])},
    Idc1: { value: .1 * bit16(1, t[11])},
    Idc2: { value: .1 * bit16(1, t[12])},
    EPSAVoltage: { value: .1 * bit16(1, t[54])},
    EPSBVoltage: { value: .1 * bit16(1, t[55])},
    EPSCVoltage: { value: .1 * bit16(1, t[56])},
    EPSACurrent: { value: .1 * bit16(0, t[57])},
[... 482 lines cut ...]
    PowerDc2:{ value: bit16(1, t[10])}
    BAT_Power:{ value: bit16(0, t[22])}
    feedInPower:{ value: bit32(0, t[29], t[30])}
    GridAPower:{ value: bit16(0, t[2])}
    GridBPower:{ value: 0}
    GridCPower:{ value: 0}
    FeedInEnergy:{ value: .01 * bit32(1, t[31], t[32])}
    ConsumeEnergy:{ value: .01 * bit32(1, t[33], t[34])}
    RunMode:{ value: bit16(1, t[12])}
    EPSAPower:{ value: bit16(0, t[43])}
    Vdc1:{ value: .1 * bit16(1, t[5])}
    Vdc2:{ value: .1 * bit16(1, t[7])}
    Idc1:{ value: .1 * bit16(1, t[6])}
    Idc2:{ value: .1 * bit16(1, t[8])}
    EPSAVoltage:{ value: .1 * bit16(1, t[41])}
    EPSACurrent:{ value: .1 * bit16(0, t[42])}
    BatteryCapacity:{ value: bit16(1, t[18])}
    BatteryVoltage:{ value: .1 * bit16(1, t[20])}
    BatteryTemperature:{ value: bit16(0, t[17])}
    GridAVoltage:{ value: .1 * bit16(1, t[0])}
    GridACurrent:{ value: .1 * bit16(0, t[1])}
    FreqacA:{ value: .01 * bit16(1, t[3])}
}

```

adambogocz, 2022-12-20:
I also have conversion table for each run mode code of each inverter type

squishykid, 2022-12-20:
Please share it all.

BTW, can I contact you by email? I have some questions

adambogocz, 2022-12-20:
Yes, its adam.bogocz@seznam.cz

adambogocz, 2022-12-20:
And here is the run mode table

```
3 = {
    0: "Wait Mode",
    1: "Check Mode",
    2: "Normal Mode",
    3: "Fault Mode",
    4: "Permanent Fault Mode",
    5: "Update Mode",
    6: "EPS check mode",
    7: "EPS mode",
    8: "Self Test",
    9: "Idle Mode",
    10: "Standby Mode"
},
4 = {
    0: "Wait Mode",
    1: "Check Mode",
    2: "Normal Mode",
    3: "Fault Mode",
    4: "Permanent Fault Mode",
    5: "Update Mode"
},
5 = {
    0: "Wait Mode",
    1: "Check Mode",
    2: "Normal Mode",
    3: "Fault Mode",
    4: "Permanent Fault Mode",
    5: "Update Mode",
    6: "EPS check mode",
    7: "EPS mode",
    8: "Self Test",
    9: "Idle Mode",
    10: "PvWakeUpBat Mode",
    11: "Standby Mode"
},
6 = {
    0: "Wait Mode",
    1: "Check Mode",
    2: "Normal Mode",
    3: "Fault Mode",
    4: "Permanent Fault Mode",
    5: "Update Mode"
},
7 = {
    0: "Wait Mode",
    1: "Check Mode",
    2: "Normal Mode",
    3: "Fault Mode",
    4: "Permanent Fault Mode",
    5: "Update Mode"
},
8 = {
    0: "Wait Mode",
    1: "Check Mode",
    2: "Normal Mode",
    3: "Fault Mode",
    4: "Permanent Fault Mode",
    5: "Update Mode"
},
9 = {
    0: "Wait Mode",
    1: "Check Mode",
    2: "Normal Mode",
    3: "Fault Mode",
    4: "Permanent Fault Mode",
    5: "Update Mode",
    8: "Self Test",
    9: "Idle Mode",
    10: "Standby Mode"
},
10,11,12 = {
    0: "Wait Mode",
    1: "Check Mode",
    2: "Normal Mode",
    3: "Fault Mode",
    4: "Permanent Fault Mode",
    5: "Update Mode",
    6: "EPS check mode",
    7: "EPS mode",
    8: "Self Test",
    9: "Idle Mode",
    10: "Gen Check Mmode",
    11: "Gen Run Mode",
    12: "RSD Standby"
},
13 = {
    0: "Wait Mode",
    1: "Check Mode",
    2: "Normal Mode",
    3: "Fault Mode",
    4: "Permanent Fault Mode",
    5: "Update Mode",
    6: "EPS check mode",
    7: "EPS mode",
    8: "Self Test",
    9: "Idle Mode",
    10: "Standby Mode"
},
14 = {
    0: "Wait Mode",
    1: "Check Mode",
    2: "Normal Mode",
    3: "Fault Mode",
    4: "Permanent Fault Mode",
    5: "Update Mode",
    6: "EPS check mode",
    7: "EPS mode",
    8: "Self Test",
    9: "Idle Mode",
    10: "Standby Mode"
},
15 = {
    0: "Waiting",
    1: "Checking",
    2: "Normal",
    3: "Fault",
    4: "Permanent Fault",
    5: "Upgrade",
    6: "EPS Waiting",
    7: "EPS",
    8: "Self Testing",
    9: "Idle",
    10: "Standby"
},
16 = {
    0: "Wait Mode",
    1: "Check Mode",
    2: "Normal Mode",
    3: "Fault Mode",
    4: "Permanent Fault Mode",
    5: "Update Mode"
},
17 = {
    0: "Wait Mode",
    1: "Check Mode",
    2: "Normal Mode",
    3: "Fault Mode",
    4: "Permanent Fault Mode",
    5: "Update Mode",
    6: "EPS check mode",
    7: "EPS mode",
    8: "Self Test",
    9: "Idle Mode",
    10: "Standby Mode"
},
18,22 = {
    0: "Wait Mode",
    1: "Check Mode",
    2: "Normal Mode",
    3: "Fault Mode",
    4: "Permanent Fault Mode",
    5: "Update Mode",
    6: "EpsCheckMode",
    7: "EpsMode"
},
19,20,21 = {
    0: "Wait Mode",
    1: "Check Mode",
    2: "Normal Mode",
    3: "Fault Mode",
    4: "Permanent Fault Mode",
    5: "Update Mode",
    6: "EPS Check Mode",
    7: "EPS Mode",
    8: "Self Test",
    9: "IdleMode",
    10: "StandbyMode",
    11: "Gen Check Mode",
    12: "Gen Run Mode",
    13: "RSD Standby"
},
100,101 = {
    0: "InitMode",
    1: "IdleMode",
    2: "StartMode",
    3: "RunMode",
    4: "FaultMode",
    5: "UpdateMode"
},
24 = {
    0: "Wait Mode",
    1: "Check Mode",
    2: "Normal Mode",
    3: "Fault Mode",
    4: "Permanent Fault Mode",
    5: "Update Mode",
    6: "EPS check mode",
    7: "EPS mode",
    8: "Self Test",
    9: "Idle Mode",
    10: "Standby Mode"
},
23 = {
    0: "Waiting",
    1: "Checking",
    2: "Normal",
    3: "Fault",
    4: "Permanent Fault",
    5: "Upgrade",
    6: "EPS Waiting",
    7: "EPS",
    8: "Self Testing",
    9: "Idle",
    10: "Standby"
}
```

squishykid, 2022-12-20:
> ConsumeEnergy: { value: .01 * bit32(1, t[36], t[37])},

Could you also please share which inverter name is which type in the list?

adambogocz, 2022-12-20:
I haven got that mapping from the same source, but it can be found here https://www.solaxcloud.com/user_api/SolaxCloud_User_Monitoring_API_V6.1.pdf