squishykid/solax #104: Qcells ESS HYB-G3 not supported

emichel11, 2023-01-03:
I have a Qcells ESS HYB-G3 which is not supported.
A query gives 300 elements
curl -X POST http://192.123.456.789/ --data "optType=ReadRealTimeData&pwd=XXXXXXXXXXX"
{"sn":"xxxxxxxxxx","ver":"3.003.02","type":14,"Data":[2290,2289,2297,0,0,0,0,0,0,0,0,0,0,0,0,0,5002,5001,5002,10,0,0,0,0,0,0,0,0,0,0,0,0,0,1,64901,65535,0,0,0,20760,0,0,2066,65535,65516,1,34,635,256,11547,4885,5644,100,0,21,0,0,0,0,0,0,0,0,0,0,0,0,0,12,0,1,0,0,0,10,0,0,0,0,0,3,0,1,0,0,0,0,0,769,0,0,0,689,0,0,0,0,0,0,0,0,0,1,28,1,21,17,256,2336,1600,130,234,148,136,32,32,1,1108,778,14649,14649,0,0,0,0,3235,3227,13474,0,20564,12339,18753,12341,18741,12868,12598,20564,12339,18754,12856,18738,13382,14648,20564,12339,18754,12856,18738,13638,13873,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1793,4097,1537,514,0,20760,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[6.000,14,"YYYYYYYYYYYYY",8,1.23,0.00,1.24,1.09,0.00,1]}%

There is a Qcell entry QCELLHYP3 but this defines only 200 elements. Following change works
diff --git a/solax/inverters/qvolt_hyb_g3_3p.py b/solax/inverters/qvolt_hyb_g3_3p.py
index f686d3e..44a320c 100644
--- a/solax/inverters/qvolt_hyb_g3_3p.py
+++ b/solax/inverters/qvolt_hyb_g3_3p.py
@@ -56,7 +56,7 @@ class QVOLTHYBG33P(Inverter):
             vol.Required("Data"): vol.Schema(
                 vol.All(
                     [vol.Coerce(float)],
-                    vol.Length(min=200, max=200),
+                    vol.Length(min=200, max=300),
                 )
             ),
             vol.Required("Information"): vol.Schema(

VadimKraus, 2023-03-12:
this seems to be a newer firmware? 
The current sample was  "ver": "2.034.06" and I can confirm it still works for my "Qcells ESS HYB-G3"

ifrh, 2023-10-06:
For the record:
The changes mentioned above in https://github.com/squishykid/solax/issues/104#issue-1517763728  seem to be the same as mentioned in https://github.com/squishykid/solax/issues/85#issuecomment-1228881287

VadimKraus, 2023-10-29:
Ist this still an issue? Looking at the code,  I guess it should be recognized as

emichel11, 2025-03-08:
Yes - it is recognized with "X3 Hybrid G4" with latest HEAD 450d41b.
It works fine for me - the data I need are catched- we shall closed the issue

closed 2025-03-08 by emichel11