squishykid/solax #105: extend class method "async def real_time_api(ip_address, port=80, pwd="")" with inverter

emichel11, 2023-01-03:
The discover process takes some time or - in my case does not work properly
Actually the method
async def real_time_api(ip_address, port=80, pwd=""):
    i = await discover(ip_address, port, pwd)
    return RealTimeAPI(i)

calls discover and this method scans all available classes in regestry
async def discover(host, port, pwd="") -> Inverter:
    failures = []
    for inverter in REGISTRY:

It shall be possible to predefine the INVERTER for real_time_api /  discover method. This will speedup 
the process for known inverters

MatthiasFraass, 2023-02-15:
Agreed.

Meanwhile this is the code for direct use of a known inverter:

```
import solax
import asyncio

from solax.inverters import QVOLTHYBG33P
from solax import RealTimeAPI

async def work():
    i = QVOLTHYBG33P._build('IP_OF_WIFI_DONGLE', 80, 'SERIAL_NUMER_OF_WIFI_DONGLE')
    r = RealTimeAPI(i)
    return await r.get_data()

loop = asyncio.new_event_loop()
asyncio.set_event_loop(loop)
data = loop.run_until_complete(work())
print(data)
```

emichel11, 2025-03-08:
lastest HEAD has improved speed

closed 2025-03-08 by emichel11