squishykid/solax #109: X3-MIC/PRO returns "Connection refused" after 2000+ms (via LAN)

ramondeklein, 2023-01-30:
I have the Solax X3-MIC/PRO (5kW) inverter and it uses the LAN adapter (Pocket LAN 2.0). The cloud connection is fine and I am able to ping the converter on 192.168.1.60. It also connects properly to the cloud and I am able to access it via Solax Cloud.

```
$ curl -v -d "optType=ReadRealTimeData&pwd=SP2JXXXX4F" -X POST http://192.168.1.60
curl: (7) Failed to connect to 192.168.1.60 port 80 after 2028 ms: Connection refused
```

I read that the WIFI dongle doesn't expose the API on the local network. I did use a WIFI dongle for two weeks before I got my LAN dongle. Maybe I need to re-enable something?

nazar-pc, 2023-02-02:
Wi-Fi does expose API on local network and works just fine, but maybe not in 100% of cases (hard to tell, manufacturer doesn't disclose that kind of information).

Mobile application doesn't support Pocket LAN as far as I can tell from reaching Android application's source code, so it may physically be unable to return you that data and with firmware being closed source we may not be able to do anything about it.

letsjustfixit, 2023-04-16:
@ramondeklein 
Try it this way (replace YOURIP with your inverter's ip: YOURPASSWORD with your admin password by default: admin)
```
curl -X POST http://YOURIP  -H 'X-Forwarded-For: "5.8.8.8"' --data 'optType=ReadSetData&pwd=YOURPASSWORD '
```

mrichtarsky, 2024-10-29:
The Pocket LAN dongle is not listening on any ports, i.e. there is no webserver running at all. That's why you get `Connection refused`. Only the Wifi models are supported by this module.

mrichtarsky, 2024-10-29:
The Pocket LAN dongle is not listening on any ports, i.e. there is no webserver running at all. That's why you get `Connection refused`. Only the Wifi models are supported by this module. As @nazar-pc is mentioned this also becomes clear by looking at the Android application, which offers a "Local Access" mode, but explicitly mentions that it works only with a Wifi dongle.

ramondeklein, 2024-10-29:
It's completely insane that you cannot read your inverter using a wired connection, but you can using a WIFI connection. The cloud connection works fine, but I don't like to have my inverter connected to the internet at all.

nazar-pc, 2024-10-29:
We're all in the same boat with you. I just blocked Internet access for the WiFi dongle on the router and collect data locally with Home Assistant and have WireGuard VPN to local instance so I can still access it remotely if needed.

I can recommend pinging Solax support and letting them know that this is something people actually want. So far their response was to use Modbus that they provide documentation for. Querying already available JSON endpoint is so much more convenient though.

mrichtarsky, 2024-10-29:
> It's completely insane that you cannot read your inverter using a wired connection, but you can using a WIFI connection. The cloud connection works fine, but I don't like to have my inverter connected to the internet at all.

You could use a wired Modbus connection (with a RS485 to TCP adapter). However, I didn't manage to retrieve any values via Modbus using the Wifi dongle (which should be the same mechanism). Not sure what I'm doing wrong, I've opened a ticket with Solax.

ramondeklein, 2024-10-29:
I just bought a new home, so I won't invest in the Solax anymore. Next time I'll be more careful if I ever need to buy another inverter again (or home battery). No local connection, then no buy...