squishykid/solax #112: "Platform solax does not generate unique IDs." for X3 Hybrid G4 due to duplicate run mode on idx 19

jurahudak, 2023-04-14:
Hi.

I'm experiencing the error below, causing stop on processing fully all the returned data = only data until index 19 are stored and displayed. "Run mode" showing 2.0.

`2023-04-14 16:57:03.789 ERROR (MainThread) [homeassistant.components.sensor] Platform solax does not generate unique IDs. ID XXX-19 already exists - ignoring sensor.solax_XXX_run_mode
`

Leaving only one of the rows in the inverters/x3_hybrid_g4.py file solves the issue.
```python
            "Run mode": (19, Units.NONE),
            "Run mode text": (19, Units.NONE, X3HybridG4._decode_run_mode),
``` 

In the Solax overview there is "Run mode" even if I leave "Run mode text" in the file.

Guess, either remove one of the lines, preferable the raw value line. Or fix the ID generation for this case.

squishykid, 2023-06-10:
I think this is something we will need to fix on our side, as well as adding a test.

squishykid, 2025-01-18:
Fixed in #149

closed 2025-01-18 by squishykid