squishykid/solax #151: Possible method to access API on LAN

andmcg1, 2024-04-23:
I'm sorry if this has been mentioned elsewhere. I have only tried this on my own device (Pocket WIFI firmware 2.034.06), so not sure if it helps with later firmware versions.

I read all about needing a reverse proxy set up via a Raspberry Pi to access the API locally, but I thought surely the web server security is not very good based on how insecure the rest of the device is. So I decided to try this on my device:

`curl -d "?optType=ReadRealTimeData&pwd=admin" -X POST 'http://192.168.99.8/'
curl: (52) Empty reply from server
`

But if you trick it and tell it you are requesting a different website, it gets data even via the LAN (wifi)
`curl -d "?optType=ReadRealTimeData&pwd=admin" -X POST 'http://192.168.99.8/' -H "Host: 5.8.8.8"
{"sn":"XXX","ver":"2.034.06","type":4,"Data":[2429,88,2119,3461,3457,32,32,1120,1125,5002,2,10038,5,10,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,8,0,0,0,0,0,8,0,0,20,0,19247,0,0,0,0,0,0,1533,0,0,0,0,0,0,23,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],         "Information":[5.000,4,"XXX",1,3.31,1.09,1.21,0.00,0.00,1]}`

DanielWeeber, 2024-06-24:
Yeah, better even if u use "X-Forwarded-For" instead of "Host". Works either way though.

squishykid, 2025-01-18:
We set this header in the library now, so this happens automagically :)

I understand that we should remove this note from the home-assistant frontend docs though.

Thanks for bringing this to our attention!

squishykid, 2025-01-18:
For example here is where we set the header: https://github.com/squishykid/solax/blob/master/solax/inverters/x1_boost.py#L73

closed 2025-01-18 by squishykid