squishykid/solax #158: Error installing solax package under WinPython

petgriffin, 2024-06-05:
Hi

I am trying to install the solax package on my PC which is running Python via Winpython.  The command line command is:

c:\WinPython\WPy64-31230\python-3.12.3.amd64\Scripts>pip install C:\Users\alun.griffiths\Downloads\solax-3.1.0.tar.gz --use-pep517

The argument --use-pep517 was suggested by pip after I tried a pip install without it, whcih failed.  The output from the above command is given below.  No idea how to fix this so would appreciate any suggestions.  Many thanks!

Alun
------------------------------------------------------------------
Processing c:\users\alun.griffiths\downloads\solax-3.1.0.tar.gz
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [47 lines of output]
      WARNING setuptools_scm.pyproject_reading toml section missing 'pyproject.toml does not contain a tool.setuptools_scm section'
      Traceback (most recent call last):
        File "C:\Users\alun.griffiths\AppData\Local\Temp\pip-build-env-c5h5fk85\normal\Lib\site-packages\setuptools_scm\_integration\pyproject_reading.py", line 36, in read_pyproject
          section = defn.get("tool", {})[tool_name]
                    ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^
      KeyError: 'setuptools_scm'
      Traceback (most recent call last):
        File "C:\WinPython\WPy64-31230\python-3.12.3.amd64\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
          main()
        File "C:\WinPython\WPy64-31230\python-3.12.3.amd64\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\WinPython\WPy64-31230\python-3.12.3.amd64\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 149, in prepare_metadata_for_build_wheel
          return hook(metadata_directory, config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\alun.griffiths\AppData\Local\Temp\pip-build-env-c5h5fk85\overlay\Lib\site-packages\setuptools\build_meta.py", line 366, in prepare_metadata_for_build_wheel
          self.run_setup()
        File "C:\Users\alun.griffiths\AppData\Local\Temp\pip-build-env-c5h5fk85\overlay\Lib\site-packages\setuptools\build_meta.py", line 487, in run_setup
          super().run_setup(setup_script=setup_script)
        File "C:\Users\alun.griffiths\AppData\Local\Temp\pip-build-env-c5h5fk85\overlay\Lib\site-packages\setuptools\build_meta.py", line 311, in run_setup
          exec(code, locals())
        File "<string>", line 6, in <module>
        File "C:\Users\alun.griffiths\AppData\Local\Temp\pip-build-env-c5h5fk85\overlay\Lib\site-packages\setuptools\__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\alun.griffiths\AppData\Local\Temp\pip-build-env-c5h5fk85\overlay\Lib\site-packages\setuptools\_distutils\core.py", line 146, in setup
          _setup_distribution = dist = klass(attrs)
                                       ^^^^^^^^^^^^
        File "C:\Users\alun.griffiths\AppData\Local\Temp\pip-build-env-c5h5fk85\overlay\Lib\site-packages\setuptools\dist.py", line 307, in __init__
          _Distribution.__init__(self, dist_attrs)
        File "C:\Users\alun.griffiths\AppData\Local\Temp\pip-build-env-c5h5fk85\overlay\Lib\site-packages\setuptools\_distutils\dist.py", line 284, in __init__
          self.finalize_options()
        File "C:\Users\alun.griffiths\AppData\Local\Temp\pip-build-env-c5h5fk85\overlay\Lib\site-packages\setuptools\dist.py", line 659, in finalize_options
          ep(self)
        File "C:\Users\alun.griffiths\AppData\Local\Temp\pip-build-env-c5h5fk85\overlay\Lib\site-packages\setuptools\dist.py", line 679, in _finalize_setup_keywords
          ep.load()(self, ep.name, value)
        File "C:\Users\alun.griffiths\AppData\Local\Temp\pip-build-env-c5h5fk85\normal\Lib\site-packages\setuptools_scm\_integration\setuptools.py", line 103, in version_keyword
          _assign_version(dist, config)
        File "C:\Users\alun.griffiths\AppData\Local\Temp\pip-build-env-c5h5fk85\normal\Lib\site-packages\setuptools_scm\_integration\setuptools.py", line 58, in _assign_version
          _version_missing(config)
        File "C:\Users\alun.griffiths\AppData\Local\Temp\pip-build-env-c5h5fk85\normal\Lib\site-packages\setuptools_scm\_get_version_impl.py", line 117, in _version_missing
          raise LookupError(
      LookupError: setuptools-scm was unable to detect version for C:\Users\alun.griffiths\AppData\Local\Temp\pip-req-build-hx4h9kp3.

      Make sure you're either building from a fully intact git repository or PyPI tarballs. Most other sources (such as GitHub's tarballs, a git checkout without the .git folder) don't contain the necessary metadata and will not work.

      For example, if you're using pip, instead of https://github.com/user/proj/archive/master.zip use git+https://github.com/user/proj.git#egg=proj
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

squishykid, 2024-06-05:
Hello, thank you for putting such a detailed bug report together! Unfortunately I am not really sure what the issue is here, and we generally only build this library for linux.

If you do figure out how to resolve this issue, I would be happy to review a PR though!

For now I will close this issue because it is for an version of python we don't test for.

closed 2024-06-05 by squishykid

petgriffin, 2024-06-05:
No Worries - thanks!  Will try to install solax on my RPi and see how I get on

Darsstar, 2024-06-05:
Try downloading and then installing the [.whl](https://files.pythonhosted.org/packages/3a/43/b5351fa2c3a9b00436d6bb2ee1cc3e32882abbb0f445784457855aad62c0/solax-3.1.1-py3-none-any.whl) file instead of the [.tar.gz](https://files.pythonhosted.org/packages/5e/dd/ae30e97aa0d19de807305b4db585fd5157d7eb26c3554ff9826fccc41f48/solax-3.1.1.tar.gz).
That way you skip having to build the wheel locally, before installing the wheel.

petgriffin, 2024-06-05:
Hi
installing the .whl file seemed to work so I ran the script on the github home page to see if the installation was OK.  It seems to work but unfortunately, my inverter seems not to be supported.  I will try the curl commands on the Wiki page and see if they produce any output.