squishykid/solax #169: Solax X1 boost does not provide identity id sensor.solax_swxxxxxxxx_exported_power anymore

japeral, 2024-07-02:
I have automations in Home Assistant to engage loads when exported power is below 100W. My solax is in never injecting power to the grid.

I have modified the firmware of my Wifi dongle to expose the API in my home wifi. I can curl the IP of the wifi dongle and I get this:
[image]

Since few weeks ago, I have seen intermittent issues where my automations does not trigger. I have been updating HA and the Solax Integration when they were available ( I shouldn't ).

I just uninstalled the Solax Integration, and installed it again, but the identity id "sensor.solax_swgxxxxxxx_exported_power" is not available anymore.

1. Is this a known issue? 
2. How can I revert to an older version of the Solax Integration ?

Thank you.

japeral, 2024-07-02:
[image]

germanotero, 2024-09-07:
Same thing here.. it looks like it takes it ok the first time, but then after a reboot it decides we have a different inverter and defaults to another version...

japeral, 2024-10-08:
I do not know how, but it healed automatically for a while, but I I am suffering this again entity sensor.solax_swxxxxxxxx_exported_power is gone again, and my automations fails...

This is the log:

`2024-10-08 11:48:41.498 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-10-08 11:48:41.502 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solarman which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-10-08 11:48:41.506 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration sonoff which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-10-08 11:48:41.510 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration localtuya which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-10-08 11:48:41.514 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration pid_controller which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-10-08 11:49:08.281 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for sensor.energy_production_today (<class 'homeassistant.components.forecast_solar.sensor.ForecastSolarSensorEntity'>) took 0.544 seconds. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+forecast_solar%22
2024-10-08 11:49:08.499 WARNING (SyncWorker_1) [homeassistant.components.rpi_power.binary_sensor] Under-voltage was detected. Consider getting a uninterruptible power supply for your Raspberry Pi.
2024-10-08 11:49:09.220 ERROR (MainThread) [homeassistant.components.automation] Automation with alias 'Todos los Microinversores Off (Imported power > 50W 2min)' failed to setup triggers and has been disabled: Unknown entity '958fe7acb8af68b1eb5c088f44cb3fe4'
2024-10-08 11:49:09.222 ERROR (MainThread) [homeassistant.components.automation] Automation with alias 'Todos los Microinversores On (Imported power < 0W 1min)' failed to setup triggers and has been disabled: Unknown entity '958fe7acb8af68b1eb5c088f44cb3fe4'
2024-10-08 11:49:10.415 DEBUG (MainThread) [homeassistant.components.solax] Finished fetching solax SWxxxxxxxx data in 3.436 seconds (success: True)
2024-10-08 11:49:10.444 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.solax_swxxxxxxxx_today_s_energy (<class 'homeassistant.components.solax.sensor.InverterSensorEntity'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total', 'total_increasing'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+solax%22
2024-10-08 11:49:12.001 WARNING (MainThread) [homeassistant.components.homeassistant.triggers.numeric_state] Error initializing 'Minero1 Off (imported below -300W 1m & Minero2-Off 1m)' trigger: In 'numeric_state' condition: unknown entity sensor.solax_swxxxxxxxx_exported_power
2024-10-08 11:49:12.003 WARNING (MainThread) [homeassistant.components.homeassistant.triggers.numeric_state] Error initializing 'Minero2 Off (imported below -300W 1m & Minero3-Off 1m)' trigger: In 'numeric_state' condition: unknown entity sensor.solax_swxxxxxxxx_exported_power
2024-10-08 11:49:12.004 WARNING (MainThread) [homeassistant.components.homeassistant.triggers.numeric_state] Error initializing 'Minero3 Off (imported below -300W 1min & Minero4-Off 30s)' trigger: In 'numeric_state' condition: unknown entity sensor.solax_swxxxxxxxx_exported_power
2024-10-08 11:49:12.006 WARNING (MainThread) [homeassistant.components.homeassistant.triggers.numeric_state] Error initializing 'Minero4 Off (imported below -300W 30s)' trigger: In 'numeric_state' condition: unknown entity sensor.solax_swxxxxxxxx_exported_power
2024-10-08 11:49:12.007 WARNING (MainThread) [homeassistant.components.homeassistant.triggers.numeric_state] Error initializing 'Minero1 On (imported above -200W 9m & cargabateria ON 9m)' trigger: In 'numeric_state' condition: unknown entity sensor.solax_swxxxxxxxx_exported_power
2024-10-08 11:49:12.009 WARNING (MainThread) [homeassistant.components.homeassistant.triggers.numeric_state] Error initializing 'Minero2 On (imported above -200W 9min & minero1-ON 5min)' trigger: In 'numeric_state' condition: unknown entity sensor.solax_swxxxxxxxx_exported_power
2024-10-08 11:49:12.011 WARNING (MainThread) [homeassistant.components.homeassistant.triggers.numeric_state] Error initializing 'Minero3 On (imported above -200W 9min & minero2-ON 9min)' trigger: In 'numeric_state' condition: unknown entity sensor.solax_swxxxxxxxx_exported_power
2024-10-08 11:49:12.015 WARNING (MainThread) [homeassistant.components.homeassistant.triggers.numeric_state] Error initializing 'Minero4 On (imported above -200W 9min & minero2-ON 9min)' trigger: In 'numeric_state' c [... cut ...]
2024-10-08 11:50:43.747 DEBUG (MainThread) [homeassistant.components.solax] Finished fetching solax SWxxxxxxxx data in 1.246 seconds (success: True)
2024-10-08 11:51:14.745 DEBUG (MainThread) [homeassistant.components.solax] Finished fetching solax SWxxxxxxxx data in 1.244 seconds (success: True)
2024-10-08 11:51:45.748 DEBUG (MainThread) [homeassistant.components.solax] Finished fetching solax SWxxxxxxxx data in 1.247 seconds (success: True)
2024-10-08 11:52:16.756 DEBUG (MainThread) [homeassistant.components.solax] Finished fetching solax SWxxxxxxxx data in 1.255 seconds (success: True)
`

japeral, 2024-10-08:
I realized then the problem reproduces, the it only detects 16 entities on the device. And when all is OK, there is 20 entities (4 more).

japeral, 2024-10-08:
I just digged more into the problem.

[image]

japeral, 2024-10-08:
The discovery detecting mechanism is obviously not working well.
Autodetect intermittent issues like this are going to be difficult to debug and test for all variants.
I believe it is going to be much robust solution to select manually the inverter type from HA from a dropdown menu.

squishykid, 2025-01-18:
Let me take a look at this, thank you for the detailed investigation! I think I can help you here.

squishykid, 2025-01-18:
@japeral can you please copy-paste the data returned from your inverter so I can use it as  test case? I can see it in your first message where you screenshot the curl command.

darmach, 2025-01-18:
@squishykid can this be related to a similar error I've seem on X4?

```
Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:773
integration: Sensor (documentation, issues)
First occurred: January 17, 2025 at 23:13:11 (1 occurrences)
Last logged: January 17, 2025 at 23:13:11

Platform solax does not generate unique IDs. ID SRUJ233UF6-168 already exists - ignoring sensor.solax_battery_mode
```

squishykid, 2025-01-18:
@danmach this is a different issue. We export a sensor with ID '168' twice in the X3 hybrid g4: https://github.com/squishykid/solax/blob/master/solax/inverters/x3_hybrid_g4.py#L134

That should be a relatively quick fix

darmach, 2025-01-18:
> [@danmach](https://github.com/danmach) this is a different issue. We export a sensor with ID '168' twice in the X3 hybrid g4: https://github.com/squishykid/solax/blob/master/solax/inverters/x3_hybrid_g4.py#L134
> 
> That should be a relatively quick fix

Roger, I'll take a stab at it

japeral, 2025-07-02:
> [@japeral](https://github.com/japeral) can you please copy-paste the data returned from your inverter so I can use it as test case? I can see it in your first message where you screenshot the curl command.

Here it is:

root@MSI:/home/jose# curl -d "optType=ReadRealTimeData&pwd=admin" -X POST http://192.168.1.36

{"sn":"SWGU2XXXXX","ver":"2.034.06","type":4,"Data":[2331,99,2374,2012,1319,66,70,1443,936,4997,2,27131,3,105,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,8,0,0,0,0,0,8,0,0,0,0,28002,0,0,0,0,0,0,1025,0,9067,2,26928,69,0,60,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],             "Information":[4.600,4,"XB502999999999",1,3.16,1.05,1.07,0.00,0.00,1]}

root@0200-05535:/home/jose#

japeral, 2025-07-22:
@squishykid @darmach could you have a look to the curl paste when you have time? thanks.

japeral, 2025-08-03:
@japeral I have no idea on the process of running a modified integration in my HA. In which folder exactly do I have to put it in the filesystem to run in, instead of the original image?

I sincerely believe the AUTO detection mode is going to be difficult to maintain. I think there is not enough singular data in the responses to identify uniquely each inverter. Also testing your auto mode is very complicated for you.

I would like to create a variant that always uses my inverter type the "X1-Boost" hardcoded in the config. Then later, add a drop down menu to select the inverter type where the IP and the password is set to make it nicer.