squishykid/solax #173: Question: why is it working without wifi proxy solax x3 hybrid g4 with wifi 3.0

vukisz, 2024-09-03:
Hello. 
I just got a new solar modules installation with solax solax x3 hybrid g4.
Wifi device type is WIFI3.0 version number on app is 3.015.02. Firmware version number when connecting to device ip (using home network): 3.001.02
I can see, that wifi created additional wifi AP. I tried accessing it via APP using local connect and it works as expected.
In HA solax integration description page I read, that newer dongles and firmware would need wifi proxy
Anyway I tried adding solax integration into home assistant from my home network. Provided device ip address (made it static from router) and as a password I used wifi dongle s/n. And it works!
Why does it work without wifi proxy and what should I do to keep it working forwards on? Prevent wifi dongle firmware updates? 
And another question can I disable that open wifi AP form the dongle itself, as I don't need it?

shockerchocker, 2024-09-03:
Just updated firmware.  They've decided to expose the info locally now in a way that's (almost) usable.
Just need them to do it in a properly tagged, documented way so we don't have to reverse engineer lists of a few hundred numbers.
This seems to be a general improvement with them - i've one of the newest X1-IES setups.  It's still got the local access via the dongle access point for 'live' data, but recent updates to the app let you have live data via Solax (10s updates) and there's a tickbox to update the front page diagram every 10s too in the latest update.

squishykid, 2025-01-18:
@vukisz we set a 'forwarded-for' header in the library, and this appears to be enough to convince the inverter that it does not need a proxy. Can you please share a link to the docs where you saw the proxy requirement? I can get this fixed up because-as you noticed- it is no longer a requirement to run a proxy.

@shockerchocker I don't have access to a new inverter like this, can you please provide a bit more information about what you are using to view the data? Is this via an app?

darmach, 2025-01-18:
@squishykid just wanted to let you know that is also works like that for me (just local network connection to solax lan ip)

X3-Hybrid-G4 
ARM Version | 1.26-1.09
Main DSP Version | 1.27
Slave DSP Version | 0.00

Wifi3.0 module
Version number | 3.018.03 (in solax cloud panel)
Firmware Version
3.001.02 (shown in local network web ui of the wifi module)

I recently talked with local Solax support, and there is a newer firmware - although I'm a bit reluctant to upgrade, as the current integrates quite nice! (btw. do we have a list of working firmares somewhere? I'd add my current if it's not listed there yet, or check for the support of newer firmwares)

squishykid, 2025-01-18:
@darmach I don't have a list of working firmware, we currently employ a guess-and-check approach 😅 . If it is not working then I hope that people raise an issue / PR :)

vukisz, 2025-01-18:
> [@vukisz](https://github.com/vukisz) we set a 'forwarded-for' header in the library, and this appears to be enough to convince the inverter that it does not need a proxy. Can you please share a link [...]

It's in home assistant documentation: 
https://www.home-assistant.io/integrations/solax/#note

It's been working  quite good for like 4 months now. And I would like to keep it that way. I hope firmware won't upgrade without asking me and break something :-)

closed 2025-01-18 by vukisz

squishykid, 2025-01-18:
@vukisz do you feel like raising a PR to [home-assistant front-end](https://github.com/home-assistant/frontend)?

Could be an easy change and win you some open-source points :)

You just have to delete the section in the docs related to the note, then link the PR here, I will go comment that it is an expected change.

(plus then I don't have to do it haha)

vukisz, 2025-01-18:
> [@vukisz](https://github.com/vukisz) do you feel like raising a PR to [home-assistant front-end](https://github.com/home-assistant/frontend)?
>
> Could be an easy change and win you some open-source [...]

Which PR I should link to? Or link to this issue?

squishykid, 2025-01-18:
When you raise a PR against https://github.com/home-assistant/frontend, then paste a link to that PR here and I will help review and get it merged. I will try and make the process nice and smooth for you :)

vukisz, 2025-01-18:
I haven't found that info in frontend. But found it here (new pr):
https://github.com/home-assistant/home-assistant.io/pull/36983

squishykid, 2025-01-18:
The frontend change looks great! I approved and left a comment. Thank you!