squishykid/solax #174: X3 Hybrid G4 failed parsing json

mkkot, 2024-09-08:
Hello!
Newer firmware here: 3.010.16

```
{"sn":"SVXXX","ver":"3.010.16","type":14,"Data":[2347,2324,2350,8,12,98,40,192,2294,2526,0,0,0,0,0,0,5003,5002,5002,2,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,31380,64736,62995,3137,65455,62996,1,47,2526,256,14891,1559,6153,100,0,35,0,0,0,0,0,0,0,0,0,0,0,0,0,16356,1,397,7204,0,0,11968,0,14500,0,23,28,14985,1,409,38,0,0,20429,8,40437,4,3396,0,22,0,0,0,0,0,0,0,0,0,1,81,1,25,75,256,3504,2400,231,300,271,247,33,33,140,1620,783,16191,16191,16191,0,0,0,3274,3266,30868,41,20564,12339,18753,12857,16688,13636,14135,20564,12339,18754,12354,16689,13380,14642,20564,12339,18754,12354,16689,13380,12339,20564,12339,18754,12354,16689,13380,14386,0,0,0,0,0,0,0,7681,4355,1537,259,0,31380,0,0,0,0,0,0,0,0,0,0,0,0,0,161,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[8.000,14,"H34A08I4188020",8,1.39,0.00,1.38,1.09,0.00,1]}
```
Unfortunately, parsing fails. Maybe because "Data" is capital?

I would like to use this input to fix the json parser, but not sure how I can put my file instead of http response.

Darsstar, 2024-09-10:
> Unfortunately, parsing fails. Maybe because "Data" is capital?

[This `str.lower()` should rule that out.](https://github.com/squishykid/solax/blob/master/solax/response_parser.py#L131)

When I locally create the ResponseParser instance myself and feed it your json it is able to handle it just fine...

mkkot, 2024-09-10:
Okay, was using wrong inverter type. Thanks!

closed 2024-09-10 by mkkot