squishykid/solax #181: X1 Smart discovery error

AthenaNetworks, 2024-12-16:
Running latest HA which includes solax-3.2.1, X1 Smart is failing discovery.

In previous versions of HA/Solax, this was working with some code tweaks but I never bothered to fork the repo at the time and I've forgotten what my changes were now.

Let me know what debug information I need to provide to further assist you with this. Also happy to provide you with SSH/Web access to my HA VM.

Thanks.

AthenaNetworks, 2024-12-16:
I managed to fix it by hackily defining huge min and max ranges here

```
    # pylint: disable=duplicate-code
    _schema = vol.Schema(
        {
            vol.Required("type", "type"): vol.All(int, 8),
            vol.Required(
                "sn",
            ): str,
            vol.Required("ver"): str,
            vol.Required("data"): vol.Schema(
                vol.All(
                    [vol.Coerce(float)],
                    vol.Length(min=0, max=10000),
                )
            ),
            vol.Required("information"): vol.Schema(vol.All(vol.Length(min=0, max=100))),
        },
        extra=vol.REMOVE_EXTRA,
    )
```

squishykid, 2024-12-16:
Hi, thank you for raising this with us! If you could supply the output from your inverter when you hit the endpoint with curl, then we can add a regression test and make sure this is fixed and doesn’t break in the future.

Issue #75 has an example of a curl command, but you might need to adjust this. Maybe it’s a GET or a POST. Let me know how you go.

AthenaNetworks, 2024-12-16:
curl -d "optType=ReadRealTimeData&pwd=SJ4QUE8RBZ" -X POST http://10.80.111.13/

```
{"sn":"SJ4QUE8RBZ","ver":"3.006.04","type":8,"Data":[2426,20,103,3470,1100,2,1,93,20,5004,2,27207,2,353,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,30,0,6537,0,0,0,0,0,0,63616,65535,37744,11,32984,16,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[8.000,8,"XS80222C212071",8,1.16,1.00,1.09,0.00,0.00,1]}
```

squishykid, 2025-01-18:
Should be fixed in the next release!

PR #184 (2025-01-18): fix discovery for x1 smart with v3 firmware
Fixes #181
closed 2025-01-18 by squishykid