squishykid/solax #188: PocketWifi V2: Unable to connect through LAN

joelekstrom, 2025-02-24:
Hello! I'm unable to connect to direct IP to my PocketWifi V2 - meaning I have to set up HA to connect to the AP provided by the PocketWifi itself.

I'm a bit confused about why - since I can `curl` with `X-Forwarded-For: 5.8.8.8` just fine, and my custom REST sensors also work fine this way, I imagine it should work with this integration too. I see in the code that it also adds the `X-Forwarded-For: 5.8.8.8` header.

I wish I could help more, but I've enabled debug logging and I get zero logs from this integration. The only thing I get is that it "Cannot connect" when trying with the direct IP, but works when I connect to `5.8.8.8` via local AP. I have an X3-Hybrid-G4.

Anything else I can do to help debug this? I would really like to be able to connect directly since the local AP is quite unreliable, and this integration will simply stop working over it sometimes. My custom REST-sensors keep working fine that go via LAN directly so I imagine it would improve my situation

joelekstrom, 2025-02-25:
OK, I think I see the problem. It's that X-Forwarded-For is not applied for [X3-Hybrid-G4](https://github.com/squishykid/solax/blob/6fe33f7e5f87b63657d6a19cdc839eaa607e047b/solax/inverters/x3_hybrid_g4.py). I'll see if I can test fixing that and submit a PR

darmach, 2025-02-25:
Hi @joelekstrom I have the same inverter. What firmware are you running on you pocket Wifi?

I have X3-Hybrid-G4 and I can connect to API address in local network just fine, both with curl and this integration.

Did you do the usual troubleshooting network-wise? (able to ping the endpoint, curl etc.)

joelekstrom, 2025-02-25:
@darmach  I'm able to ping it, and `curl` works fine, but requires `X-Forwarded-For: 5.8.8.8`. Do you have Pocket Wifi 3+? This is an issue for Pocket Wifi 2 specifically.

```
curl -d "optType=ReadRealTimeData&pwd=..." -X POST http://192.168.1.128 -H "X-Forwarded-For: 5.8.8.8"

{"sn":"SWSGFTFUBH","ver":"2.035.01","type":14,"Data":[2338,2315,2228,17,13,8,341,198,54,593,4384,3691,6,8,274,320,5000,5001,5001,2,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,23290,0,1,2317,0,0,1,42,593,256,14382,6410,6402,356,0,33,0,0,0,0,0,0,0,0,0,0,0,0,0,13466,3,9,7049,0,0,56781,0,63566,0,0,1,29571,3,10,0,0,0,61795,8,49997,39,0,0,3529,0,0,0,0,0,0,0,0,0,1,10,1,18,10,256,2628,1800,298,200,224,189,32,32,485,1106,525,10537,10537,0,0,0,0,3222,3213,43138,183,21302,14389,18497,12345,16690,12355,13368,21302,14389,18497,12345,16690,12355,13368,21302,14389,18498,12345,16691,12355,14646,0,0,0,0,0,0,0,0,0,0,0,0,0,0,257,770,260,1537,0,23290,0,0,0,0,0,0,0,0,0,0,0,0,0,161,96,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],		"Information":[12.000,14,"H34T12H7128015",1,1.50,0.00,1.47,1.07,0.00,1]}
```

darmach, 2025-02-25:
@joelekstrom Ah, really sorry for the commotion then, indeed, mine is 3.0!

joelekstrom, 2025-02-25:
No problem! I see this header is applied for some other inverters so should be an easy fix. Just need to figure out a python/homeassistant environment to test

PR #189 (2025-02-26): Fix X3 Hybrid G4 with PocketWiFi V2 not working via LAN
It would only work when connected directly to the PocketWiFi AP, since the V2 requires an X-Forwarded-For: header.

I have tested locally both with discovery and direct access and the data is populated correctly.

Fixes #188
closed 2025-02-26 by squishykid