squishykid/solax #191: Please add support for X1-Micro 2 in 1

Mandrake1967, 2025-04-09:
Hi, have been trying to use the Solax integration to add my new Micro Inverter.
It does not get detected automatically and when I try to manually add it, the connection is refused.
I only bought this damn inverter as I thought it would integrate with HA!
Any help or guidance very much appreciated!

Mandrake1967, 2025-04-09:
OK, I fail at the first hurdle
When I try to connect the Inverters Host IP I get
Request method for this URI is not handled by server

I can connect via Pocket Wifi to the hotspot, but I don't see any data either. I wonder if local support has been not been added to the firmware yet. I think these are quite new devices for Solax
I can see everything via the Solax cloud normally

Mandrake1967, 2025-05-05:
Hello, I have made some progress!
The X1 was on an original firmware and with the help of Solax we upgraded the microinverter.
I can now connect to the inverter locally via its IP address to a web browser, however the solax integration is not working.
Any steps I can follow?

CurlyMoo, 2025-08-16:
@Mandrake1967 I'll be receiving the 2 in 1 micro inverter today. What would be that latest firmware version? That will help me check if i'm running the latest version. If i have it up and running, i will can help add support.

CurlyMoo, 2025-08-20:
I realized i forgot to capture the initial connection to the Solax inverter through the local IP.

The initial request is can be done by this simple request. This works for both for the internal IP of the inverter as for the DHCP IP from the router.
```
curl -d "optType=getModuleInfo" -v -X POST http://x.x.x.x
```

With this response:
```json
{"code":200,"message":"success","data":{"version":"2.0","encryption":0,"category":2,"deviceType":14,"inverters":[{"sn":"XXXXXXXXXXXXXX","category":1,"deviceType":28}]}}
```

CurlyMoo, 2025-08-21:
So the respones from the inverter contain all values in plain hex. This is what i found thus far.

Example HEX:
```
24 24 70 00 08 04 01 1c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 05 0e d5 67 c3 fd a6 d4 48 50 b3 de d7 07 dd 7b c2 c6 95 7e ea bb 5b 53 4a b6 a2 56 9e 39 ca 7a 23 ee 2d 00 81 00 00 28 00 20 09 03 00 4d 00 89 13 c4 01 cd 01 08 00 0b 00 26 00 33 00 02 00 41 00 00 00 20 00 22 00 22 00 03 00 00 00 00 00 00 00 67 66
```
Serial intentionally blanked with zero's.

```
Byte 0-1 = START
Byte 2-3 = Response type
Byte 4-7 = ?
Byte 8-21 = Inverter Serial
Byte 22-71 = ?
Byte 72-73 = Inverter current
Byte 74-75 = Inverter grid current
Byte 76-77 = Inverter grid hertz
Byte 78-79 = Solar PV1 V
Byte 80-81 = Solar PV2 V
Byte 82-83 = Solar PV1 A
Byte 84-85 = Solar PV2 A
Byte 86-87 = Solar PV1 W
Byte 88-89 = Solar PV2 W
```

CurlyMoo, 2025-08-22:
I managed to receive data from the inverter by sending one of the requests back to the inverter. In this case:

```
24 24 45 00 08 04 01 1c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 09 2b 3c d3 4f 18 42 bd b9 41 53 86 f2 04 e9 65 77 d2 9d cf c0 82 45 92 9f 03 51 45 6a 65 db fd 03 00 01 00 00 b1 9b
```

I intentionally left the device id blank.

I then sent this string base64 back to the inverter:

```
JCRFAAgEARwAAAAAAAAAAAAAAAAAAAAAAAAAAAAECSs8008YQr25QVOG8gTpZXfSnc/AgkWSnwNRRWpl2/0DAAEAALGb
```

```
curl \
-d "JCRFAAgEARwAAAAAAAAAAAAAAAAAAAAAAAAAAAAECSs8008YQr25QVOG8gTpZXfSnc/AgkWSnwNRRWpl2/0DAAEAALGb" \
-X POST http://x.x.x.x
```

It doesn't matter if you use the local IP of the internal network IP of the router.

With this request i get the response i need which i can then decode with the above logic.

@Mandrake1967 Can you let me know what the reponse is when you sent: 
```
curl -d "optType=getModuleInfo" -X POST http://x.x.x.x
```

Guigui37, 2025-08-27:
Hey, I also got the X1-Micro 2 in 1, also not working at this time in HA, also running FW version 008.03.
Here is the result of the command
```
$ curl -d "optType=getModuleInfo" -X POST http://192.168.1.225
 {"code":200,"message":"success","data":{"version":"2.0","encryption":0,"category":2,"deviceType":14,"inverters":[{"sn":"SERIALNUMBER","category":1,"deviceType":28}]}}
```
I am available if you want me to test anything.

Thanks !

CurlyMoo, 2025-08-27:
Can you use this PHP script:
```php
<?PHP 
$ip = "http://x.x.x.x/";
$serial = "XXXXXXXXXXXXXX";

$curl = curl_init();
curl_setopt($curl, CURLOPT_URL, $ip);
curl_setopt($curl, CURLOPT_POST, true);
curl_setopt($curl, CURLOPT_RETURNTRANSFER, true);
curl_setopt($curl, CURLOPT_CONNECTTIMEOUT, 3); 
curl_setopt($curl, CURLOPT_TIMEOUT, 3); //timeout in seconds

$hex = explode(" ", "24 24 45 00 08 04 01 1c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 09 2b 3c d3 4f 18 42 bd b9 41 53 86 f2 04 e9 65 77 d2 9d cf c0 82 45 92 9f 03 51 45 6a 65 db fd 03 00 01 00 00 b1 9b");
$i = 0;
foreach (str_split($serial) as $chr) {
	$hex[8+$i] = bin2hex($chr);
	$i++;
}

$data = base64_encode(hex2bin(implode($hex)));

curl_setopt($curl, CURLOPT_POSTFIELDS, $data);

$resp = curl_exec($curl);
curl_close($curl);

$resp = str_replace("\n", "", $resp);

$decoded = base64_decode($resp);

if(strlen($decoded) == 112 && ord($decoded[2]) == 0x70 && strcmp(substr($decoded, 8, 22), '30M3Y10115028H')) {
	echo 'PV1: '.(((float)((ord($decoded[79]) << 8) | ord($decoded[78])))/10)."v / ".(((float)((ord($decoded[83]) << 8) | ord($decoded[82])))/10)."A / ".((int)((ord($decoded[87]) << 8) | ord($decoded[86])))."W<br />";
	echo 'PV2: '.(((float)((ord($decoded[81]) << 8) | ord($decoded[80])))/10)."v / ".(((float)((ord($decoded[85]) << 8) | ord($decoded[84])))/10)."A / ".((int)((ord($decoded[89]) << 8) | ord($decoded[88])))."W<br />";
	
	echo 'Grid: '.(((float)((ord($decoded[71]) << 8) | ord($decoded[70])))/10)."v / ".(((float)((ord($decoded[73]) << 8) | ord($decoded[72])))/10)."A / ".(((int)((ord($decoded[75]) << 8) | ord($decoded[74]))))."W / ".(((float)((ord($decoded[77]) << 8) | ord($decoded[76])))/100)."Hz <br />";
	
	echo "Opgewekt: ".((int)((ord($decoded[97]) << 8) | ord($decoded[96])))." / ".(((float)(((ord($decoded[95]) << 24) | (ord($decoded[94]) << 16) | (ord($decoded[93]) << 8) | ord($decoded[92]))))/10)."<br />";
}
?>
```

CurlyMoo, 2025-08-27:
What i forgot to mention in the previous post is that you first need to convert the hex to binary.

You can use this online tool to directly encode hex as base64 without having to transform everything to binary first:
https://cryptii.com/pipes/hex-to-base64

CurlyMoo, 2025-08-27:
My firmware is a little bit older:
```
Firmware information
wifi 007.03 DSP 007.04
```

CurlyMoo, 2025-08-27:
So the theory was that the last two bytes are a checksum. That would explain why it doesn't work in your case. With another serial, we would need another checksum. With the help of ChatGPT i came up with this checksum calculator that worked for all payloads i captured before.

```php
<?PHP 
/**
 * CRC16 (poly=0x8005, init=0x0000, refin=false, refout=false, xorout=0x0000)
 * Big-endian result (high byte first).
 */
function crc16_nonref(string $data) {
    $poly = 0x8005;
    $reg  = 0x0000;
    $mask = 0xFFFF;

    $bytes = unpack("C*", $data);
    foreach ($bytes as $b) {
        $reg ^= ($b << 8);
        for ($i = 0; $i < 8; $i++) {
            if ($reg & 0x8000) {
                $reg = (($reg << 1) ^ $poly) & $mask;
            } else {
                $reg = ($reg << 1) & $mask;
            }
        }
    }
    return str_split(dechex($reg & 0xFFFF), 2);
}

$ip = "http://x.x.x.x/";
$serial = "XXXXXXXXXXXXXX";

$curl = curl_init();
curl_setopt($curl, CURLOPT_URL, $ip);
curl_setopt($curl, CURLOPT_POST, true);
curl_setopt($curl, CURLOPT_RETURNTRANSFER, true);
curl_setopt($curl, CURLOPT_CONNECTTIMEOUT, 3); 
curl_setopt($curl, CURLOPT_TIMEOUT, 3); //timeout in seconds

$hex = explode(" ", "24 24 45 00 08 04 01 1c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 09 2b 3c d3 4f 18 42 bd b9 41 53 86 f2 04 e9 65 77 d2 9d cf c0 82 45 92 9f 03 51 45 6a 65 db fd 03 00 01 00 00 00 00");
$i = 0;
foreach (str_split($serial) as $chr) {
	$hex[8+$i] = bin2hex($chr);
	$i++;
}

$crc = crc16_nonref(substr(hex2bin(implode($hex)), 0, -2));
$hex[67] = $crc[0];
$hex[68] = $crc[1];	

$data = base64_encode(hex2bin(implode($hex)));

curl_setopt($curl, CURLOPT_POSTFIELDS, $data);

$resp = curl_exec($curl);
curl_close($curl);

$resp = str_replace("\n", "", $resp);

$decoded = base64_decode($resp);

if(strlen($decoded) == 112 && ord($decoded[2]) == 0x70 && strcmp(substr($decoded, 8, 22), $serial)) {
	echo 'PV1: '.(((float)((ord($decoded[79]) << 8) | ord($decoded[78])))/10)."v / ".(((float)((ord($decoded[83]) << 8) | ord($decoded[82])))/10)."A / ".((int)((ord($decoded[87]) << 8) | ord($decoded[86])))."W<br />";
	echo 'PV2: '.(((float)((ord($decoded[81]) << 8) | ord($decoded[80])))/10)."v / ".(((float)((ord($decoded[85]) << 8) | ord($decoded[84])))/10)."A / ".((int)((ord($decoded[89]) << 8) | ord($decoded[88])))."W<br />";
	
	echo 'Grid: '.(((float)((ord($decoded[71]) << 8) | ord($decoded[70])))/10)."v / ".(((float)((ord($decoded[73]) << 8) | ord($decoded[72])))/10)."A / ".(((int)((ord($decoded[75]) << 8) | ord($decoded[74]))))."W / ".(((float)((ord($decoded[77]) << 8) | ord($decoded[76])))/100)."Hz <br />";
	
	echo "Generated: ".((int)((ord($decoded[97]) << 8) | ord($decoded[96])))." / ".(((float)(((ord($decoded[95]) << 24) | (ord($decoded[94]) << 16) | (ord($decoded[93]) << 8) | ord($decoded[92]))))/10)."<br />";
}
?>

CurlyMoo, 2025-08-27:
It would be great if you could help decode the unknown bits i haven't decoded yet. Once the requests work that is.

Guigui37, 2025-08-28:
It works !

`$ php testsolax2.php 
PV1: 36.1v / 0A / 0W<br />PV2: 36.1v / 0A / 0W<br />Grid: 234.7v / 0A / 0W / 49.98Hz <br />Generated: 0 / 4.5<br />guigui@guigui-laptop-12th:~/solax$`

(but no sun for now, only rain).

I will check the other bits.

CurlyMoo, 2025-08-28:
I also see that you have just installed the inverter. Hardly any kWh generated in total. Did you do it yourself?

CurlyMoo, 2025-08-28:
I did found some requests and responses that use a different payload in byte 4 - 7 `08 04 02 0e`. In case such a request is made, the request byte 2 - 3 are: `45 00`, but the responses container either `61 00` or `4f 00`. I'm not sure if those are incomplete requests and responses, because the number of total bits of the response is far less.

I also believe that only bits 30 - 64 contain the hash (in both requests and response), because in the requests bit 65 is always 00 and bit 66 and 67 contain the CRC.

I believe that in the responses bits. I believe the response bits 65 - 69 also contain data.

CurlyMoo, 2025-08-28:
Some additional info. Sending this hex also works fine:
```
24 24 45 00 08 04 01 1c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00
```
The `01` as the end is required.

Guigui37, 2025-08-28:
I found a mistake on the day generated.
Before generating energy:
Generated: 0 / 4.5
After having generated 0.6kWh:
Generated: 6 / 5.1
it should not be 6 but 0.6

CurlyMoo, 2025-08-31:
@Guigui37 @Mandrake1967 I noticed that with each startup of the inverter the SSID is again broadcasted despite what you've set.

For now, the only solution seems to be to disable the SSID again after each boot through the API:
```
curl -d '{"show":0}' -H 'token: XXXXXX' http://x.x.x.x/api/wifi/setSsidVisibility
```

Guigui37, 2025-09-02:
I confirm it change the value of netSet.
I will check tomorrow if it switch back to 1 or not.

CurlyMoo, 2025-09-02:
@Mandrake1967 Have you found the time to test as well?

CurlyMoo, 2025-09-03:
But did you check if it actually stopped broadcasting despite the setting?

Guigui37, 2025-09-03:
I just checked, it stopped broadcasting the SSID, the wifi networks remains, but hidden.
I can check again tomorrow without touching anything.
FYI I set the show to 0 with the command you gave me, not through the android app.

CurlyMoo, 2025-09-09:
Did you manage to "check tomorrow"?

Guigui37, 2025-09-21:
Hey @Mandrake1967 , did you had time to check ?

CurlyMoo, 2025-09-21:
@Guigui37 and were you able to confirm that the SSID broadcasting still stopped the next day?

Knetus56, 2026-08-28:
https://github.com/Knetus56/solax_local  update

tszyszko, 2026-08-28:
I've tested v1.4.0 but not having any luck. Home assistant can connect to inverter but I'm just getting unknown / 0.0 values

Knetus56, 2026-08-28:
open issue

tszyszko, 2026-08-28:
Turns out I am running 004.6 firmware, so may need to update to work

webpatrick, 2026-09-01:
> Turns out I am running 004.6 firmware, so may need to update to work

I send a message to Solax Support and my inverters were updated wihtin minutes.