squishykid/solax #192: new solax dongle firmware requires username + password authorization

soppem, 2025-05-18:
The latest firmware requires a username and password. However, the integration does not take into account the username and therefore authorization is not possible.

[image]
[image]

darmach, 2025-05-23:
Which dongle model no and fw version is that?

soppem, 2025-05-25:
> Does it work with the `admin` username?

Logging in to the dongle's web interface with the admin account and the set password works. But in the Hass integration, only the password is entered.

webknjaz, 2025-05-29:
@soppem you probably shouldn't expose your device's serial number in public